OpenInverterGateway/OpenInverterGateway #4: Flashing the ShineWiFi-X module

Cruzxia, 2022-01-25:
To flash the ShineWiFi-X(Usb), you need to temporarily remove R29 and R30. 
This prevents the onboard USB serial IC interfering with the serial data you are flashing. 
(Don't forget to put the resistors back once the module has been flashed) 

On the ShineWiFi-X module connect your serial programmer Tx to RX0 (on the ShineWiFi-X PCB), and Rx to TX0(on the ShineWiFi-X PCB)

Lastly if you are using MQTT and have two systems on the same broker, you will have to change the name of the MQTT_LAST_WILL
so that they are unique for each system. otherwise both systems keep disconnecting.
I added a name definition and altered the Mqtt... statement in the code as follows:
#define MQTT_LAST_WILL    "S1Growatt" // Must be unique if more than one system
if (MqttClient.connect(MQTT_LAST_WILL, MQTT_TOPIC, 1, 1, "{\"Status\": \"Disconnected\" }"))

Great project
regards Cruzxia

StevenBoelhouwer, 2022-01-25:
If you install the correct drivers and connect GPIO0 to GND when powering the ShineWiFi-X on it will let you flash the firmware over the onboard USB serial. No need to desolder and resolder anything.

Cruzxia, 2022-01-25:
Hi Steven Thanks for the prompt response. I did try to use the onboard usb serial, but it would not communicate. Prehaps the usb driver on my pc was wrong. It is all up an running now on two systems.Thanks again.CruzxiaSent from my Galaxy

StevenBoelhouwer, 2022-01-25:
Get the correct (USB UART) driver here: https://www.maxlinear.com/support/design-tools/software-drivers.
Shorten the GPIO0 to GND on powering up and its ready to be flashed. From top of my head the serial speed should be set to "115200". 

I'm running a forked version of this code a couple of months already. I've added HomeAssistant discovery and WiFiManager (for easier setup of WiFi) to my code but it needs cleaning up before I want to initiate a pull request.

jwortelboer, 2022-01-25:
Thanks Steven, that worked like a charm!  Uploading worked with everything default. No soldering, just shortening the two pin holes during boot and voila. Now I have to wait till daylight to see the code running on my inverter ;-)   Yes the WiFi setup could be more generic, Configuration during compile time has some drawbacks, but this is already great fun to play with.  Looking forward to your additions/modifications.

achim-70, 2022-08-19:
Hi,
I have the ShinWifi-X stick with the soldered resitor. After connecting with my PC all three leds are glowing. But about one minute later they turn of. Is the stick at this phase still in flashing mode? The compiling process takes always longer and when the upload process wants to start the leds are turned of and I get the error message "esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header".

Cees2006, 2022-08-21:
@achim-70  No, it's not in flash mode. You must connect GPIO0 to GND before plugged in (see the description here). then disconnect gpio0 from gnd. then only the red and blue led will light up (I think).

achim-70, 2022-08-22:
Hi @Cees2006,

thanks for the hint about the red and blue light. That got me further. I simply bridged GPIO0 and GND with a paper clip and connected the stick to the PC. Then the stick was in programming mode. I don't understand why a resistor is soldered on shown [on the picture in the documentation](https://github.com/otti/Growatt_ShineWiFi-S/blob/master/Doc/ShineWiFi-X/ShineWiFi-X_PCB_Bot.jpg). But it doesn't matter to me now because I've achieved my goal and was able to program the stick.

Btw I disconnected the bridge after the programming finished.

[image]

MichelFreund, 2022-09-04:
> Hi @Cees2006,
>
> thanks for the hint about the red and blue light. That got me further. I simply bridged GPIO0 and GND with a paper clip and connected the stick to the PC. Then the stick was in [...]

Hi achim-70, thanks for the info. Which driver / tool did you use to do the flashing? Im having troubles finding the right steps to flash my shinewifi-x.

achim-70, 2022-09-05:
@MichelFreund

Driver: https://www.maxlinear.com/content/document.ashx?id=1596
Tool: Arduino IDE (https://www.arduino.cc/en/software)

crasu, 2022-12-04:
Flashing process is now documented [here](https://github.com/otti/Growatt_ShineWiFi-S/blob/master/Doc/ShineWiFi-X/ShineWiFi-X.md#loading-firmware)

closed 2022-12-04 by crasu