OpenInverterGateway/OpenInverterGateway #5: Original FW

soulraven, 2022-01-30:
hello,
do you have any idea where to find the original firmware? I have two WIFI-F sticks, both bricked after a failed firmware update from OSS platform.

sohnvonoff, 2022-02-02:
hi soulraven,
i hope it's that what you need.
[FlashBackup_8MB.zip](https://github.com/otti/Growatt_ShineWiFi-S/files/7989960/FlashBackup_8MB.zip)
Urgent!
Install the driver for UART (XR21V14xx) to flash direct over USB. (without solder) Only jumper between GPIO00 and GND.
[https://www.maxlinear.com/support/design-tools/software-drivers](url)
The default windows driver don't work with XR21V14xx (UART in ShineWifi-x)
I hope it works, I haven't tested it myself.

soulraven, 2022-02-02:
hello,
the modules have ESP-07S, that is having only 4MB not 8MB. I guess is from  ESP8266.

WIFI-F is USB connection, but the Data line are in fact RX/TX pins from the MCU.

sohnvonoff, 2022-02-02:
sorry, I read somewhere  8MB Flash. 
here the  4MB version
The flash is only up 0x6c4c0 written. behind is only 0xff
[FlashBackup_4MB.zip](https://github.com/otti/Growatt_ShineWiFi-S/files/7990474/FlashBackup_4MB.zip)
This file is readout from original Growatt-ShineWIFI-X with esptool.py

soulraven, 2022-02-03:
thx, is working, I manage to flash the ESP, the problem is now that is from WIFI-X not WIFI-F. and I am stuck with the config for WI-FI network. The procedure is hard or bad translate on Shine app.
The best news is that somehow the firmware knows the original SN from the sticks, i guess is calculated from chipID and flashID. or something like this.

ricardordl, 2022-02-03:
Hi,

I'm having the same problem here, I managed to program the esp with original firmware but I couldn't configure the network, the Wifi configuration page is not accessible and the app doesn't work. I believe something is missing, maybe some file in the spiffs. Could you describe what parameters you used to upload the firmware?

soulraven, 2022-02-03:
to write the firmware i have used: esptool.py -b 115200 --port COM5 write_flash -fm dio -fs detect 0x0 FlashBackup_4MB.bin
But keep in mind that the original firmware file from WIFI-X is not containing any html elements or any web server. I guess is a blank TCP server that is only connect to Growatt server and send data. 

I have in mind a solution to get the original FW files from Growatt server using FOTA. But i need more help with that. In the firmware I have found some references to user1.bin file. but I not find any url or path to that file.

soulraven, 2022-02-03:
yes,
is the jump to user1 memory location, from the boot sequence.
The memory is mapped something like, boot,user1,user2. Or something like this.

The bin file contains one, or more sections, spit by memory locations.

soulraven, 2022-02-09:
I have dump from a brand new WIFI stick.
The stick is Growatt Shine WIFI-F (shite case)
The version or firmware is 3.0.0.0.

waite902, 2022-02-25:
Thanks soulraven.  I managed to brick my shinewifi-F during an update as well.  Hoping to flash the image you posted.  Can you clarify what process and drivers you used to connect to the USB on the -F?  (since it is different to the -X with no Maxim XR21V14xx chip)?

soulraven, 2022-02-25:
hi,
follow the pin headers from the stick and from this repository, and connect a USB-RS232 convertor using 3V power supply.
from there using the esptool flash the esp-07S, geek in mind the module has 4Mb. You can flash only using the pin header, is more easy. The stick in fact has a rs485 to rs232 converter connected to D+, D- on USB pins, are not USB compatible, only the connector is USB.

> Thanks soulraven. I managed to brick my shinewifi-F during an update as well. Hoping to flash the image you posted. Can you clarify what process and drivers you used to connect to the USB on the -F? (since it is different to the -X with no Maxim XR21V14xx chip)?

gekkehenkie11, 2022-03-05:
> tks, this is what i got when listen the serial port. I don't know if it means something but we have same reference to user1 here:

@ricardordl 
What baudrate did you use? I used an UART to log the original FW, I tried with a lot of baudrates, first got always garbled up chars, but then tried 72.000 and finally got some cleartext:

>  ets Jan  8 2013,rst cause:2, boot mode:(3,1)

>load 0x40100000, len 2408, room 16
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0
tail 8
chksum 0xd8
csum 0xd8

>2nd boot version : 1.6
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(1024KB+1024KB)
jump to run user1 @ 1000

>OS SDK ver: 2.0.0(e271380) compiled @ Mar 30 2018 18:54:06
phy ver: 1055_1, pp ver: 10.7

>rf cal sector: 1019
tcpip_task_hdl : 40107a00, prio:10,stack:512
idle_task_hdl : 40107ab0,prio:0, stack:384
▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒8, prio:2,stack:512

See how you have different flash sizes. I guess you're not on Shinewifi-X?

Also, the 'jump to run user! @ 1000' is where it does exactly that: the bootloader jumps to the user code. The program entry indeed is at offset 0x1000 in the dumped firmware from @sohnvonoff above (I checked with IDA). So the first 0x1000 bytes are the bootloader. I also saw that indeed that "OS SDK ver" string is in the bytes below that so that indeed must be user code (the 'growatt' program). 

Anyway, my problem is that, as you can see on the lowest line, after dumping this first part it garbles up again. I want to read all output, the uart is dumping more but it's garbled up. Not sure what I'm doing wrong. Maybe its the baudrate? I was wondering if any of you guys had better luck dumping the original FW output via the UART?

BTW @soulraven where is that reddit discussion? I would like to get more info the original FW

soulraven, 2022-03-05:
I have uploaded the original fwn dump here and also is on my git, look for Growatt-monitor inside the doc folder.

gekkehenkie11, 2022-03-05:
But that's shinewifi-F, right? I'm on shinewifi-X

soulraven, 2022-03-05:
> But that's shinewifi-F, right? I'm on shinewifi-X

Have you try this one?
Then author dumped from X version

https://github.com/otti/Growatt_ShineWiFi-S/files/7989960/FlashBackup_8MB.zip

gekkehenkie11, 2022-03-05:
With the help of the comments elsewhere on this page I could dump my own (just needed to short GND/GPO0 also for dumping the FW). BTW I was wrong above, the program entry is not 0x1000, that's just the offset of the user image in the file. Program entry code is shown with "image_info" option of esptool.py.

soulraven, 2022-03-05:
> With the help of the comments elsewhere on this page I could dump my own (just needed to short GND/GPO0 also for dumping the FW). BTW I was wrong above, the program entry is not 0x1000, that's just [...]

happy to hear that, is normal that every action done on the eeprom is done with the device in boot mode.

gekkehenkie11, 2022-03-06:
Not sure if this is the right place, but also don't know a better place: I wanted to give a little info on my update on reverse engineering the firmware with IDA pro, it might help someone out who is also trying to reverse engineer the original firwmare. 

So the firmware has 2 images. The first 0x1000 is the bootloader, after that is the growatt user code. You cut the file at 0x1000 so you'll end up with 2 images. Both can be analyzed with the "image_info" option of esptool.py. It will tell you the segments, loading addresses and the lengths. One important thing is that if you load the segments to the right addresses in IDA, you should cut off the headers (first 8 bytes). 

So the problem I've been facing is that esptool shows address '0' for the big section in the 2nd image. Obviously that's not the right address. I actually wrote a small tool to find the address, I did so by using the known length of 3 strings following each other. Turned out that the loading address was 0x4020101E, which btw is also the lowest 0x402xxxxx address that I had seen as pointer in the file. Not sure if there's an easier way to find it.

Anyway, I now have all sections loaded at the right sections. Next going to look for some libraries to generate some FLAIR/FLIRT files for IDA

soulraven, 2022-03-06:
great to hear that, i have also try with IDA Pro but i don't have find many information regarding the functions. Personally i was interested in the OTA procedure and the RS485 protocol.
I have to admit that was the first time in IDA and only using the free version, and somehow manage to extract the elf file and try to decompile using xtensa plugin.

gekkehenkie11, 2022-03-13:
This is not really a good project for people without IDA experience I'm afraid, much easier to just start with a windows exe and learn from that. Anyway quite a few things were new for me too, so going to share some more info, maybe somebody might find it useful:
First of all, those addresses in the 0x4000xxxx region are in ROM. This actually makes it great for reverse engineering, since we don't need any flair signatures for this 'kernel library', we just the addresess!! For example ets_printf  print is 0x400024cc. The complete list is here: https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/ld/eagle.rom.addr.v6.ld

Secondly, all memory registers are mapped here: https://github.com/pfalcon/esp8266-re-wiki-mirror/blob/master/Memory_Map.mw

So it should be easy to make an IDC script file that labels all those functions and addresses in IDA.

gekkehenkie11, 2022-03-13:
BTW if anyone managed to log output from the serial port beyond the bootloader, please let me know how! I just can't get it to show correct output, not sure what i'm doing wrong. With IDA I clearly see the firmware sets either 9600 or 115200 as baudrate (I assume this is also a general FW for shinewifi_S and shinewifi_X so that's why it selects either of those 2) but somehow I'm not getting correct output, only garbled characters...

soulraven, 2022-03-15:
Will be if you can document the procese and share with us. Personally in this time I have reverse the api on iOS application and document it using OpenApi. Now I am working on a small framework that will include some of the libs what are oriented to Growatt application.

gekkehenkie11, 2022-03-15:
Well i mainly just followed the guide above. I have IDA 7 so I didnt need to adapt that "IDB2PAT" file, it just ran out of the box.

gekkehenkie11, 2022-03-19:
I don't have access to my inverter for another week but I did reverse engineer the uart init code a bit more. So the Shinewifi-X initializes both the UART's. The Uart0 is initialized with 9600 baud and Uart1 (TX only) is initialized with 115200. Both  use standard parameters (UART_WordLength_8b, no parity, 1 stop bit, no flow control). However one thing that's different from the RTOS SDK2 2.0.0 example code is that they set ART_RxFlowThresh to 0xF80. Not sure if that has any impact. And then, they set the UART Printport to UART1. WHich is kinda funny since I did 99% of my testing on UART0. So for debug output, connect your UART to the TX1 serial port and set baudrate to 115,200.

gekkehenkie11, 2022-03-23:
Yihaaaaa FINALLY output!! I just put the device in my PC usb port (instead of inverter). Will try it on an inverter later today and post full log output

gekkehenkie11, 2022-03-24:
Just wanted to share how to enable debug output. Search your FW dump for : 22 41  00 22 4C 00 (all hex) and then change that 0x4C into 0x41. Not sure if it's any different in any other FW's, but on Shinewifi-X it should work (probably also on other FW's). And then just connect as described above.

BeoQ, 2022-03-25:
Hello gekkehenkie11,

Stumbled across this thread, and noticed you have different WiFi-Stick.
This modified Firmware here was originally run on a ShineWiFi-S (showing a 9-pin serial connector).
We managed to make it work on a ShineWiFi-X (black, showing an USB-Connector).
The main difference was modifying the BaudRate in the code (9600Bd vs 115200Bd), and some Modbus Register locations due to newer Protocol version.
BTW: I have also successfully tried to use a Wemos-D1 and a NodeMCU instead of the ShineWiFi-X on my Converter; Their USB-to-Serial chip did work for me without any modification.

Apparently now there is a ShineWiFi-F version (white, showing an USB-Connector), that you seem to have.
Did you connect it to you PC using USB (probably via extension cable) to flash? - This did work fine with the ShinWiFi-X, because its internal USB-to-Serial converter is already connected right.
Did you get any Board-identification from esptool? (i.e. does it report a esp8266 or esp8255 or other?)

Did you manage to open the enclosure of the Stick, an take out the PCB? 
if yes, could you share pictures of both sides of the PCB and how to open it's enclosure?

I would expect, that the -F version of the Stick is basically the same as the -X version: A esp8266 (-variant) connected to the Converter thru a USB-to-Serial-Bridge chip, along with some EEprom and a Battery-Backed-RTC. The latter two are not used by our firmware here. 
In the Firmware, we're just using one serial (the 8266's default rx0/tx0) for communication to the converter.
There are also debug messages on the same serial, the converter seems to ignore them happily (since they don't start with its Modbus-Address, and have no valid Modbus-CRC)

Suggestion: our firmware here tries some auto-detection of the converter, and then selects the baudrate on serial0 appropriately. Since this is likely to fail on an not-yet-implemented converter (your 3000TL), you better disable/comment out the autodetection, and hardcode the baudrate. I suggest to start with 115200Bd.
If you still have the original FW running, you can try to connect it to your converter, and additionally listen to tx0/rx0 on the PCB, This way you should find out the correct baud rate for shure. (do exepct binary-Modbus-communication here. A typical message starts with the Modbus-Address of the converter, which is 0x01 in our case).

Your Dump suggests 9600, 38400, 115200 Baud.
The finding of 74880 Bd on Tx1 seems somewhat unusual.

Once you get the Baudrate right, you can peek around in the various Modbus-Registers using our Firmware's Modbus_RW feature. Well, just the Input Registers for now (as opposed to the Holding registers). 

Pleas share your results!

Good Luck,
Flo

soulraven, 2022-03-25:
> Apparently now there is a ShineWiFi-F version (white, showing an USB-Connector), that you seem to have.
> Did you connect it to you PC using USB (probably via extension cable) to flash? - This did [...]

i will answer in partial on you questions:

> Apparently now there is a ShineWiFi-F version (white, showing an USB-Connector), that you seem to have.
> Did you connect it to you PC using USB (probably via extension cable) to flash? - This did [...]

Yes, the board has a serial to modbus converter on USB data pins and also has header pins on serial side, is the method used to flash the board. The module is a ESP-07S and is report as ESP8266. 

> I would expect, that the -F version of the Stick is basically the same as the -X version: A esp8266 (-variant) connected to the Converter thru a USB-to-Serial-Bridge chip, along with some EEprom and [...]

Firmware base the boards are not the same. I have personally flash WIFI-X firmware on WIFI-F and is not working, the mapping of leds is different and also the WIFI in AP mode is working strange.

More information will have when a decompiled version of the firmware will be available

BeoQ, 2022-03-25:
Soulraven,

i had a similar problem last year, when i tried to use the -X: 
I had overwritten the original FW without backup; so i was stuck to finding out myself how to talk to my converter.
I knew from otti's firmware, that the protocol is very likely Modbus, and i had found various Documentations on the Register set.

Once the correct Baudrate was found, i tried to read back some of the registers from the converter, by hardcoding them; this way we found which protocol version is working for which converter.

I have also tried disassembling an original firmware, which was interesting, but ultimately not really helpful.
Still I'm looking forward to your findings!

Cheers,
Flo

gekkehenkie11, 2022-03-25:
Hi @BeoQ ! Thanks for your reaction, nice to see more people showing up modding stuff! I have a shinewifi-X! About your comments, the baudrate on TX1 is 115,200 (you said 74880). So the bootloader outputs at 74880, then after that Growatt initializes the UART1 at 115,200 and they use that TX1 for debug output.

About the changing of baudrate, i SUSPECT that only happens because I put the stick in my computer instead of an inverter. I haven't tried it yet in my inverter, still dont have access to it. I suspect the stick just tries to communicate with the inverter and when it fails it just tries it at different baudrate (so on UART0 of course, on UART1 it just stays 115,200 throught the session)

BeoQ, 2022-03-25:
Hi Gekkehenkie,

The Baudrate for the Modbus depends on the Converter you have. If it was shipped with a ShinWiFi-X, then it should work at 115200 Baud. At least my MIC-600TL-X is working fine for a few months with this setting.

I makes sense, that the Stick tries different Baudrates to find a converter, because the converter is the Modbus-slave, it sits there and waits unitl someone talks to it an the right speed and the right address.

My suggestion of the process it this:
I suggest you first try to get the stick to be connected to your WiFi by editing the respective defines in ShineWiFi-S_ModBus.ino
Note that Wifi credentials must be surrounded in double quotes only (like: "MySecretPassword" ),
I also suggest to add a username and password for firmware update, because this simplifies later firmware uploads.

To hardcode a certain Baudrate, look in void Growatt::begin() in Growatt.cpp
For testing, I did modify the if statement, that it always runs along the else-path (for the ShineWiFi-X at 115200Bd).

Now recompile and upload; 
You're right, that stick does not make sense using at a PC, only for testing and uploading of firmware (connect GPIO0 to Gnd while plugging in to get the ESP to progamming mode).

To check if WiFi works, you just need to power the Stick via USB, using your PC is good enough for that. Verify that it connects to your Wifi-Network and gets an IP Address. If it does not, check your WiFi credentials and recompile/upload.

If it does, you can try this IP-address in your favourite Webbrowser using http://<IPaddress> (not https !).
Now you should see some basic status, and should be able to RW-Modbus to test if you can access some Registers and get sensible values in return.

If you don't, you can try other Modbus Registers, or different Baudrates. Remember to recompile an upload.

Good Luck,
Flo

gekkehenkie11, 2022-03-27:
Hi @BeoQ 
Well my goal is a bit different from most visitors here. I work for a solar installation firm, we use a lot Growatt, we like the brand, but the shinewifi-X is not very compatible, it doesn't like to communicate with certain routers. So my goal is to find out why that is and hopefully fix it. That's why I'm reverse engineering the original firmware. Made some very good progress already and am hopeful it will work out, but we'll see :) So I'm posting some of the info I find along the way here to hopefully help some others, although this might not be the best place to post it, I also don't know of a better place and actually I'm happy I found this page, helped me a lot for example @soulraven's instructions on how to flash were very welcome!

soulraven, 2022-03-28:
@gekkehenkie1 have you manage to find how the serial number is calculated? is interesting to use any ESP module to send data back to the growatt servers.

gekkehenkie11, 2022-04-02:
Hey @soulraven Not sure, haven't checked on that. Just finished up my project, succesfully patched Growatt's FW to work with some of the 'problematic' routers that some of our clients have. Was way more work than I had hoped, not sure if it was worth all that time, but it works anyway :) If you have any specific questions regarding reverse engineering the code just let me know, I know all about it now :)

soulraven, 2022-04-02:
Regading regading the protocol, Have you found some info how the data is send back to Growatt servers? I want to mock-up the sticks and send data directly from my server. Soon I will push a "framework" and some apps made for this framework. The intend is to get data from different sources, rs485, serial, proxy the sticks and upload to Growatt server, influxdb, MQTT. The framework is inspired by Django and HomeAssistent.

gekkehenkie11, 2022-04-02:
I only really worked on the wifi connection part, havent studied any other processes. Best way to start is just enable debug output, like explained above, buy an UART on amazon, solder ground and TX1 of the shinewifi to ground and RX of your uart and start logging output.
BTW I ran all of my tests just with the dongle in a PC (since I was just interested in the wifi connection part and that works without inverter), so I'm not sure if this output would be the same when connected to an inverter (I guess not) but here's the part where it talks to growatt.com:

https://pastebin.com/A962EVdz

gekkehenkie11, 2022-04-03:
@BeoQ I think the confusion is that you thought that my comments are regarding your modified FW :) I'm not using that FW (I tried it, loved it, but I've been doing a FW mod project for the original Growatt FW, for our clients). So all my comments in this thread only related to Growatt's original FW.

BeoQ, 2022-04-03:
Hi gekkehenkie11,

Well, this project is all about modifying the ShineWiFi's firmware, so that was just a first guess.

The ShineWifi-X is internally a somewhat ordinary (ESP8266-based) ESP07-Module, with a relatively good internal antenna. 
What makes it difficult to run the ShineWiFi-X's original firmware on some other ESP8266 based board, is that there is an additional EEPROM and a RTC with backup battery, that you would need to add to the circuit.

If you write that the ShineWiFi-X is a bit picky with certain routers, i suspect this is with the original firmware.

Normally these ESP8266-based devices perform somewhat well with a broad range of routers, so I would suspect the original firmware to be the cause for abnormal behaviour.

Are you forced to use Growatt's Protocol (and probably Growatt's servers)?

If yes, you're stuck to using the ShineWiFi-X with original firmware, or you might consider wiring all your converters with the RS485 communication and get just one of the RS485 communication gateway boxes from Growatt, serving multiple converters.

I haven't tried that box myself, I have only used the (wired-) communication of the converters to talk to it. It works with the identical Modbus-protocol, but on 9600Bd. However; i did not find how to change the ModbusAdress if you have multiple converters on the same RS485-Bus (I did not need to, so I didn't look)

If no (not stuck to Growatt's Servers), and you are happy with sending your data to a MQTT, then this project here may help you. Some improvement would likely help, to support a use-case like yours: 3-Phase and multiple PV-Strings support, improved auto-deteciton of protocol, generating a part of the mqtt-path from the converters serial-no, and so on.
Nothing extremely complicated to a seasoned programmer, but too much for my spare-time.

@soulraven, gekkehenkie,
There is another project on github named ["grott"](https://github.com/johanmeijer/grott) that focuses on the connection protocol between the ShineWifi-Sticks and Growatt's Server. They have implemented something to collect data sent from unmodified ShineWiFi-Sticks, and can even replace the Growatt' servers if you don't want your data there.

Good luck!

soulraven, 2022-04-03:
> Hi gekkehenkie11,
>
> Well, this project is all about modifying the ShineWiFi's firmware, so that was just a first guess.
>
> The ShineWifi-X is internally a somewhat ordinary (ESP8266-based) [...]

hello,
i know that project ["grott"](https://github.com/johanmeijer/grott), is a part of my implementation, hope is working, never tested yet, for the moment i am working on the framework and some implementation of [GrowattServer](https://github.com/indykoning/PyPi_GrowattServer) but in async mode. In a few weeks hope to try also the RS485 mode, and get data from that mode.

gekkehenkie11, 2022-04-03:
> so I would suspect the original firmware to be the cause for abnormal behaviour.

Yes, correct. More specifically it was a bug in the v2.0.0 version of the SDK Growatt used. I fixed it by writing some simple wifi STA code and compiled with several version of that SDK to find out which version solved it and then specifically, which patch. Luckily the fix was still in the v2.x version because the v3.x version of the SDK is quite different. So merging it with the current Growatt FW code was not too hard

ZAPotter, 2022-04-21:
> thx, is working, I manage to flash the ESP, the problem is now that is from WIFI-X not WIFI-F. and I am stuck with the config for WI-FI network. The procedure is hard or bad translate on Shine app. [...]

Good day

My Wifi-F also bricked during an update (all 3 LED's stay on all the time).

I downloaded the Wifi-F v3.0.0.0 firmware. But unable to flash it. GPIO0 to ground for start up (tried keeping it connected as well) RX-> TX and TX -> RX (I confirmed the print on the board is right). but no luck. I tried esptool.py and FlashESP8266.exe (I use this to flash my Sonoff's).

Looks like the green LED is connected to GPIO0.

Thanks

sohnvonoff, 2022-04-22:
Hi ZAPotter,
Urgent!
Install the driver for UART (XR21V14xx) to flash direct over USB. (without solder) Only jumper between GPIO00 and GND.
[https://www.maxlinear.com/support/design-tools/software-drivers](https://github.com/otti/Growatt_ShineWiFi-S/issues/url)
The default windows driver don't work with XR21V14xx (UART in ShineWifi-x)

ZAPotter, 2022-04-22:
Thank you.

I tried that, but it didn't even detect a COM port. That was a very old one, had firmware 1.7.6.6 on it. Took it back and they exchanged it for me. The new one has firmware 3.0.0.0 and is working great.

Solpowersun, 2022-08-23:
> Tengo volcado de un dispositivo WIFI nuevo. El stick es Growatt Shine WIFI-F (caso de mierda) La versión o firmware es 3.0.0.0.
> 
> https://github.com/soulraven/growatt_esp_monitor/raw/main/flash_ORIGINAL_FWv_3.0.0.0_HWv_1.7.6.0.bin

What is the way if i want to do backup original firmware in my shinewifi-x before to update otti firmware mod?

Thanks

hugocrank, 2022-10-26:
after a broken Update on one of my wifi-x (Growatt support has crashed the stick) i upload the firmware from here and what should i say, its runining again! thank you!
very nice is to so that the modified Update interval i change bevor the crash to 1 Minute is still active!
in the firmware version 3.1.0.0 is a webinterface included to change the intervall between 1-5 minute. after reanimation with firmware 3.1.0.2 and the crashed newest 3.1.0.5 the solution / webinterface is down in the firmware .-( is there still a workaround known to change the time in a newer firmware? and crazy is that the stick have the 1 min. interval after firmware flashing...

crasu, 2022-12-02:
Closing this issue as it is a bit off topic

closed 2022-12-02 by crasu