OpenInverterGateway/OpenInverterGateway #6: Undef stick in debug log / watchdog

Cruzxia, 2022-02-02:
I am using the Wifi-X module & TL-X inverter
The Debug reports:
#0: Setup()
#1: WiFi reconnected
**#2: Error: Undef. Stick**
It all functions correctly collecting data. 

With regard to the watchdog, I could not see any implementation in the code. 
I have had one system crash twice on me. (no mqtt data or access via the htpp) 
I did notice that the green LED hat stopped flashing and was just constantly on. 
so I am uncertain if it was a Wi-Fi disconnect or if the processor hung up. 
I unplugged and plugged the USB back in to reboot, and all was working again. 
In theory the watch dog should have rebooted the module. 
Is it possible to add a watchdog to your code. 

Thanks
Cruzxia

otti, 2022-02-03:
Hi Cruzxia,

are you using the latest version? I have fixed a probleme where the stick could hung up every morning after the inverter powers on. It seems, that the inverter powers the stick but is not ready to talk to it just after sunrise. That is the reason why the debug reports the "Undef. Stick".

I have not implemented a watchdog yet. But feel free to add some code. 

Regards,
Timo

Cruzxia, 2022-02-07:
Hi Timo
Yes I am using the latest version, It powers up ok in the morning. it was late in the afternoon that it stopped. 
I have since re-flashed that unit and it seems to be stable now. 

I will take a look at the watch dog, best I can tell the hardware and software watchdog is on by default on the 8266
It may be that it was stuck in a loop that had a delay and kept clearing the watchdog. On my esp32 projects I have used an interrupt timer for the watchdog function, and clear the timer at the end of the main loop, that way if it gets stuck in a library loop the system will reboot after about 120 seconds.  (needs to be long enough for OTA) 

Lastly I noticed that there time zone in the graphing html. So depending on your location the time on the graph wont be correct.  

Regards
Cruzxia

closed 2022-12-04 by crasu