OpenInverterGateway/OpenInverterGateway #10: Please excuse me but I dont understand what to do exactly

hucky58, 2022-03-11:
I own a Growatt TLX Inverter with ShineX stick. Growatt stopped the possibility to read most of the submitted values.
So i think i have to flash the stick with new firmware and then i am able to read the stick by myself. But i dont find this imagefile that i have to flash. I am wrong here? I only find some source codes to make the reading program.

gekkehenkie11, 2022-03-13:
You have to first edit in the values you want to use into the sourcecode (like your wifi password, wifi SSID, the ip of your server you want to use) and then build the image with the Arduino IDE. It will give you a few errors, for example you will need the right board manager URL (the .json link that is also in the first lines of the source code) in the preferences part of the IDE, then you need to select the ESP8622 board in the tools settings, you need to add a few libraries that it misses and edit a header of one library to increase one setting from 256 to 512. It will show you the error, you can google it and should be easy to solve.

I haven't flashed the image myself yet, no time yet, but I think it should be either possible directly with the Arduino IDE or else with esptools.py, like explained elsewhere in this 'issues' section of this site.

Don't forget to put the device into flash mode by shorting (jumpering) GPIO0 and GND, also explained on another thread in the 'issues' section, and install the right windows driver.

gekkehenkie11, 2022-03-13:
Actually I flashed it myself today, with the esptool.py that was given in one of the other 'issue' discussions here. Works perfectly, really awesome stuff!

EmHenkey, 2022-08-14:
Hello, 
I tried to compile the project but I am not really a master in using Arduino SDK. I have issues in adding missing libraries. Can anyone give me a short howto or a quick guide what libraries are necessairy?
Thanks in advance.
Martin

crasu, 2022-12-02:
@EmHenkey  May be you want to use [platform](https://platformio.org/install) io? This will download the required libs itself.

closed 2022-12-02 by crasu