OpenInverterGateway/OpenInverterGateway #11: Results on a generic ESP8266MOD (Generic Lolin) ESp12

the-butterfry, 2022-03-16:
Seems to work fine. I'm not on location of the inverter though, but will test when I am.

I'll be testing on a SPF 3000TL LVM-24 Do we need to add the decoding/offset for that? I think [this](https://github.com/johanmeijer/grott) project has that sorted out. Or is it magic(??) It sure would be nice to have a more better written firmware than the ShineWifi modules. I dread re-setting those things up, 1/10 chance that it actually works whenever I have to mess with them.

BeoQ, 2022-03-25:
Hello Butterfry,

There is a significant difference how these two projects work:

- Grott uses the standard Firmware on the Wifi-Sticks, and intercepts the communication to the Servers from Growatt. I.e. it talks the protocol they send across the internet, and is limited to whatever the Stick reads from the converter and sends to their servers. Obviously this is ultimately based on data read out from the converter.
- Our project replaces the Firmware on on the WiFi-Stick, and communicates directly to the converter using its Modbus-Interface. This allows to read out whatever data you want, if you know in which register it is located, and modify the code appropriately.
- I do think it is worthwhile to share knowledge. Using one or the other depends if you want to keep sharing your data to Growatt or not. 
If you don't want/need to share the data and you're willing to flash your WiFi-stick, there is little point on using Grott.
If you don't want/not allowed to flash your WiFiStick, then it is better to use Grott.

I did skim through the available protocol documentation:
- The older inverters apparently use Protocol v3.05 from 2013.
- The newer inverters apparently use protocol v1.05 from 2018. 
- The newer protocol allows for additional inverter functionality, which is not implemented here (yet).
I don't know (did not search) if these Island-Inverter use a different protocol. YMMV.

Since the converter you mention attaches Batteries, i guess it uses the newer protocol.
This has provisions to read out al lot of Data related to the battery.
We currently read out only some data for one mppt-string and for one AC-phase, is should be somewhat easy to extend that.

To read it yourself, search the net for: "Modbus RTU Protocol v1.05 2018.pdf"
The Section about Holding Registers shows the somewhat static information.
The Section about Input Regsters show the dynamic data, that you probably want.
Note that you need to use ReadHoldingRegisters() or ReadInputtRegisters() respectively.

FYI: I have tried to use a Wemos-D1 to run our Firmware, attached using it's USB-cable: Works flawlessly, you just don't have the fancy Status-Leds.

Cheers,
Flo

the-butterfry, 2022-03-25:
Yup, I got excited when I saw what you're doing here and just had the "smash-these-projects-together" thought. 

As for battery info - It only works if your BMS speaks Growatt via the RS485. Personally I'm harvesting my battery info directly from the BLE in the BMS. But I'm sure we would want to have that available as many aftermarket batteries now speak Growatt. Grott already has all that available anyhow, so there isn't anything to figure out there. I just have DiY batteries, that don't speak that language for feedback to the Growatt unit. But future builds will, to be sure.

As for the viability of a smashed project, It might be best attacked from the other side since Grott is holding all the register/offset/decode information and other features which might just be a matter of putting that in a more ESPable format.  A combined code might also be a candidate for an ESP32 variant anyhow. 

The goal for me honestly is a simplification of the system. As we know the Growatt firmware is lacking and frustrating to use and reconfigure. I'd love it if I could do exactly what your project is doing - Direct to MQTT or even Influx socket of some sort without bouncing it off of a secondary computer (raspberry). I'm running an MQTT Server on the onsite router which holds the info for an offsite TIG stack - so that would totally eliminate the complexity of the raspberry onsite.

Thanks for taking a look at my excitement. I'm not the person to smash this together, but I'll bring it up to @[johanmeijer](https://github.com/johanmeijer)

closed 2022-12-04 by crasu