OpenInverterGateway/OpenInverterGateway #13: What kind of USB-to-serial cable to use for flashing?

Jordi24, 2022-03-26:
After having had quite some issues with the Growatt servers and my ShineWiFi-S, I came across this solution to run everything locally and it looks very promising. Although I have experience with DIY projects, having no USB connector directly on the board is a first. Therefore, I was wondering what kind of USB-to-serial cable you would advise to use? Are there any specific things that I should consider?

Jordi24, 2022-04-03:
@otti, @StevenBoelhouwer, @gekkehenkie11, I saw that you were so kind to answer questions from others in the past. Maybe you could help me as well?

otti, 2022-04-03:
Hi Jordi24,
I would recommend an USB to serial converter with a chipset from FTDI. Drivers for FTDI chips are usually included in nearly every operating system. So they will work out of the box.
The WiFi stick operates at 3.3 V. So I would recommend a version where you can switch between 3.3 and 5V.

This is the converter I have used:
https://www.ebay.de/itm/164880183161

BR,
Otti

gekkehenkie11, 2022-04-03:
I used this one myself: https://www.amazon.com/gp/product/B07D6LLX19/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Btw if any of you wants my IDB database, just let me know. I used IDA v7.0 so I guess it will only load with that version. I managed to label a lot of functions, by generating FLIRT signatures with some demo code generated with ESP8266_RTOS_SDK v2.0.0 (that's the one that growatt used). 

Like I said in the thread, my goal was to fix their buggy wifi implementation, causing the dongle not to work with some routers. I quickly found out that the problem was not in the growatt user code, but in that v2.0.0 version of the SDK. So what I did is that I compiled some basic wifi STA connection code with different versions of the SDK to see which one did work and which ones didn't. That way I could determine the exact patch in the SDK that fixed it. Then the last step was of course to merge that patch into the exisiting growatt image. I was lucky that the fix was still in v2.x, because the v3.x framework is quite different and would have been much harder to merge with the growatt image.

Regards,
Henkie

gekkehenkie11, 2022-04-03:
BTW, not sure if you're aware but if you have the USB version you don't need to buy an UART at all. Just stick the wifi dongle into your PC, short GPIO0 and GND (this will put the firmware into a FW dowload mode), install the correct windows driver for the UART(very important! Also explained in another thread here) and then you can just flash it.

I only used the UART for getting debug output from Growatt (since my goal was not to flash a completely different FW, but make a mod for the exisiting Growatt firmware)

Jordi24, 2022-04-03:
@otti and @gekkehenkie11, thank you for your answers and suggestions. I will have a look at the USB to serial converters you used and see which one is available. Thanks again!

EvilForge, 2022-07-19:
https://diysolarforum.com/threads/hacking-the-new-growatt-wifi-f-modules.43231/#post-548996

I have the newer ShineWifi-F and I had to cut the traces to the SP3232EE chip, and the pull-up resistors, before I could get a USB-Serial cable to work.  I just tried to plug it in and see if a COM port would show - but it doesn't appear to be the case with this version.  I'm also trying to get data from a SPF 3000TL LVM-ES, but so far, the only valid data i see to get is temperature. 

Nevertheless... yes USB-Serial will work but you have to remove the interfering serial chip and pullups - then once you have it flashed, you can use OTA remotely.

closed 2022-12-04 by crasu