OpenInverterGateway/OpenInverterGateway #14: WIFI-F - Can't get into boot mode

JohnGalt1717, 2022-04-07:
I can't for the life of me get this into boot mode. I've done this dozens of stuff with no issues but no joy here.

I have v3.3, gnd, rx to rx and tx to tx on the usb uart that I always use and then short gpio0 and gnd using the pins that I continuity tested on the board and it looks like it's in boot mode (one red light), but no joy uploading.

Is there something I'm missing?

otti, 2022-04-08:
Hi John,

Cabling looks OK to me.
A power cycle is needed while GPIO0 is tied to GND. Did you do that?
What kind of Growatt stick do you use?

gekkehenkie11, 2022-04-10:
Not sure about the wifi-F, but on the Wifi-X you should connect the UART normal way so RX to TX and TX to RX.

JohnGalt1717, 2022-04-11:
ShineWifi-F

Power cycled with the battery out and GPIO0 to ground. Tried TX-TX and TX-RX and neither respond.  The red light and only the red light is on, which I would assume was firmware mode, but still no response.

mstieg, 2022-04-24:
I am also about to try this but need to ask, otti, is your sketch likely to work on the wifi-f stick?

JohnGalt1717, 2022-04-25:
My solution was to desolder the ESP07 and replace it with an ESP12F which works just fine and is pin compatible.

mstieg, 2022-04-25:
James,

Thanks for the response. Where did you source the ESP12F?

Since the 07 is surface mounted, a like module would be needed, correct?

-Mark

JohnGalt1717, 2022-04-25:
Just went on Amazon and get an esp-12f in search and you’ll find it. 

The 12f has additional pins in the bottom but otherwise is identical for your purposes and it works perfectly. (Also works using esphome that I believe uses this code). 

There’s also an esp32-c3-12f which is also the same pinout that works too.

closed 2022-12-04 by crasu