OpenInverterGateway/OpenInverterGateway #17: Shinewifi-X wifi working, but no webserver found

jhaake2308, 2022-06-10:
Hi, after reading a lot about the project I managed today to flash my shinewifi-x after bringing it into boot-mode by connecting GPIO0 to GND when inserting the stick (not using seperate cabling but the built-in usb3). The compilation of the sketch is running w/o errors, I managed to work out these by changing the MQTT-supported-packet-size and adding the necessary packets.

I installed the drivers from [here](https://www.maxlinear.com/support/design-tools/software-drivers) as mentioned in the wiki, its the same behavior with driver version 2.6.0.0 and 2.2.5.0 (I am on Win10 64bit here). 

The process of uploading the sketch seems to be successfull, after ejecting it from usb and reinserting it I almost instantly get a ping-answer from the IP-Address of the stick. I also can telnet to the IP on port 80 (telnet 192.168.x.x 80) and seem to have a connection, but in browser I am not getting a response. 

[image]

[image]

Although there was enough sun and energy (>1kwh) I did not reinsert the stick into the inverter until now, but configured the 
#define SIMULATE_INVERTER to 1 to eliminate this as a root cause. 

When inserting the stick now, the red LED is flashing for 1-2 seconds and then changing to permanent red. Ah and one more curious thing, when connecting GPIO0 to GND, not only the RED led is on (labeled as LED2), but also blue (LED3) 

Any ideas? Forgive me my long text, I am stuck here and have no idea what to try next :/ 

Thanks and BR 
Julius

PS Pictures of my shinewifi-X:

[image]

[image]

JonEllis1234, 2022-06-11:
Hi. Just flashed my wifi stick for first time as well, but managed to recreate your problem because my MQTT server requires authentication, and so the line 
   if (MqttClient.connect("Growatt", MQTT_TOPIC, 1, 1, "{\"Status\": \"Disconnected\" }"))
needed to be replaced with
   if (MqttClient.connect("Growatt", "mqttusernm", "mqttpassword", MQTT_TOPIC, 1, 1, "{\"Status\": \"Disconnected\" }"))
At which point once compiled the web server springs into life. (obvious mqttusernm and mqttpassword would be whatever your user/pass are set to in MQTT. Hope that's the same as your issue and it fixes it. Please say if it does. Obviously I would suggest that long term a variable should be used and put in a nicer place in the file.

jhaake2308, 2022-06-12:
hi @JonEllis1234 
you kicked me into the right direction. Until today I didnt realize, that a running mqtt-server is necessary for this to run. I had in mind to just grep the necessary information by the built-in json-frontend. Fortunately my whole setup @ home is built up in node-red so it was not a real task to get a mqtt-server running. And see, the website is coming right up after mqttserver says "one client connected". Thanks for your help!

BTW, I first tried to get mqtt out of the way by setting MQTT_SUPPORTED to 0, but it seems this variable is not honored, as the sketch is throwing variuos errors concerning the mqtt-packetsize and so on. Commenting out was also not really successful, so setting up the mqtt-server was the better way :)

Thx again, smooth thing here
Julius

closed 2022-06-14 by jhaake2308