OpenInverterGateway/OpenInverterGateway #19: How to use postCommunicationModbus?

hansaplasst, 2022-06-29:
Hi,

I'm trying to figure out the correct registers for PV1 and PV2 Power since I want to use these values in my ESPHome device. I thought using the [RW Modbus] could tell me which registers to use. But I can't figure out how this works?

Can anyone help me with that?
Br. Hansa

superfunnl, 2022-06-30:
Depending on your RTU Protocol (which i believe is tied to your growatt inverter model and/or serial/usb stick), you can find it in the specification:
(start at page 25) v1.0.5 (newest version i believe): https://www.photovoltaicsolar.in/Growatt_Manual/MAX%20Series%20Modbus%20RTU%20Protocol.pdf
(start at page 10) v3.0.5 (older version i believe): https://docplayer.net/161194234-Growatt-pv-inverter-modbus-rs485-rtu-protocol.html

I believe the input registers do get read if i look at Growatt.cpp line 22 and 32.
Line 22 (input register 28) matches on the v3.0.5 specification for total energy.
Line 32 (input register 55) matches on the v1.05 specification for total energy.

I did found other projects on github (e.g. https://github.com/nygma2004/growatt2mqtt/blob/main/Growatt2mqtt_1p2s.ino start on line 135) which matches the input registers on the v3.0.5 specification.

hansaplasst, 2022-06-30:
Lots of Thanks!!

The [doc](https://forum.iobroker.net/assets/uploads/files/1550402978050-growatt-pv-inverter-modbus-rs485-rtu-protocol-v3.05-%E5%AE%A2%E6%88%B7%E7%89%88.pdf) I found has the same specification. I also tried out the regs you proposed this afternoon. IIRC, Reg 5 gave me the right value for PV1. Reg 9 however was 0.

I Also tried out 6 and 10, which also gave unexpected results. Maybe I programmed it wrong? I'm not a programmer.

Growatt.cpp (ln 70)
`...
    if (res == Modbus.ku8MBSuccess)
    {
      _Data.InverterStatus     = (eGrowattStatus_t)Modbus.getResponseBuffer(0);
      _Data.u32DcPower         = (Modbus.getResponseBuffer(5) << 16) + Modbus.getResponseBuffer(9);
      _Data.u16DcPowerPV1      =  Modbus.getResponseBuffer(5);
      _Data.u16DcPowerPV2      =  Modbus.getResponseBuffer(9);
      _Data.u16DcVoltage       =  Modbus.getResponseBuffer(3);
      _Data.u16DcInputCurrent  =  Modbus.getResponseBuffer(4);
   
      //return true;
    }...`

Growatt.h (ln 24)
`...
    float            GetDcPower();
    float            GetDcPowerPV1();
    float            GetDcPowerPV2();
    float            GetDcVoltage();
...`

Index.h (ln 84)
`...
 	  var obj = JSON.parse(this.responseText);
	  document.getElementById("AcPower").innerHTML = "Power: " + obj.AcPower + " W"; 
	  document.getElementById("EnergyToday").innerHTML = "Today: " + obj.EnergyToday + " kWh"; 
	  document.getElementById("EnergyTotal").innerHTML = "Total: " + obj.EnergyTotal + " kWh"; 
    document.getElementById("DcVoltage").innerHTML = "DC Voltage: " + obj.DcVoltage + " V"; 
    document.getElementById("DcPower").innerHTML = "DC Power Total: " + obj.DcPower + " W"; 
    document.getElementById("DcPowerPV1").innerHTML = "DC Power String 1: " + obj.DcPowerPV1 + " W"; 
    document.getElementById("DcPowerPV2").innerHTML = "DC Power String 2: " + obj.DcPowerPV2 + " W"; 
...`

ShineWiFi-S_ModBus.ino (ln 521)
`...
  sprintf(Buffer, "%s  \"DcVoltage\": %.1f,\r\n",       Buffer, Inverter.GetDcVoltage());
  sprintf(Buffer, "%s  \"DcPower\": %.1f,\r\n",         Buffer, Inverter.GetDcPower());
  sprintf(Buffer, "%s  \"DcPowerPV1\": %.1f,\r\n",      Buffer, Inverter.GetDcPowerPV1());
  sprintf(Buffer, "%s  \"DcPowerPV2\": %.1f,\r\n",      Buffer, Inverter.GetDcPowerPV2());
`

What am I missing?
Is it the "shift left +" thingy? Has something to do with u16 to u32 conversion but I don't understand it fully. Not a programmer here.

superfunnl, 2022-07-01:
Not a programmer as well. Do you have an ShineWifi-X or ShineWifi-S?

I think for the ShineWifi-X you need the following:
`res = Modbus.readInputRegisters(0, 9);` line 70 Growatt.cpp

hansaplasst, 2022-07-02:
Thanks. Seems obvious to read from 0 to 9

This is what I've tested so far.
[image]
It's getting dark now. So maybe tomorrow I'll tweak a little more.

BTW I've found working solution [here](https://community.home-assistant.io/t/esphome-modbus-growatt-shinewifi-s/369171/10) using ESPHome. I've modified this code a bit using reg 6 and 10. These gave me the correct power output. of PV1 and PV2. So we're definitely missing something with the above code...

superfunnl, 2022-07-10:
I am waiting on my wemos d1 so i can try it my self on the SYSCOM port. Currently the ShineWifi-X is working as I intended with domoticz so I wont mess around with it.

I did even found an 1.20 version PDF which has an date of 2022 and mentions the TL-XE series (newest inverter series is believe, at least the one I have).
https://github.com/nygma2004/growatt2mqtt/blob/main/Growatt%20PV%20Inverter%20Modbus%20RS485%20RTU%20Protocol%20v120.pdf

hansaplasst, 2022-07-10:
Thanks! 
The v1.20 also mentions that PV1 and PV2 power should be read from the registers 5/6 for PV1 and 9/10 for PV2 string. Therefore I would assume that below code should work to get the proper values from both strings.

```
res = Modbus.readInputRegisters(0, 10); // line 70 Growatt.cpp
```

and adding around line 74

```
      _Data.u32DcPowerPV1         = (Modbus.getResponseBuffer(5) << 16) + Modbus.getResponseBuffer(6);
      _Data.u32DcPowerPV2         = (Modbus.getResponseBuffer(9) << 16) + Modbus.getResponseBuffer(10);
```

I haven't tested this since the [ESPHome code](https://community.home-assistant.io/t/esphome-modbus-growatt-shinewifi-s/369171/10) I am currently using gives me the proper output. 

Yes please let us know this works via the SYSCOM port!

superfunnl, 2022-07-19:
Small update on my end, i got the SYSCOM port working, but it wouldnt read any of the PV1 and PV2. It did manage to retreive the other registers. I got it now running with own-customized code of https://github.com/nygma2004/growatt2mqtt.

Next up, compare code why it doesnt work with this repo but it does with the other one.

hansaplasst, 2022-07-20:
Nice one!
So you're saying you can read PV2 DC-Power from SYSCOM with this repo code? Do you have the ability to check your repo with the ShineWifi-X or ShineWifi-S? Because I couldn't get it to work.

superfunnl, 2022-07-26:
I ended up using the other repo and haven't checked this repo any further for the issues we are facing. (still not a programmer ;) )
[image]

hansaplasst, 2022-07-30:
No problem @superfunnl ;-)
Glad you got it working!

crasu, 2022-12-04:
After #61 you should now be able to get invert PV Power values (1-7).

closed 2022-12-04 by crasu