OpenInverterGateway/OpenInverterGateway #20: ESP8266 - NodeMCU

Holzzer, 2022-07-24:
Hallo Otti, ich wollte mich bedanken für das gute Programm. Ich habe gerade mit einem NodeMCU-Board und über den normalen USB-Port die Verbndung zum Growatt MIC 1000 TL-X  aufgebaut. Zuvor hatte ich sehr viel Zeit mit der separaten RS485 Schnittstelle erfolglos verbracht.
Beim Kompilieren deines Programms müsste ich zwei Änderungen vornehmen (siehe Bilder). Vielleicht kannst du ja diese Erfahrungen mit einfliessen lassen.
Nochmal vielen Dank,

Holzzer
[image]
[image]

[image]
[image]
[image]
[image]
r

otti, 2022-07-25:
Moin Holzzer,

danke für dein Lob und deine Anmerkungen.
Mich wundert etwas warum das überhaupt funktioniert. Verwendet der NodeMcu genau den gleichen seriell nach USB Konverter wie der original Stick? Ich kann mir nicht vorstellen, dass der Growatt Inverter pauschal alle seriell nach USB Konverter unterstützt. Aber ist ja gut, dass es funktioniert ;-)
Kannst du zu deinen Screenshots noch etwas mehr schreiben?
Screenshot 3: Hat es ohne das #include nicht funktioniert? 
Scrrenshot 4: Existiert das #define MQTT_MAX_PACKET_SIZE bei dir nicht, oder warum hast du es durch eine Zahl ersetzt?

Gruß,
Otti

BeoQ, 2022-07-30:
Hallo beinand,

Ich denke, daß auf den Growatt's irgendetwas Linux-artiges läuft, und deshalb verschiedene USB-Serial Converter unterstützt werden.
Mein Original ShineWiFi-X hat einen Chip von Exar (siehe Hardware-Doku).
Eine Weile hatte ich einen Wemos-D1-Clone mit CH340 Chip am laufen (der ist auf den neueren NodeMCUs auch drauf, glaub ich jedenfalls), 
Mein alter NodeMCU mit dem Silabs-Chip hatte auch funktioniert. 
Nur einen FT232 hatte ich noch nicht dran. ;-)

In meinem Fork hatte ich angefangen da was aufzuschreiben, aber noch lange nicht pull-request-tauglich.
Momentan hab ich leider grad nicht so die Zeit damit rumzuspielen oder an der Firmware zu basteln.

Happy hacking!
   Flo

Holzzer, 2022-07-31:
Hallo Otti,

hier mal ein Update, der NodeMcu lief die ganze Woche ohne Probleme durch.

  1.  Der Webserver "ESP8266WebServer httpServer(80);"  konnte nicht gestartet werden, deshalb habe ich die Zeile hinzugefügt.
  2.  "#include esp8266webserver.h"

  1.  genau, die Variable "MQTT_MAX_PACKET_SIZE " war bei mir nicht definiert. So habe ich die schnell die Anzahl fixiert.

Gruss,

Holzzer

Bernd9000, 2022-08-02:
Auch von mir Danke an Otti, klappt super.
Würde gern an meinem Growatt MIC 600 TL-X die Ausgangsleistung einstellen.
Weiß jemand ob das hiermit machbar ist ?

Bernd9000, 2022-08-02:
Hab mal die neuste Version "2022-07-31 Update" installiert.
Wifi Verbindung klappt, Anmeldung am MQTT Broker auch.
Es wird kurz der Status im MQTT Datenpunkt angezeigt und paar Sekunden später ist der Status auf "Disconnected".
Auf der Web-Seite erscheint dann gleichzeitig wieder die Wifi Config Seite, Wifi Verbindung ist aber noch vorhanden.
Hat das noch jemand ?

BeoQ, 2022-08-02:
Hallo Bernd9000,

ja, Leistung drosseln geht, auch übers Webinterface vom WifiStick.
Leider nur in Prozent der Nominalleistung aber nicht darüber hinaus.
In der Spec v1.05 von 2018 gibt es hier zwei Holding-Register:

- Register 03 "Active P Rate" - "Inverter Max output active power" in percent 0-100 <--- Den brauchst du
- Register 04 "Reactive P Rate" - "Inverter Max output reactive power" in percent 0-100

In der Spec steht zwar, daß man auch 255 für Maximalleistung schreiben kann, aber das führte bei mir zu einer Modbus-Fehlermeldung.

Leider sind die Register 06 & 07 "Pmax" "Normal Power" nicht beschreibbar.

Interessant ist das Setzen der Max Ausgangsleitung, wenn man den Umrichter als Batterie-Inverter mit Eigenverbrauchsoptimierung einsetzen möchte. 
Im Prinzip möglich, Spannungslage (am PV-Eingang) bei der dieser Startet liegt bei ca 50..55V.

Happy throttling,
  Flo

Bernd9000, 2022-08-02:
Okay, Danke, probier ich Morgen aus.
Und das mit dem Fehler der neuen Firmware am ESP hab ich glaube ich schon gefunden.
Die neue Firmware denkt ich hätte den Taster gedrückt `(adcValue< 50)`  und löst damit den eigenen Access Point aus.
Da ich nicht weiß wie der Taster im Orginal verdrahtet ist hab ich in mit einem 1K Widerstand nach +3,3V gezogen.
Das werde ich dann Morgen auch testen

Bernd9000, 2022-08-03:
Hab mal getestet mit Holding Register 03, funktioniert nicht.
Wenn ich den Wert lese kommen verschindene Werte zwischen 900 und über 1000.
Versuch ich den zu beschreiben kommt :Writeing Input register is not allowed

Bernd9000, 2022-08-03:
@otti 
Hab den Fehler gefunden.
Bitte korrigieren falls ich falsch liege.

Zeile 472:
if (httpServer.arg("rd") == "Holding") // Read/Write Holding Registers

Falsch, die Variable "rd" kann nie den Wert "Holding" annehmen.
"Holding" wird in die Variable "holding" geschrieben.

Richtig:
 if (httpServer.arg("holding") == "Holding") // Read/Write Holding Registers

Und dann funktioniert es auch mit der Leistungssteuerung :-)

otti, 2022-08-03:
Hi @Bernd9000,

ich habe es geändert. Danke für den Hinweis.

achim-70, 2022-08-14:
Hallo @Bernd9000,
Du hast ja wohl auch einen Wemos D1 am Growatt MIC 600 TL-X im Einsatz. Mit der Kombination würde ich auch gerne arbeiten, will aber noch nicht so wirklich klappen. Die Software ist drauf, WLAN und MQTT eingerichtet und die Oberfläche kann ich auch aufrufen. Nur kommen dort wohl keine Daten an. In der Grafik wandern die Uhrzeiten immer schön brav nach links weg, aber Werte fehlen und auch in der Liste darunter nur "undefined" (bei Power, Today usw.). 
Unter "Log" wird mir immer nur "Error: Unknown Shine Stick" angezeigt und auf den Seite "Json" sehe ich nur "Status: Disconnected". Muss ich denn außer in der Config.h in einer der anderen Dateien noch Einträge vornehmen, wenn ich den Wemos zum Laufen bringen will? Den Vorschlag von Holzzer weiter oben bzgl. JasonStrin[512]... hab ich auch ausprobiert, hat aber nichts gebracht.
Hier mal noch meine Werte aus der Config.h:
#define MQTT_SUPPORTED 1
#define PINGER_SUPPORTED 0
#define ENABLE_DEBUG_OUTPUT 1 (auch schon mit 0 probiert)
#define ENABLE_WEB_DEBUG 1
#define SIMULATE_INVERTER 0
#define AP_BUTTON_PRESSED ( digitalRead(1) == HIGH )
Die Variablen HOSTNAME, APPassword, UPDATE_USER und UPDATE_PASSWORD hab ich noch geändert, die Werte dürfte hier aber wohl nicht relevant sein.

Bernd9000, 2022-08-14:
@achim-70
Ich hab den wemos D1 Mini v3, weiß nicht wie es bei anderen Versionen aussieht.
Zu Anfang klappte es bei mir nicht weil ich den wemos nackt, also ohne Taster und LED angeschlossen hatte.
Das führte dazu das ständig neu gebootet wurde. Da bin ich erst hintergekommen als ich gesehen hatte das ständig der Access-Point vom wemos aufgespannt wurde. Und dachte ich mir schon das es am Tastereingang liegen könnte.
Da ich zu faul war ein Taster anzulöten hab ich in der config.h folgende Zeile geändert:
#define AP_BUTTON_PRESSED ( digitalRead(D7) == LOW )
Den Digitaleingang D7 habe ich benutzt weil dieser der einzige war der beim booten stabil war, also nicht wie andere Eingänge hin und her sprang von High auf Low. 
Und dann funktionierte es endlich, andere Daten in der congfig.h habe ich nicht geändert.
Also schau mal ob der Access Point vom wemos im WLAN auftaucht.
Klappt es denn mit MQTT ? Wird ein Datenpunkt im MQTT Broker angelegt ?

Bernd9000, 2022-08-14:
Ach ja, den Analog Eingang habe ich natürlich deaktiviert:
//#define AP_BUTTON_PRESSED ( analogRead(A0) < 50 )

achim-70, 2022-08-14:
> Ach ja, den Analog Eingang habe ich natürlich deaktiviert: //#define AP_BUTTON_PRESSED ( analogRead(A0) < 50 )

Hab ich ja auch gemacht, weil ich vorher ebenfalls ständig der AP Modus hatte. Ich hab dann einfach den anderen Eintrag aktiviert: #define AP_BUTTON_PRESSED ( digitalRead(1) == HIGH ), womit das Problem mit dem ständigen AP Modus behoben war. Ich hab jetzt auch mal Deine Einstellung genommen, aber da ist dann wieder der AP Modus ständig da gewesen. Ich muss allerdings auch gestehen, dass ich nicht explizit den Wemos D1 mini gekauft habe, sondern - ich sage mal - einen Klon ([XTVTX](https://www.amazon.de/gp/product/B093GR7MJM/ref=ppx_yo_dt_b_asin_title_o00_s01?ie=UTF8&psc=1)). Da in der Beschreibung steht "100% kompatibel mit Arduino und WeMos", bin ich mal davon ausgegangen, dass es keine Probleme bereitet.
Hast Du eigentlich beim Flashen als Board "Generic ESP8266 Module" verwendet, wie es in der Beschreibung steht oder ein anderes gewählt (z.B. LOLIN(WEMOS) D1 R2 & mini)?

Bernd9000, 2022-08-15:
Ich hab das genommen:
[image]

achim-70, 2022-08-15:
Guten Morgen,
ich konnte gestern Abend nicht mehr weiter forschen, war dann ja dunkel...
Also mit "NodeMCU 1.0 (ESP-12E Module)" bin ich auch nicht weiter gekommen, auch nicht mit dem "NodeMCU 0.9 (ESP-12 Module)". Ich will mal noch ein paar andere ausprobieren. Mir ist aufgefallen, dass mein Board ja 12F hat, aber ich würde da mal eine Abwärtskompatibilität unterstellen.

Zwischendurch mal noch eine andere Frage: Hast Du in Deinem Log auch den Eintrag "Error: Unknown Shine Stick"? Der ist mir von Anfang an aufgefallen und ich dachte, der kommt, weil ich eben nicht den Original-Stick verwende.

Bernd9000, 2022-08-15:
Zu Anfang, wenn die Modbus Verbindung noch nicht geklappt hat steht bei mir auch "Error: Unknown Shine Stick"
Erst wenn die Verbindung läuft, steht bei mir "ShineWiFi-X (USB) found".
Mal ein Beispiel vom Debug Log, wie man sieht klappt auch das Daten Update nicht immer zuverlässig.

#0: Setup()
#1: ShineWiFi-X (USB) found
#2: UpdateData() NOT successful
#3: UpdateData() successful
#4: UpdateData() NOT successful
#5: UpdateData() successful
#6: UpdateData() NOT successful
#7: UpdateData() successful
#8: UpdateData() successful
#9: UpdateData() successful
#10: UpdateData() NOT successful
#11: UpdateData() successful
#12: UpdateData() successful
#13: UpdateData() NOT successful
#14: UpdateData() successful
#15: UpdateData() NOT successful
#16: UpdateData() successful
#17: UpdateData() successful
#18: UpdateData() successful
#19: UpdateData() successful
#20: UpdateData() successful
#21: UpdateData() successful
#22: UpdateData() successful
#23: UpdateData() successful

Bernd9000, 2022-08-15:
Sonst würde ich an deiner Stelle noch weitere "WEB_DEBUG_PRINT" im Quellcode einbinden um den Fehler einzukreisen.

achim-70, 2022-08-15:
Ok, ich schau mal, wo ich sinnvollerweise noch "WEB_DEBUG_PRINT" unterbringe und berichte dann.

Zum Thema ModbusMaster:
In der einleitenden Beschreibung der ShineWiFi-ModBus.ino steht (nur): "Download ModbusMaster by Doc Walker"
(@otti, vielleicht bei Gelegenheit in der Beschreibung noch ergänzen, so wie bei WifiManager und MQTT Client)
Ich hab mir das dann [hier](https://github.com/4-20ma/ModbusMaster) runtergeladen und über Sketch -> Bibliothek einbinden -> .ZIP Bibliotherk hinzufügen installiert. Sieht dann so aus:
[image]

Die Aussage "Enlighten your Arduine to be a Modbus master." hat mich  zunächst etwas grübeln lassen, aber ich gehe davon aus, dass das in der Growatt.cpp erfolgt.

Die Ausgabe beim Compilieren und Hochladen zeigt u.a. folgendes an:
`Compiling library "ModbusMaster-master"`
`Bibliothek ModbusMaster-master in Version 2.0.1 im Ordner: C:\Users\ich\Documents\Arduino\libraries\ModbusMaster-master  wird verwendet`

Fehlermeldung hab ich keine gefunden. Hab ich was beim Einrichten übersehen? Nutzt Du auch die Version 2.0.1?

Bernd9000, 2022-08-15:
JA, ist bei mir auch die Version 2.0.1

achim-70, 2022-08-15:
Hey Bernd,
erstmal vielen Dank, dass Du immer so schnell auf meine Posts reagiert hast. Ich bin jetzt erstmal mit meinem Latein am Ende. Hab noch zahlreiche andere Boards ausgewählt und auch mal [modbusmaster-esp8266-master](https://github.com/tekk/modbusmaster-esp8266) statt "nur" ModbusMaster-master verwendet, hat aber alles nix geholfen. Bei Deiner Idee mit dem WEB_DEBUG_PRINT bin ich auch nicht recht voran gekommen, weil ich keinen "Einstiegspunkt" gefunden habe, wo ich hätte hier was ins Log schreiben lassen können bzgl. Modbus. Ich hab vorhin andere Hardware bestellt, soll morgen sogar schon da sein. Dann sehen wir weiter.

achim-70, 2022-08-17:
Nun hab ich neue Hardware (D1 mini v3 von AZ Delivery). Das Flashen ging auf Anhieb durch und der AP "GrowattConfig" wird auch angeboten. Nur kann ich nicht wirklich eine Verbindung aufbauen. Ich hab es von vier verschiedenen Geräten (2 x Handy, 2 x PC/Laptop) versucht, immer das gleiche Verhalten. Zum einen verschwindet die SSID immer wieder mal, kommt aber wieder zurück, ohne dass ich das Board neustarte. Die blau LED leuchtet dauerhaft. Wenn ich es mal schaffe, das Passwort einzugeben, kommt die Meldung "Netzwerkanforderungen werden überprüft" (mal länger, mal nur ganz kurz) und danach die Meldung "Keine Verbindung mit diesem Netzwerk möglich". Einmal ist es mir bisher gelungen, dass sich eine Verbindung aufgebaut hat, aber dann hat sich die Startseite des AP nicht geöffnet und über 192.168.4.1 war sie auch nicht zu erreichen :-(

Bernd9000, 2022-08-17:
Ja, da hab ich die wemos auch her.
Hast Du an den Taster gedacht ?
#define AP_BUTTON_PRESSED ( digitalRead(D7) == LOW )

achim-70, 2022-08-17:
Oh weh, daran hab ich nicht mehr gedacht, Asche auf mein Haupt...
Aber leider klappt auch das nicht. Gleiches Verhalten, wie zuvor beschrieben.
Ich hab dann mal testweise eines der mitgelieferten Beispiele für "NodeMCU 1.0" installiert (ESP8266WiFi/WiFiAccessPoint). Da hab ich exakt das gleiche Verhalten und dort ist ja nichts definiert à la "AP_BUTTON_PRESSED". Somit würde ich das Problem nicht bei dieser Variablen sehen. Aber wo dann? Hab schon mal nach dem Problem eines instabilen AP gegoogelt und bin da nicht der erste, aber die Lösungen, die ich bisher gefunden haben, haben nicht geholfen. Hab auch die Werte bei "Flash Size" alle probiert, sind bei dem NodeMCU zum Glück nur vier Einträge. Damit bin ich aber auch nicht weiter gekommen.
Ach ja, ich hab auch mal die "geforkte" Variante von BeoQ drauf gespielt. Da gibt es ja keinen AP Modus und man kann die Zugangsdaten zum WLAN direkt in der ShineWiFi-ModBus.ino erfassen. Nach dem Flashen hat dann die blaue LED nur geblinkt und es wurde keine Verbindung zu meinem WLAN aufgebaut.

BeoQ, 2022-08-17:
Servus beinand,

Wenn schon eines der bei Arduino mitgelieferten Beispiele nicht tut, dann liegt der Fehler häufig in den Compile-Einstellungen der IDE. Schau mal, ob dort als Board der Generic ESP8266 oder NodeMCU oder LoLin D1 oder sowas eingetragen ist. Auch die Größe der Flash-Size bzw OTA-Speicher macht manchmal Ärger: häufig 1M oder 4M.

Bei den Beispielen solltest du den NodeMCU oder D1 am PC angesteckt lassen, und sehen ob du im Seriellen Monitor (via USB-Programmierkabel) etwas aufschlussreiches siehst. Die meisten Beispiele nutzen den Seriellen Port für Debug-Ausgaben...
Mein Tip wäre, das beispiel ESP8266Webserver/HelloServer auszuprobieren. Der sollte auch Debug über den Monitor liefern.

Mein Fork von der Firmware ist den Neuerungen von Otti etwas hinterher, ich hab noch nicht die Zeit gefunden das nachzuziehen. Wie auch immer: damit hatte ich es schonmal auf meinem Wemos-D1-Clone am TL600-X testweise für ein paar Tage erfolgreich am laufen.
Und ja, einen AP kann diese Version nicht, entweder bucht er sich im konfigurierten WLAN ein, oder eben nicht. ;-)

Cheers,
  Flo

Bernd9000, 2022-08-17:
Was mir gerade noch einfällt, hast Du auch das 5GHz Band bei deinem Wlan ?
Das muste ich bei allen Wemos immer abschalten bei der ersten Verbindung  da diese nur mit dem 2,4 GHz Band arbeiten.
Hat es dann einmal geklappt konnte ich das 5GHz Band wieder einschalten

Bernd9000, 2022-08-18:
@otti 
Hinweis:
Es gibt Probleme falls man mehr als ein Modul hat und mit dem MQTT Broker verbunden ist.
Der Broker kann dann die Anmeldung nicht mehr zuordnen weil der selbe Name benutzt wird ("GrowattWL")

 if (MqttClient.connect("GrowattWL", mqttuser.c_str(), mqttpwd.c_str(), mqtttopic.c_str(), 1, 1, "{\"Status\": \"Disconnected\" }"))

Ich hab das durch eine weitere Variable "INVERTER_NAME" in config.h gelöst.
Wenn ich die Variable noch in der Index.h (<h2>Growatt Inverter</h2>) einfügen könnte, das wäre Perfekt.
Aber das ist wohl nicht so einfach...

otti, 2022-08-18:
Hi @Bernd9000 ,

ich habe ein define mit dem namen MQTT_ID in der Config.h hinzugefügt.
Die Anzeige der ID in der Webseite ist wirklich nicht so einfach zu realisieren. Man könnte die ID auch noch mit in das JSON Paket packen. Dann wäre es recht simpel.

Danke für eure aktive mitarbeit

achim-70, 2022-08-18:
Hallo Flo,
danke, dass Du Dich auch an meiner Lösungssuche beteiligst!
Die von Dir genannten Boards hab ich alle schon rauf und runter getestet, hab mich aber jetzt mal auf das NodeMCU 1.0 "eingeschossen", weil das ja bei Bernd funktioniert hat. Hattest Du das auch verwendet?

Hier jetzt mal alle meine Einstellungen (über Werkzeuge):
Upload Speed: 115000
Debug Port: Disabled
Flash Size: 4 MB (FS:1MB OTA~1019KB)
C++ Exceptions: Disabled
IwIP Variant: v2 Lower Memory
Builtin Led: 2
Debug Level: None
MMU: 32KB Cache + 32 KB IRAM (balanced)
Non-32-Bit Access: Use pgm_read macros…
SSL Support: All SSL ciphers
Stack Protection: Disabled
VTables: Flash
Erase Flash: All Flash Contents
CPU Frequency: 80 MHz

Vielleicht fällt euch ja noch was dazu ein. Bei "Flash Size" hab ich auch schon alle vier Möglichkeiten probiert.

Im seriellen Monitor kommt erst das:
{$l��|�$�|�l�c<����;�c�#��gn�l'o���cx��l{lsd8�o��l��co�|�Ǐc��g'�l��l`�og�dor���g;s�`x�o�s������cg�|�c��og��d`�og�dgsǛ�gr$`{ۓnsl`�����l`��g�l

Nach einigen Minunten wurde das noch angezeigt:
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9

Der HelloServer verbindet sich mit meinem WLAN und ich kann die Seite auch aufrufen, die die Meldung "hello from esp8266!" ausgibt. Der serielle Monitor zeigt folgendes an:
...........................................................................
Connected to <meine SSID>
IP address: 192.168.0.117
MDNS responder started
HTTP server started
A useless web hook has passed
(this hook is in 0x402010ac area (401x=IRAM 402x=FLASH))
A useless web hook has passed
(this hook is in 0x402010ac area (401x=IRAM 402x=FLASH))

Ist das mit dem "useless web hook" relevant?

@Bernd9000: Ich hab das 5GHz Band mal ausgeschaltet, hat aber auch keine Besserung gebracht. Weder der AP Mode im "otti-Original" war stabil, noch hat sich der Fork von Flo im WLAN angemeldet.

Bernd9000, 2022-08-19:
@achim-70 
Die Einstellungen unter Werzeuge hab ich genau so, daran liegt es wohl nicht.
Die Hieroglyphen Im seriellen Monitor hab ich auch, manchmal hilft es die Baudrate zu verstellen, keine Ahnung woran das liegt.
Du könntest noch schauen ob alle Bibliotheken aktuell sind und die neuste Version von Otti mal testen.

achim-70, 2022-08-19:
Ich hab die ganze Zeit mit der 2.0.0-rc9.1 gearbeitet und mir jetzt auch mal die 1.8.19 runter geladen. Aber mit keinem anderen Ergebis. Will die Version mal noch auf einem Gerät ausprobieren, auf dem die neuere Version nicht schon drauf ist. Außerdem hab ich bisland unter Windows 11 gearbeitet und probiere mal noch 10 aus. Die Bibliotheken sind aktuell und die neue Version von Otti hab ich auch schon probiert.

Immerhin hab ich aktuell den Zwischenstand, dass das HelloServer-Beispiel funktioniert, das Flo vorgeschlagen hatte.

Ich hab am Wechselrichter seit Februar 2021 den ShineWifi-X betrieben mit der Datenübertragung zu Growatt. Ist aber vor einigen Wochen ausgefallen und nur noch sporadisch ans Netz gegangen. Das Problem haben wohl etliche andere auch. Daher hab ich mich mal dran gesetzt, diese Version hier zu verwenden. Zunächst eben mit einem Wemos D1. Seit gestern hab ich aber auch einen Stick, mit dem aufgelöteten Widerstand (hat mir ein Kollege gemacht, löten kann ich nicht wirklich gut). Erste Versuche sind aber auch gescheitert, wobei ich noch nicht verstanden habe, wann und wie lange genau der Stick im Programmier-Modus bleibt. Aber dazu gibt es ja einen anderen Thread. Dort werde ich mich wohl auch mal blicken lassen...

Spexx-, 2022-08-19:
Also mit den Kommentaren hier habe ich es nun auch endlich zum Laufen bekommen. Ich hatte anfangs noch die Meldung "Disconnected", das konnte ich aber auch beheben indem ich in der Config.h folgenden Eintrag verwendet habe:
 #define AP_BUTTON_PRESSED ( digitalRead(D7) == LOW ). Ich nutze einen D1 Mini (Clone von AZ-Delivery).

Im Log habe ich paar Einträge mit "NOT successful", aber vermutlich kann man das ignorieren?

Passt nicht ganz ins Thema aber hat jemand bereits Änderungen vorgenommen, damit ich die Werte als einzelne Datenpunkte im ioBroker verfügbar habe? Mit JSON kann ich nicht ganz so viel anfangen...

achim-70, 2022-08-19:
@Spexx- 
Mit welcher Version von Arduino IDE arbeitest Du und auf welchem Betriebssystem?

Spexx-, 2022-08-19:
Meine Arduino IDE ist nicht ganz aktuell, ich nutze die Version 1.8.5 unter Windows 10.

Habe zuvor alle Bibliotheken aktualisiert sowie auch das Esp8266 Board (gab sonst Fehlermeldungen beim Kompilieren).

Bernd9000, 2022-08-19:
@Spexx- 
Das mit dem "NOT successful" im Log habe ich auch. Das sagt einfach nur das der Wechselrichter nicht geantwortet hat über Modbus. Ich weiß nicht ob das am Wechselrichter liegt weil der voll beschäftigt ist und nicht antworten kann oder ob es am wemos liegt.Man könnte mal versuchen den Abfrage Intervall höher zu setzen in der config.h (REFRESH_TIMER).

Wegen dem JSON Format benutze ich im IOBroker Node-red.

achim-70, 2022-08-20:
Kurzer Zwischenstand: Unter Windows 10 und Arduino IDE 1.8.19 hab ich den Fork von Flo auf Anhieb zum Laufen gebracht und die Daten landen in meinem FHEM. Jetzt brauche ich mal ein paar Tage Pause, dann werde ich mich auf die Suche machen, ob es tatsächlich am Windows 11 liegt oder die Arduino IDE Version 2.0.0 noch nicht so ganz will, wie sie soll.

Cees2006, 2022-08-21:
@Spexx- Wenn du noch Hilfe benötigst, wie du den JSON in iobroker-Datenpunkte auflösen kannst, dann lass es mich wissen. Bei mir läuft das mit einem Script. Gruss & special thx to @otti for this ino-project ;-)

achim-70, 2022-08-23:
Nächster Zwischenstand: In meiner Win 10 & Arduino IDE 1.8.19 Umgebung hab ich auch die Version von otti ohne Probleme sowohl auf den D1 mini, als auch auf den ShineWiFi-X Stick flashen können. Beide haben auch nach der Konfiguration über den AP Modus Daten aus dem Wechselrichter geliefert.
Unter Win 11 & Arduino IDE 1.8.19 konnte ich jetzt zumindest mal den Stick flashen und anschließend konfigurieren, aber der D1 mini stellt nach dem Flashen noch keinen stabilen AP zur Verfügung. Jetzt bin ich erstmal ein paar Tage weg. Weitere Nachforschungen daher erst ab nächster Woche.

Solpowersun, 2022-08-23:
> Siguiente estado intermedio: en mi entorno Win 10 y Arduino IDE 1.8.19, también pude flashear la versión otti tanto en el D1 mini como en el dispositivo ShineWiFi-X sin ningún problema. Ambos también entregaron datos del inversor después de la configuración a través del modo AP. Bajo Win 11 y Arduino IDE 1.8.19 pude al menos flashear el dispositivo y luego configurarlo, pero el D1 mini no proporciona un AP estable después del flasheo. Ahora estaré fuera unos días. Más investigaciones, por lo tanto, solo a partir de la próxima semana.

Hi i would like to update my shinewifi-x but i want to do backup first, do you know the way? 
My shinewifi-x has version firmware 3.1.0.5 ap mode not work with default original firmware...

Let me know when you can.

Thanksss

Spexx-, 2022-08-28:
Ich habe immer noch ein WLAN-Netzwerk mit dem Namen "FaryLink...". Wie lässt sich das am Einfachsten deaktivieren?

Bernd9000, 2022-08-29:
@otti 
Ich würde gerne den Holding Register 3 im JSON mit aufnehmen (Ausgangsleistung in %).
Ich krieg das irgendwie nicht hin, hast Du eine Idee?

otti, 2022-08-29:
Hi @Bernd9000 ,

ich habe mir den Pull Request #30 von Tom noch nicht vollständig angeschaut. Aber beim Überfliegen schaut es so aus, als hätte er die Möglichkeit eingebaut, sich alle verfügbaren Daten frei zusammenstellen zu können. Das sollte demnächst also verfügbar sein.

Bernd9000, 2022-08-29:
Danke, werd ich mir Morgen anschauen.

zinserjan, 2022-09-10:
@Bernd9000 der PR ist übrigens gemerged. Mit v.1.20 müsstest du bei deinem 600TL-X die Ausgangsleistung in % bekommen:
```json
{
...
  "OnOff": 1,
  "CmdMemoryState": 0,
  "ActivePowerRate": 100,
}
``` 
Funktioniert jedenfalls bei meinem 1000TL-X und der sollte gleich funktionieren.

Setzen muss man das leider noch manuell über die Register (irgendwann hoffentlich mal über MQTT), aber zumindest werden diese schon mal ausgelesen. Wenn ich das richtig verstanden habe muss man CmdMemoryState (Register 2) noch auf 1 setzen, damit der Inverter die Einstellungen beim Neustart behält.

ansem187, 2022-09-13:
@zinserjan ich regle den 1500TL-X auch über das Register 3. Derzeit noch so weit reduziert um den Grundbedarf abzudecken. Ich hatte bereits versucht via MQTT Subscribe den aktuellen Verbrauch meines Smartmeters abzufangen, um dadurch den WR für eine Nulleinspeisung regeln zu können. Leider reichen meine Programmierkenntnisse wohl nicht genug aus. Zumindest lief der Stick nach meiner Modifizierung nicht mehr sauber hoch. Wenn man diese möglichkeit irgendwie einbauen kann, könnte man damit sogar den SDM630 ersetzen.

Griizly, 2022-09-27:
Hi ! Sorry I don't speak german but I translated this issue and I see that some have the same issue as me,
I have flashed a Wemos D1 mini pro with a CH340 chip, modbus protocol defined is 1.20 ( my inverter is a growatt MIN 4200 TL-XH )
I enabled the webdebug and now I see that my board seems to not communicate with my inverter because I see :
```
#0: Setup()
#1: Error: Unknown Shine Stick
#2: Error: Unknown Shine Stick
#3: Error: Unknown Shine Stick
```

did someone managed to get it working ? 

thanks !

Bernd9000, 2022-09-27:
@ansem187 
Als Tipp,
Ich hab den Code auf dem Stick so gelassen wie es ist und regel den Wechselrichter über ein HTTP POST Befehl .
Das ganze mach ich über ioBroker und Node-red.

Adresse:
http://xxx.xxx.xxx.xxx/postCommunicationModbus_p

Werte:
    "reg": 3,
    "val": wert,
    "type": '16b',
    "operation":'W',
    "registerType":'H'

"wert" ist hier meine errechnete Variable die die Ausgangleistung in Prozent enthält.
Die Werte Netzbezug / Einspeisung muß Du natürlich erstmal haben zum errechnen der Differenzleistung.
Jedenfalls funktioniert das als Nulleinspeisung ganz gut.

Bernd9000, 2022-09-28:
> Hi ! Sorry I don't speak german but I translated this issue and I see that some have the same issue as me, I have flashed a Wemos D1 mini pro with a CH340 chip, modbus protocol defined is 1.20 ( my [...]

Hi
try this in File "config.h"
#define ENABLE_DEBUG_OUTPUT 0

Griizly, 2022-09-28:
Hi ! I will try this after work and let you know if it worked ! thanks !

Griizly, 2022-09-28:
Still not working, mqtt error is normal i don't bother to set the mqtt anymore 
[image]

Griizly, 2022-09-28:
Should I restart my inverter after inserting my board maybe ? ( Which i checked, he has a CH340 on it not a ftdi )

Bernd9000, 2022-09-28:
> Should I restart my inverter after inserting my board maybe ? ( Which i checked, he has a CH340 on it not a ftdi )
yes, restart inverter
my stick is not always recognized too.
And wait after start....

Griizly, 2022-09-29:
It works !! thanks a lot !!

closed 2022-12-01 by crasu