OpenInverterGateway/OpenInverterGateway #23: Error: Unknown Shine Stick

Blueforcer, 2022-08-01:
i flashed this firmware on the ShineWifi-X

in debug log it always tell me 
Error: Unknown Shine Stick

how to solve this?

achim-70, 2022-08-23:
@Blueforcer have you found a solution?

Blueforcer, 2022-08-23:
Yes i just removed the complete if clause
and set the type manually with
_eDevice = ShineWiFi_X; // USB

achim-70, 2022-08-23:
Ok, can you then close this thread?

Blueforcer, 2022-08-23:
Its still a bug, istn it?

achim-70, 2022-08-23:
Did you already flashed the new version from last week?

DocPolyester, 2022-09-24:
I also had an issue with connecting to inverters ModBus.
I own a MIC 1500TL-X and a  ShineWiFi_X USB and both new I got it 2 weeks ago.

Solution for me was a change of ModBus adress in Growatt.cpp
>>>
   // init communication with the inverter
    Serial.begin(115200);
    Modbus.begin(0, serial); // originally here is Modbus.begin(1, serial); which didnt work with my new inverter
    res = Modbus.readInputRegisters(0, 1);
 <<<

Maybe this helps someone to solve "unkown shine stick" issue

DocPolyester, 2022-09-24:
Dealin with same issue.
My setup:
Growatt MIC 1500TL-X and ShineWifi-X USB.

Currently I think its some timing issue, after inserting  a delay of 1.5sec in Growatt.cpp it seems the InputRegisters are ready...

Serial.begin(115200);
Modbus.begin(1, serial);
delay(1500);
res = Modbus.readInputRegisters(0, 1);
if(res == Modbus.ku8MBSuccess) {
..
I am not the ModBus library expert maybe there is a better solution...

crasu, 2022-09-26:
@DocPolyester @Blueforcer could you try to force an inverter with the functionallity from this pr #40?

crasu, 2022-12-04:
Closed due to no replay

closed 2022-12-04 by crasu