OpenInverterGateway/OpenInverterGateway #24: Error at compile in Arduino IDE

Stef293, 2022-08-02:
Hi great job,

but i have trouble to compile the Project. 

ShineWiFi-ModBus.ino:71:2: error: #error change MQTT_MAX_PACKET_SIZE to 512
   71 | #error change MQTT_MAX_PACKET_SIZE to 512
      |  ^~~~~

exit status 1

Compilation error: #error change MQTT_MAX_PACKET_SIZE to 512

If i disable mqtt it works but is active, i  get the message.
Thx for help.

otti, 2022-08-03:
Hi Stef293,

the #define MQTT_MAX_PACKET_SIZE can be found here:
C:\Users\<user>\Documents\Arduino\libraries\pubsubclient-master\src\PubSubClient.h
Set its value to 512 and it will compile without errors

Regards,
Otti

Stef293, 2022-08-03:
thx, it works.
one more question, why you have the inverter data as json format?
it's not better to send it as datapoint in iobroker?

greetings
Stef

otti, 2022-08-03:
Hi,

the answer is simple. I do not use iobroker :-)
JSON is one of the standard data formats for IoT devices. This data packet will be tranfered to a MQTT broker of your choice. It is also possible to get the data with a RESTful api by calling http://<ip>/status.
I think both methods will be supported by iobroker

achim-70, 2022-08-23:
@Stef293 could this thread be closed?

closed 2022-08-29 by otti