OpenInverterGateway/OpenInverterGateway #31: d1mini clone

That-Dude, 2022-08-26:
Hi, I successfully installed onto my D1 mini clone (after changing AP_BUTTON_PRESSED) but my weblog just repeats with:
```
#1: Error: Unknown Shine Stick
#2: Error: Unknown Shine Stick
#3: Error: Unknown Shine Stick
```

My inverter model is Growatt MIN 3600TL-XE

Immediately after start-up it does make a single connection to my MQTT server with ```{"Status": "Disconnected" }``` but then stops sending MQTT messages.

Do you have any idea what I am missing?

my Config.h file:
```
#define MQTT_SUPPORTED 1
#define PINGER_SUPPORTED 0
#define ENABLE_DEBUG_OUTPUT 0
#define ENABLE_WEB_DEBUG 1
#define MQTT_ID           "GrowattWL"
#define SIMULATE_INVERTER 0
#define SIMULATE_DEVICE ShineWiFi_X
#define HOSTNAME          "Growatt"
#define APPassword        "growsolar"
#define UPDATE_USER       "admin"
#define UPDATE_PASSWORD   "admin"
// #define AP_BUTTON_PRESSED ( analogRead(A0) < 50 )
#define AP_BUTTON_PRESSED ( digitalRead(D7) == LOW )
#define REFRESH_TIMER 5000 // 5s default
#define WIFI_RETRY_TIMER 120000 // 120s default
#define LED_TIMER 500 //  0.5s default
#define BUTTON_TIMER 500 //  0.5s default
#if PINGER_SUPPORTED == 1
#define GATEWAY_IP IPAddress(192, 168, 178, 1)
#endif
```

This is my D1 Mini clone:

[image]

[image]

That-Dude, 2022-08-26:
Reading a the open issues I'm going to guess this is because my D1-mini clone does not have a CH340 USB-serial controller, that FTDI chip is incompatible.

I've ordered a CH340 based ESP8266 to test this, will update soon.

That-Dude, 2022-08-30:
The problem was the FTDI chipset on my 8266 device, as soon as I flashed to a CH340 variant everything worked.

closed 2022-08-30 by That-Dude

Griizly, 2022-09-27:
Hi ! I also have the wemos d1 but with a CH340 chip, which modbus version did you used ? I still get the unknow shine stick error :s

That-Dude, 2022-09-27:
> 

I'm not sure what you mean by 'which modbus version'. I just built and flashed with Arduino IDE and plugged it into the USB port on the inverter - it work instantly.

Griizly, 2022-09-27:
It's exactly what I did, but for a strange reason it does not detect my board, I keep getting a message in the webdebug saying this: 
#0: Setup()
#1: Error: Unknown Shine Stick

I will propabily start again from scratch 

For the modbus version you can specify it in the Config.h file, author set it by default on 124 ( 1.24 )
[image]

from what I see from your config file I see that it is an older version because now some settings have changed

That-Dude, 2022-09-27:
Oh I see! It depends on the Inverter model, mine was 120. Why not just try the other two to rule them out?

Do you have anything connected the modbus pins in the sys/com port? Have you reboot the Inverter? Try a different brand of d1 mini clone.

Griizly, 2022-09-27:
Mine is very similar to yours, it's a MIN 4200 TL - XH

I did try the 2 others versions but no changes, I never tried to reboot the inverter, do I need to do it in order for the inverter to accept the board ?

I will try to compile an older version of the code, I extrapolate a bit and took the latest version before your issue which is the 18 Aug

Nothing is connected to the COM ports

Only other board I have is a nodemcu esp-12E ( ESP8266 ) but I think it's not with a CH340 chip :s