OpenInverterGateway/OpenInverterGateway #32: Export control via registers?

That-Dude, 2022-08-26:
Hi, using the ability to write to registers, has anybody managed to enable export control on their inverter?

That-Dude, 2022-08-30:
Is is possible by writing to holding register 122

0: Disable exportLimit;
1: Enable 485 exportLimit;
2: Enable 232 exportLimit;
3: Enable CT exportLimit;

However, I'm not sure what the long term effects of writing to registers frequently will have on the EEPROM. I'd appreciate an opinions on this.

closed 2022-08-30 by That-Dude

ansem187, 2022-08-30:
Hello I am currently trying to realize a zero export. Do you have a code example to write the registers? I want to send the current consumption via MQTT from the smart meter to the ShineWifi stick to dynamically control the export limit.