OpenInverterGateway/OpenInverterGateway #42: 2022-08-24 Update: compiler error

EmHenkey, 2022-09-11:
Moin and hello,
did anyone compiled this update?
I got this error: 
***Growatt.cpp: In member function 'void Growatt::CreateJson(char*, const char*)':
Growatt.cpp:303:30: error: 'MQTT_MAX_PACKET_SIZE' was not declared in this scope
  303 |   serializeJson(doc, Buffer, MQTT_MAX_PACKET_SIZE);

Gruß Martin

tommarek, 2022-09-11:
Are you using Arduino IDE or platformio?

EmHenkey, 2022-09-11:
Arduino IDE

tommarek, 2022-09-11:
I thought so... The issue is that when using platformio this constant is taken from the platformio.ini but when using Arduino IDE it doesn't get initialised. This is my bug, sorry. I will fix this first thing tomorrow morning.

tommarek, 2022-09-12:
should be fixed in #44

closed 2022-09-12 by otti