OpenInverterGateway/OpenInverterGateway #46: autodiscovery mode for mqtt?

MichelFreund, 2022-09-17:
Unlike other devices which I import in for Home Assistant, the growatt firmware doesn't have itself autodiscovered in Home Assistant. 

Would adding this functionality be an option?

otti, 2022-09-17:
Sounds like a nice feature
[https://www.home-assistant.io/docs/mqtt/discovery/](https://www.home-assistant.io/docs/mqtt/discovery/)

Autodiscovery is also support by other systems like ESPHome. It seems, that there is no profile specially dedicated to inverters. But I think "Sensor" or "Number" can be used instead. 

If someone is willing to implement autodiscovery, I would appreciate that

MichelFreund, 2022-09-18:
If I knew how exactly, I would for sure do this....

Looks to me its a bit like described here:

https://roelofjanelsinga.com/articles/mqtt-discovery-with-an-arduino/

And here is someone who transformed your code to a yaml for esphome, contains some hints too:

https://community.home-assistant.io/t/esphome-modbus-growatt-shinewifi-s/369171

Ill try to dive into this, but my arduino / programmings knowledge is quite limited.

mgerald21, 2022-09-18:
I would suggest a mapping like this ...

PV1Voltage, PV2Voltage, GridL1Voltage, GridL3Voltage, GridL3Voltage ->
  - platform: mqtt                            
    name: "Voltage X"                          
    state_topic: "...topic"
    unit_of_measurement: "V"                              
    icon: 'mdi:alpha-v-circle-outline'                               
    value_template: "{{ value_json.value|float }}"

PV1InputCurrent, PV2InputCurrent, GridL1OutputCurrent, GridL2OutputCurrent, GridL3OutputCurrent
  - platform: mqtt                   
    name: "Current X"                             
    state_topic: "...topic"
    unit_of_measurement: "A"           
    icon: 'mdi:alpha-a-circle-outline'                               
    value_template: "{{ value_json.value|float }}"

InputPower, PV1InputPower, PV2InputPower, OutputPower, GridL1OutputPower, GridL2OutputPower, GridL3OutputPower
  - platform: mqtt                         
    name: "Power X"          
    state_topic: "...topic"    
    unit_of_measurement: kW                      
    device_class: power                      
    value_template: "{{ value_json.value|float }}"

GridFrequency
  - platform: mqtt              
    name: "Grid frequency"                    
    state_topic: "...topic"
    unit_of_measurement: Hz                                                           
    icon: 'mdi:current-ac'
    value_template: "{{ value_json.value|float }}"                                        

EnergyToday, EnergyTotal, PV1EnergyToday, PV1EnergyTotal, PV2EnergyToday, PV2EnergyTotal
  - platform: mqtt                    
    name: "Energy X"              
    state_topic: "..topic" 
    unit_of_measurement: kWh                      
    device_class: power                                          
    value_template: "{{ value_json.value|float }}"

InverterTemperature, InverterIPMTemperature
  - platform: mqtt                    
    name: "Temperature X"              
    state_topic: "..topic" 
    unit_of_measurement: °C                      
    device_class: temperature                                          
    value_template: "{{ value_json.value|float }}"

MichelFreund, 2022-09-20:
I had to define class to Energy to be able to show in Energy Dashboard:

mqtt:
  sensor:
    - state_topic: "solar"
      device_class: "energy"
      name: "OutputPower"
      state_class: "total_increasing"
      unit_of_measurement: "kW"
      value_template: "{{ value_json.OutputPower|float }}"

  sensor:
    - state_topic: "solar"
      device_class: "energy"
      name: "TodayGenerateEnergy"
      state_class: "total_increasing"
      unit_of_measurement: "kWh"
      value_template: "{{ value_json.TodayGenerateEnergy|float }}"

  sensor:
    - state_topic: "solar"
      device_class: "energy"
      name: "TotalGenerateEnergy"
      state_class: "total_increasing"
      unit_of_measurement: "kWh"
      value_template: "{{ value_json.TotalGenerateEnergy|float }}"

Have to work out the right state_classes though.

StevenBoelhouwer, 2022-09-20:
> If I knew how exactly, I would for sure do this....
>
> Looks to me its a bit like described here:
>
> https://roelofjanelsinga.com/articles/mqtt-discovery-with-an-arduino/
>
> And here is someone who [...]

Check my fork 😊

MichelFreund, 2022-09-20:
Thats great! If i had only checked upon existing forks 😂.

@otti , maybe a merge into original?

StevenBoelhouwer, 2022-09-20:
Yes, fork is behind and I’ve changed a couple things on my own like WiFi manager, ArduinoJSON library. Didn’t have much time to look at latest changes as it works great for me already. I don’t have much time atm but I can have a look sometime next week.
Or look at my source which contains the auto discovery and implement that.

crasu, 2022-09-24:
The fork has quite some incompatible changes, but I guess its possible to take over the discovery functionallity.

Griizly, 2022-09-26:
I know it's not really the good place to ask but could someone please explain how to add data into homeassistant ?
I have mosquito running on my homeassistant
I flashed the code on my nodemcu board, set up my wifi and entered mqtt server info but nothings shows up in home assistant
[image]

MixelF, 2022-09-26:
- platform: mqtt
  state_topic: "solar"
  device_class: "power"
  name: "InputPower"
  state_class: "measurement"
  unit_of_measurement: "kW"
  value_template: "{{ value_json.InputPower|float }}"

- platform: mqtt
  state_topic: "PV1Voltage"
  name: "PV1Voltage"
  unit_of_measurement: "V"
  device_class: "voltage"
  icon: mdi:flash
  value_template: "{{ value_json.PV1Voltage|float }}"

- platform: mqtt
  state_topic: "PV1InputCurrent"
  name: "PV1InputCurrent"
  unit_of_measurement: "A"
  device_class: "Current"
  icon: mdi:flash
  value_template: "{{ value_json.PV1InputCurrent|float }}"

- platform: mqtt
  state_topic: "PV1InputPower"
  name: "PV1InputPower"
  unit_of_measurement: "kW"
  device_class: "power"
  icon: mdi:flash
  value_template: "{{ value_json.PV1InputPower|float }}"

- platform: mqtt
  state_topic: "PV2Voltage"
  name: "PV2Voltage"
  unit_of_measurement: "V"
  device_class: "voltage"
  icon: mdi:flash
  value_template: "{{ value_json.PV2Voltage|float }}"

- platform: mqtt
  state_topic: "PV2InputCurrent"
  name: "PV2InputCurrent"
  unit_of_measurement: "A"
  device_class: "current"
  icon: mdi:flash
  value_template: "{{ value_json.PV2InputCurrent|float }}"

- platform: mqtt
  state_topic: "PV2InputPower"
  name: "PV2InputPower"
  unit_of_measurement: "kW"
  device_class: "power"
  icon: mdi:flash
  value_template: "{{ value_json.PV2InputPower|float }}"

- platform: mqtt
  state_topic: "solar"
  device_class: "energy"
  name: "OutputPower"
  state_class: "measurement"
  unit_of_measurement: "kW"
  value_template: "{{ value_json.OutputPower|float }}"

- platform: mqtt
  state_topic: "solar"
  device_class: "frequency"
  name: "GridFrequency"
  state_class: "measurement"
  unit_of_measurement: "Hz"
[... 140 lines cut ...]
  value_template: "{{ value_json.PVEnergyTotal|float }}"
   
- platform: mqtt
  state_topic: "solar"
  name: "InverterTemperature"
  unit_of_measurement: C
  device_class: temperature
  icon: mdi:flash
  value_template: "{{ value_json.InverterTemperature|float }}"
  
- platform: mqtt
  name: "TemperatureInsideIPM"
  state_topic: "solar"
  unit_of_measurement: C
  device_class: temperature
  icon: mdi:thermometer
  value_template: "{{ value_json.TemperatureInsideIPM|float }}"

- platform: mqtt
  name: "BoostTemperature"
  state_topic: "solar"
  unit_of_measurement: C
  device_class: temperature
  icon: mdi:thermometer
  value_template: "{{ value_json.BoostTemperature|float }}"

MixelF, 2022-09-26:
put that in sensors.yaml and reference to it in configuration.yaml ;-)

MixelF, 2022-09-26:
probably also could have combined a lot of those entries into one, but well, it was a quick one

Griizly, 2022-09-26:
Also I see that my inverter is on FW 1.0 it is a growatt MIN4200TL-XH
[image]

MixelF, 2022-09-26:
I dont know your board. But apparantly it doesnt get data from your inverter indeed.

Griizly, 2022-09-26:
Yeah I think thats the issue, I have a brand new MIN 4200TL-HX inverter

I also tested with a wemos D1 mini pro I had and same behavior, i opened an issue, maube the author can give me some hints or advices :)
thanks for trying to help me !

Griizly, 2022-09-26:
I will have to wait until tomorrow before being sure but I think that the problem was between the chair and the keyboard --'
I'm using a "generic" esp8266, but I didn't commented this line in the config.h file which is specific to growatt board...
`#define AP_BUTTON_PRESSED ( analogRead(A0) < 50 )`
Now I see the graph, which is all zeroes for now, it never showed before...
[image]

I'll let you know if it's working tomorrow

Griizly, 2022-09-27:
Unfortunatelly still nothing :s
I will try the 2 other modbus version, here intried the 1.24
[image]
[image]

MixelF, 2022-09-27:
What do your home-assistant and mqtt broker logs say? Have you tried without setting an mqtt user/password in both broker, firmware and home assistant?

Griizly, 2022-09-27:
when I connect my board to the serial monitor I can see that it manage to connect to the mqtt broker so I think that is not the problem
[image]

I really think that the issue is that my inverter does not have the same modbus protocols as the one available here, I will try version 3.05 and 1.20 and see what happens

Griizly, 2022-09-27:
I enabled webdebug and I see this : 
```
#0: Setup()
#1: Error: Unknown Shine Stick
#2: Error: Unknown Shine Stick
#3: Error: Unknown Shine Stick
```

Griizly, 2022-09-29:
I had to restart my inverter + setting this to 0 
#define ENABLE_DEBUG_OUTPUT 0

thanks for helping me ! It is now integrated in my home assistant :D 
[image]

Solpowersun, 2022-10-27:
> Tuve que reiniciar mi inversor + configurando esto en 0 #define ENABLE_DEBUG_OUTPUT 0
> 
> gracias por ayudarme ! Ahora está integrado en mi asistente doméstico :D [image]

hi, i have a growatt min 6000TL-XH i would like to get the same result! 
I have a shinewifi-x but i don't know what is the modbus version for update firmware...
Can you see the information in real time every x secs?

let me know! thanks a lot!

Solpowersun, 2022-11-02:
Let me know please....

Griizly, 2022-11-02:
Hi ! sorry I completely forgot to reply sooner
I'm using modbus 1.24 with a wemos D1 board, I don't have a shinewifi-x
regarding the refresh rate, home assistant get data drom my board every 3-4 seconds

Solpowersun, 2022-11-02:
can you get data in real time every 3-4 seconds?
wow very interesting, is it possible updating the shinewifi-x ?

anybody know as i check my modbus version? my shinewifi-x has firmware v3.1.0.5

Please let me know!!!

Solpowersun, 2022-11-02:
> Hi ! sorry I completely forgot to reply sooner I'm using modbus 1.24 with a wemos D1 board, I don't have a shinewifi-x regarding the refresh rate, home assistant get data drom my board every 3-4 seconds

thanks a lot for your reply, i follow reading about the better way for getting home assistant real time with shinewifi-x...

Solpowersun, 2022-11-03:
> Hola ! lo siento, olvidé por completo responder antes . Estoy usando modbus 1.24 con una placa wemos D1, no tengo un shinewifi-x con respecto a la frecuencia de actualización, el asistente doméstico obtiene datos de mi placa cada 3-4 segundos.

Where can i see my modbus version in my shinewifi-x?

crasu, 2022-12-01:
@Solpowersun I do not think that you can get the modbus version the stick is using from the gui of the stick.  You have to try which one works. I am using version 124 for my growat tl-x

closed 2022-12-01 by crasu

pasquinadeus, 2023-10-04:
Hi Everyone, 

I found this thread to be very useful for setting up my homeassistant instance after installing the firmware on my ShineWifi-S for an SPH3600 hybrid inverter. Even so I think it took way too long to setup the following information, so I leave it here for possible newcomers. 

First I spent half an hour trying to confirm whether I should flash the firmware at 0x0000 or 0x1000 (this was my problem as I don't have much experience in ESP8266 and was a bit afraid of bricking it - probably recoverable if it did happen). For anyone taking this up in the future, the following works for the published binary file, after connecting the ttl-serial adapter booting the dongle in the correct flashing mode (as described in the wiki):

```
esptool.py erase_flash
esptool.py write_flash 0x0000 firmware-ShineWifiS.bin
```

After setting up mqtt in homeassistant (both the integration and the broker add-on), and confirming that the messages are being published in "energy/solar" (as described in the setup information), place the following in the configuration.yaml file (some fixes on what MixelF wrote above, trying to format it correctly for yaml):

```
mqtt:
  sensor:
    - state_topic: "energy/solar"
      name: "Growatt.InverterStatus"
      unique_id: "mqtt_growatt_inverterstatus"
      value_template: "{{ float(value_json.InverterStatus) | int }}"
      
    - state_topic: "energy/solar"
      name: "Growatt.TotalGenerateEnergy"
      unique_id: "mqtt_growatt_totalgenerateenergy"
      unit_of_measurement: "kWh"
      value_template: "{{ float(value_json.TotalGenerateEnergy) | round(1) }}"
      device_class: energy
      state_class: total_increasing
#      json_attributes_topic: "energy/solar"
#      last_reset_topic: "energy/solar"
#      last_reset_value_template: "1970-01-01T00:00:00+00:00"
#      payload_available: "1"
#      availability_mode: latest
#      availability_topic: "energy/solar"
#      availability_template: "{{ value_json.InverterStatus }}"
      
    - state_topic: "energy/solar"
      name: "Growatt.InputPower"
      unique_id: "mqtt_growatt_inputpower"
      device_class: "power"
      state_class: "measurement"
      unit_of_measurement: "W"
      value_template: "{{ value_json.InputPower|float }}"
      json_attributes_topic: "energy/solar"
      payload_available: "1"

    - state_topic: "energy/solar"
      name: "Growatt.PV1Voltage"
      unique_id: "mqtt_growatt_pv1voltage"
      unit_of_measurement: "V"
      device_class: "voltage"
      icon: mdi:flash
      value_template: "{{ value_json.PV1Voltage|float }}"

    - state_topic: "energy/solar"
      name: "Growatt.PV1InputCurrent"
      unique_id: "mqtt_growatt_pv1inputcurrent"
      unit_of_measurement: "A"
      device_class: "Current"
      icon: mdi:flash
      value_template: "{{ value_json.PV1InputCurrent|float }}"

    - state_topic: "energy/solar"
      name: "Growatt.PV1InputPower"
      unique_id: "mqtt_growatt_pv1inputpower"
      unit_of_measurement: "W"
      device_class: "power"
      icon: mdi:flash
      value_template: "{{ value_json.PV1InputPower|float }}"

[... 340 lines cut ...]
      value_template: "{{ float(value_json.BatteryState) | int }}"

    - state_topic: "energy/solar"
      name: "Growatt.BatteryTemperature"
      unique_id: "mqtt_growatt_batterytemperature"
      unit_of_measurement: C
      device_class: temperature
      icon: mdi:thermometer
      value_template: "{{ value_json.BatteryTemperature|float }}"      
      
    - state_topic: "energy/solar"
      name: "Growatt.BatterySOC"
      unique_id: "mqtt_growatt_batterysoc"
      unit_of_measurement: '%'
      state_class: measurement
      device_class: battery
      icon: mdi:battery
      value_template: "{{ value_json.SOC|int }}"     
```

My inverter is a single phase model, hence I left out the 6 values related to phases 2 and 3. All other information is exposed above. 

Autodiscovery for this would be great, as the above still took a few hours to setup (upvoting this feature request) :).

Anyway, hope it helps and thanks for the software. Growatt is currently punishing homeassistant folks who go directly to their servers with the available integration. The homeassistant crowd is mostly going the Grott way, but I prefer this solution.