OpenInverterGateway/OpenInverterGateway #51: Unable to compile on Arduino IDE 2.0.0

Griizly, 2022-09-21:
Hi i'm trying to compile project but i keep getting these errors

Error was too long to share here so I pasted it here : https://sharetext.me/91cunfpojy

[image]

Any help compiling this would be appreciated :) thanks !

Griizly, 2022-09-21:
I've set the board to my nodemcu board

Here is another link : https://justpaste.it/8ofrs
Error is too long to paste here in comment

MixelF, 2022-09-21:
Did you set the right libraries?

Griizly, 2022-09-21:
I think so, error seems to come from the PubSubClient lib and Wifi manager 
I'm using latest version which is 2.8.0

don't know where it come from 
this is one of the error I see in the output : C:\Users\Maison\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.cpp:547: multiple definition of `PubSubClient::write(unsigned char)';

MixelF, 2022-09-21:
Well, im not the expert, but check carefully if you have the right library. I remember picking one which was a fork; gave me lots of errors too. Author is key 😉

Griizly, 2022-09-21:
I deleted everything and started back from scratch, I managed to flash my nodemcu board
[image]
it's connected to my wifi now :)
I just have to figure out how to get graphs and data

MixelF, 2022-09-21:
Great, nice work! I have used an mqtt broker and on that I subscripe within home assistant. But there are of course loads and loads more possible routes towards visualisation 😉

closed 2022-09-22 by Griizly