OpenInverterGateway/OpenInverterGateway #59: Growatt 2000 s no values

spottich, 2022-11-10:
I have an growatt 2000s  and in the front end isnt arriving any value from the inverter.
The firmware is installed with defaults.
In the attachement is a screenshot with the fault.
Second question ist, is it possible to give a fixed ip in the menue of the stick? 
[image]

Derirrepeter, 2022-11-15:
I have the same problem with my SPH3600 with battery same output, i hope anyone can help to get this running

Derirrepeter, 2022-11-16:
so i invested a lot of time, but in the end its working. i had to change the baud rate from 9600baud to 38400baud and i used the prot 124. and thanks to all who worked on this here its absoltly amazing.

crasu, 2022-12-04:
@spottich I would be interessted to hear if your inverter need 38400baud as well and add this to the docs.

closed 2022-12-04 by crasu