OpenInverterGateway/OpenInverterGateway #64: Fehler beim Kompilieren mit aktivem Webdebug.

smartboart, 2022-11-24:
collect2.exe: error: ld returned 1 exit status
exit status 1
Fehler beim Kompilieren für das Board Generic ESP8266 Module.

`c:/users/michael/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: sketch\ShineWiFi-ModBus.ino.cpp.o:D:\Ablage\01_smart home\Growatt\GrowattShine_neues Projekt\Growatt_ShineWiFi-S-master\SRC\ShineWiFi-ModBus/WebDebug.h:5: multiple definition of `acWebDebug'; sketch\ShineMqtt.cpp.o:D:\Ablage\01_smart home\Growatt\GrowattShine_neues Projekt\Growatt_ShineWiFi-S-master\SRC\ShineWiFi-ModBus/WebDebug.h:5: first defined here
c:/users/michael/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: sketch\ShineWiFi-ModBus.ino.cpp.o:D:\Ablage\01_smart home\Growatt\GrowattShine_neues Projekt\Growatt_ShineWiFi-S-master\SRC\ShineWiFi-ModBus/WebDebug.h:6: multiple definition of `u16WebMsgNo'; sketch\ShineMqtt.cpp.o:D:\Ablage\01_smart home\Growatt\GrowattShine_neues Projekt\Growatt_ShineWiFi-S-master\SRC\ShineWiFi-ModBus/WebDebug.h:6: first defined here`

crasu, 2022-11-28:
@smartboart Could you try the fix from the pr?

smartboart, 2022-11-29:
@crasu hello, i have tried it and the compilation is working as soon as we fix the issue #63 from @domicreation
change in ShineMqtt.h
change- 
#include <ShineWifi.h>
to 
#include "ShineWifi.h"

smartboart, 2022-11-29:
.

commit 783e7862 (2022-12-28): Fix compilation error with WEBDEBUG enabled

Fixes #64