OpenInverterGateway/OpenInverterGateway #65: Total Time  Wert falsch  bzw. ändert sich nicht / wird nicht aktualisiert MQTT 305

smartboart, 2022-11-25:
Ich hatte ein eigenes Programm laufen, welche die Totaltime mittels 30 31 übertragen hat. Dort hat es funktioniert.
mit multiplikator 0.0013888888888889 wurde die Zeit in Stunden ausgegeben.
Hier ein Auszug..
`publishFloat("energy/growatt/ttotal", glueFloat(node.getResponseBuffer(30), node.getResponseBuffer(31)));`

Hier aus der Mosbustabelle..
30 Time total H Work time total (high) 0.5S
31 Time total L Work time total (low) 0.5S

smartboart, 2022-11-29:
hab hier mal getestet.Wenn der InputRegister über die WebGui RWModbus abgerufen wird, wird ein Wert ausgelesen, welcher auch hochzählt.
Wenn ich im code die Input Register  30 und 31 einzeln als 16Bit abrufe wird 31 auch ordentlich hochgezählt. 30 bleibt statisch.
Das Ergebnis war in allen Tests aber nicht zielführend. Eine vernünftige Zeitausgabe bekomme ich nicht.

`    Protocol.InputRegisters[P305_OPERATING_TIME] = sGrowattModbusReg_t{30, 31, SIZE_32BIT, "OperatingTime", 0.5, SECONDS, true, false}; // #11
    Protocol.InputRegisters[P305_OPERATING_TIME_H] = sGrowattModbusReg_t{30, 0, SIZE_16BIT, "OperatingTimeH", 0.5, NONE, true, false}; // #12
    Protocol.InputRegisters[P305_OPERATING_TIME_L] = sGrowattModbusReg_t{31, 0, SIZE_16BIT, "OperatingTimeL", 0.5, NONE, true, false}; // #13
    Protocol.InputRegisters[P305_TEMPERATURE] = sGrowattModbusReg_t{32, 0, SIZE_16BIT, "Temperature", 0.1, TEMPERATURE, true, false}; // #14
    Protocol.InputRegisters[P305_FAULTCODE] = sGrowattModbusReg_t{40, 0, SIZE_16BIT, "FaultCode", 1, NONE, true, false}; // #15`

[image]

smartboart, 2022-11-29:
Update...
Also Inputregister 30 H und 31 L einzeln abgefragt wie oben mit 16Bit zählen beide hoch.

Denke die Lösung muss sein beides zu kombinieren...

crasu, 2023-01-03:
Unfortunatly I do not poses an inverter which supports protocol version 304. to test this? Did you have any success in fixing this. The issue seems to be a bit strange.

smartboart, 2023-01-04:
Helo crasu, no i was not able to find a solution.
I also tested with a external script by calculating with  the high and low byte values  sent by mqtt . But i never was able to get a correct operating time.
So my fix was to remove the operating time from the sketch completely..

closed 2023-01-04 by crasu