OpenInverterGateway/OpenInverterGateway #72: Error: Unknown Shine Stick(cant connect to Modbus)

dasLicht26, 2022-12-08:
Hi, unfortunately the stick (Shine-S) can not connect to the inverter (SPH4600). It does not detect the modbus.
I guess it could be because of the changed COM-port (HoldingRegister 30). 
But I can not reset this now without Stick, reset the inverter does not help either.

Debug give me this: "Error: Unknown Shine Stick" 
and i dont recive any Data.

Do you have an Idea?
Otherwise I have to buy a new stick and do without MQTT :(

dasLicht26, 2022-12-09:
maybe https://github.com/otti/Growatt_ShineWiFi-S/issues/60 is the solution

crasu, 2022-12-23:
@dasLicht26 Did it work before you changed the Com port?

crasu, 2022-12-30:
closed due to inactivity

closed 2022-12-30 by crasu

xeno27, 2023-04-05:
I have an SPH4600 and it works with the ShineWifi S Stick.