OpenInverterGateway/OpenInverterGateway #76: Issues reading data from MIC 600TL-X

Henrik99, 2022-12-18:
I flashed my Shine X stick with the firmware, but i can't get any real data from it.
It is connected to a MIC 600TL-X inverter.
The Debug output says its connected and reading data:
#0: Setup()
#1: ShineWiFi-X (USB) found
#2: ReadData() successful
...

But the values are constantly the same and wrong.
[image]
I have the Shine X stick with the CH340 serial converter.

What should i try to get it to work?

AditaNi, 2022-12-18:
Are you sure you have the right libraries?

Henrik99, 2022-12-18:
I am not 100% sure if I have the right one. I got some errors while compiling and had to try a couple different verisons of the modbus libary until it compiled.
What version are you using?

Henrik99, 2022-12-20:
@AditaNi you wehre ringt!
I used an incompatible version of the ModbusMaster libary.
I changed it to the 2.0.0 and now it works perfectly.

closed 2022-12-20 by Henrik99

AditaNi, 2022-12-20:
Thanks. I'm glad that i could help you.