OpenInverterGateway/OpenInverterGateway #79: Transmit the mqttclient Name to the mqtt broker

smartboart, 2022-12-26:
Hello in the mqtt Broker the clientname is not precent..
Maybe it is possible to add this information too...
Best regards..

crasu, 2022-12-30:
I shortly skimed through the mqtt spec - there seems to be no such thing as the clientname.

The client is identified by its id. Which is set here. Are you sure you are using the latest version?
https://github.com/otti/Growatt_ShineWiFi-S/blob/22c38a1ff2c1b4daf249b476d6e7ce28df7af69d/SRC/ShineWiFi-ModBus/ShineMqtt.cpp#L62

closed 2022-12-30 by crasu

PapaSpik3, 2023-01-04:
if you change getId().c_str() to HOSTNAME it displayes the right information also in the broker

smartboart, 2023-01-04:
Hello, thank you . The change to HOSTNAME fixed it...

Best reagards.