OpenInverterGateway/OpenInverterGateway #81: nodemcuv3

PapaSpik3, 2022-12-31:
Hello Otti,

Thanks for your firmware. I try to use it on an nodemcuv3 (LoLin but it dowsn't work (wrong Shine Stick). 
It use the CH340G chip and I defined in the platformio.ini ShineWifiX.
I also changed the board in the ini but doesnt helped.

I have a MIC 600TL-X from this year. 

Did I have to configure something on the MIC600 or is the nodemcuv3 not working?

Best regards

crasu, 2023-01-03:
Did you select the correct protocol version?

PapaSpik3, 2023-01-03:
I compiled with:

#define GROWATT_MODBUS_VERSION 305
#define GROWATT_MODBUS_VERSION 124
#define GROWATT_MODBUS_VERSION 120

All doesnt work. Web GUI looks different with all the versions but no data :-S

124 should be the right one (as documented). I changed the cable too. But also dont helped.

I get a lot of warnings during compilation is this normal?

PapaSpik3, 2023-01-04:
Now its working I didnt know what is different but what I did was:

On Linux Mint with PlatformIO:

1. Add user to dialout gfroup to use serial port (sudo usermod -a -G dialout $USER)
2. Install https://docs.platformio.org/en/latest/core/installation/udev-rules.html (warning that its missing)
3. Today PlatformIO ask to install C/C++ Extension Pack during compilation (didn't know if needed but I installed it)

I changed some code for easier use...

Config.h

// MQTT settings
#define MQTT_SERVER       "10.1.2.3"
#define MQTT_PORT         "1883"
#define MQTT_TOPIC        "energy/solar"
#define MQTT_USER         ""
#define MQTT_PASSWORD     ""

ShineWiFi-ModBus.ino

203-207:
    config->mqttserver = load_from_file(serverfile, MQTT_SERVER);
    config->mqttport = load_from_file(portfile, MQTT_PORT);
    config->mqtttopic = load_from_file(topicfile, MQTT_TOPIC);
    config->mqttuser = load_from_file(userfile, MQTT_USER);
    config->mqttpwd = load_from_file(secretfile, MQTT_PASSWORD);

283:     bool res = wm.autoConnect(HOSTNAME, APPassword); // password protected wificonfig ap

PapaSpik3, 2023-01-04:
thanks for the development!

closed 2023-01-04 by PapaSpik3

crasu, 2023-01-04:
Just flashed the latest release to my lolin32 connected to my MIC-600-TL - works!
{code}
#0: Setup()
#1: ShineWiFi-X (USB) found
#2: ReadData() NOT successful
#3: ReadData() successful
#4: ReadData() successful
#5: ReadData() successful
#6: ReadData() successful
#7: ReadData() successful
#8: ReadData() successful
#9: ReadData() successful
{code}

You are not supposed to get any compile warnings from the project itself. Just from the included libs.

PapaSpik3, 2023-01-04:
Yes thats true.
One missing point. I disabled line 66 (//#define AP_BUTTON_PRESSED ( analogRead(A0) < 50 ))
and set one digital input to low otherwise its starts all the time AP and connects to my local wifi (#define AP_BUTTON_PRESSED ( digitalRead(5) == LOW )).

best regards