OpenInverterGateway/OpenInverterGateway #83: Home assistant and mqtt

AditaNi, 2023-01-04:
Hi, 

since the last update regarding Mqtt, in Homeassistant i do not have any values anymore. 

What change was performed?

Regards
Adrian

crasu, 2023-01-04:
Some off the value names have changed when we added multi protocol support. I added a section in the readme on how to configure ha.

closed 2023-01-04 by crasu

AditaNi, 2023-01-04:
Hi,

it seems not to work.

TemplateError('ValueError: Template error: float got invalid input 'None' when rendering template '{{ float(state_attr('sensor.growatt_inverter_1', 'OutputPower')) }}' but no default was specified') while processing template 'Template("{{ float(state_attr('sensor.growatt_inverter_1', 'OutputPower')) }}")' for attribute '_attr_native_value' in entity 'sensor.growatt_inverter_ac_power'
7:05:53 PM – (ERROR) helpers/template_entity.py
Error while processing template: Template("{{ float(state_attr('sensor.growatt_inverter_1', 'OutputPower')) }}")
7:05:53 PM – (ERROR) helpers/template.py

AditaNi, 2023-01-04:
...and now:

TemplateError('ValueError: Template error: float got invalid input 'None' when rendering template '{{ float(state_attr('sensor.growatt_inverter', 'OutputPower')) }}' but no default was specified') while processing template 'Template("{{ float(state_attr('sensor.growatt_inverter', 'OutputPower')) }}")' for attribute '_attr_native_value' in entity 'sensor.growatt_inverter_ac_power'
7:18:08 PM – (ERROR) helpers/template_entity.py
Error while processing template: Template("{{ float(state_attr('sensor.growatt_inverter', 'OutputPower')) }}")
7:18:08 PM – (ERROR) helpers/template.py

crasu, 2023-01-04:
Please try again there was a typo in the readme file. You might need to wait until your inverter is delivering power.

AditaNi, 2023-01-04:
i will try with SOC...to see if it is working

AditaNi, 2023-01-04:
actually I cannot because I don't understand 100% your code for HA. I'm blocked!

crasu, 2023-01-04:
Which protocol version are u using

AditaNi, 2023-01-04:
for HA :
Home Assistant 2022.12.8
Frontend 20221213.1 - latest

your code i have arranged like:

mqtt:
  sensor:
    - name: "Growatt inverter"
      state_topic: "energy/solar/growatt/"
      unique_id: "Growatt.TotalGenerateEnergy"
      unit_of_measurement: "kWh"
      value_template: '{{ float(value_json.TotalGenerateEnergy) | round(1)  }}'
      json_attributes_topic: "energy/solar/growatt/"
      device_class: energy
      state_class: 'total_increasing'    
      last_reset_topic: 'energy/solar/growatt/'
      last_reset_value_template: '1970-01-01T00:00:00+00:00'
      payload_available: 1
      availability_mode: 'latest'
      availability_topic: "energy/solar/growatt/"
      availability_template: '{{ value_json.InverterStatus }}'

crasu, 2023-01-04:
Which protocol version do you use in config.h?

AditaNi, 2023-01-04:
aaa...
the default one : #define GROWATT_MODBUS_VERSION 124
My Growatt is SPH4000 one PH

crasu, 2023-01-04:
1. The protocol version should work
2. Have you fully restarted ha after changing the settings (ha sometimes does not correctly reconfigure without a full restart)
3. Can you see values coming in under dev tools -> states -> sensor.growatt_inverter
4. Can you see the mqtt messages on your mqtt broker?

AditaNi, 2023-01-04:
....but with this:
[image]

is working:
[image]

and on the website:
[image]

...so why?

crasu, 2023-01-04:
This issue is that your inverter state is set to 8

AditaNi, 2023-01-04:
...actually 👍 

The value of one of the two sensors (or both) was unknown.

The float filter cannot convert the value “unknown” to a number.

When it cannot convert the value, it can report a default value.

In all previous versions of Home Assistant, the default value was zero.

Starting in version 2021.10.0, you should specify the default value. If you don’t, and float cannot convert the value, it will be reported as a warning message and Home Assistant uses zero as a default value.

Starting in version 2021.12.0, you must specify the default value. If you don’t, and float cannot convert the value, it will be reported as an error message and the template will fail.

float(0)

That’s correct… default is the variable name. You aren’t supposed to copy the docs exactly, you’re suppose to supply a default.

source: https://community.home-assistant.io/t/template-warning-float-got-invalid-input-unknown-when-rendering-template/353198/12

AditaNi, 2023-01-04:
...but where should this change be done.
because the error says:
TemplateError('ValueError: Template error: float got invalid input 'None' when rendering template '{{float(state_attr('sensor.growatt_inverter', 'OutputPower'))}}' but no default was specified') while processing template 'Template("{{float(state_attr('sensor.growatt_inverter', 'OutputPower'))}}")' for attribute '_attr_native_value' in entity 'sensor.growatt_inverter_ac_power'

AditaNi, 2023-01-05:
.....so there is nothing which can help me further?

PR #85 (2023-01-04): Split readme and add homeassistant config
Closes #83