OpenInverterGateway/OpenInverterGateway #90: MQTT disconnected due to protocol error.

edstromanders, 2023-01-09:
Hi, I have compiled and tested the software using PlatformIO and an ESP8266 Nodemcu board with som problems.
I had the default config.h with MQTT enabled and managed to set parameters in the web page for Wifi and MQTT.  When trying to connect to MQTT it fails and tries to reconnect. 
**Serial monitor**
`*wm:[1] Connect to new AP [SUCCESS]
*wm:[1] Got IP Address:
*wm:[1] 192.168.0.144
*wm:[2] shutdownConfigPortal
*wm:[2] restoring usermode STA
*wm:[2] wifi status: WL_CONNECTED
*wm:[2] wifi mode: STA
*wm:[2] configportal closed
*wm:[1] config portal exiting
connected...yeey :)
MqttServer: 192.168.0.211
MqttPort: 1883
MqttTopic: /solar
MqttServer: 192.168.0.211
MqttUser:
MqttTopic: /solar
Attempting MQTT connection...failed, rc=-4 try again in 5 seconds
MqttServer: 192.168.0.211
MqttUser:
MqttTopic: /solar
Attempting MQTT connection...failed, rc=-4 try again in 5 seconds
MqttServer: 192.168.0.211
MqttUser:
`
The MQTT broker logs some error messages
`1673284498: Client <unknown> disconnected due to protocol error.
1673284518: New connection from 192.168.0.144:62507 on port 1883.
1673284518: Client <unknown> disconnected due to protocol error.
1673284538: New connection from 192.168.0.144:52895 on port 1883.
1673284538: Client <unknown> disconnected due to protocol error.
`
The MQTT broker is working well for other sensors and computers

I then tries to configure the software by changing Config.h
`// Setting this define to 0 will disable the MQTT functionality
#define MQTT_SUPPORTED 0`

but this results in strange compile errors
`   |-- ESP8266WiFi @ 1.0
Building in release mode
Compiling .pio\build\nodemcuv2\src\Growatt.cpp.o
Compiling .pio\build\nodemcuv2\src\ShineMqtt.cpp.o
Compiling .pio\build\nodemcuv2\src\ShineWiFi-ModBus.ino.cpp.o
src\Growatt.cpp:311:2: warning: #warning simulating the inverter [-Wcpp]
  311 | #warning simulating the inverter
      |  ^~~~~~~
src\Growatt.cpp:374:2: warning: #warning simulating the inverter [-Wcpp]
  374 | #warning simulating the inverter
      |  ^~~~~~~
src\Growatt.cpp: In member function 'void Growatt::begin(Stream&)':
src\Growatt.cpp:50:11: warning: unused variable 'res' [-Wunused-variable]
   50 |   uint8_t res;
      |           ^~~
src\Growatt.cpp: In member function 'void Growatt::CreateUIJson(char*)':
src\Growatt.cpp:334:15: warning: unused variable 'unitStr' [-Wunused-variable]
  334 |   const char* unitStr[] = {"", "W", "kWh", "V", "A", "s", "%", "Hz", "C"};
      |               ^~~~~~~
C:/Users/AE/Documents/PlatformIO/Projects/Growatt Wifi/src/ShineWiFi-ModBus.ino:58:1: error: 'ShineMqtt' does not name a type
   58 | ShineMqtt shineMqtt(espClient);
      | ^~~~~~~~~
C:/Users/AE/Documents/PlatformIO/Projects/Growatt Wifi/src/ShineWiFi-ModBus.ino:116:6: error: variable or field 'loadConfig' declared void
  116 | void WiFi_Reconnect()
      |      ^~~~~~~~~~
C:/Users/AE/Documents/PlatformIO/Projects/Growatt Wifi/src/ShineWiFi-ModBus.ino:116:17: error: 'MqttConfig' was not declared in this scope
  116 | void WiFi_Reconnect()
      |                 ^~~~~     
C:/Users/AE/Documents/PlatformIO/Projects/Growatt Wifi/src/ShineWiFi-ModBus.ino:116:29: error: 'config' was not declared in this scope
  116 | void WiFi_Reconnect()
      |                             ^
C:/Users/AE/Documents/PlatformIO/Projects/Growatt Wifi/src/ShineWiFi-ModBus.ino:117:6: error: variable or field 'saveConfig' declared void
  117 | {
      |      ^
C:/Users/AE/Documents/PlatformIO/Projects/Growatt Wifi/src/ShineWiFi-ModBus.ino:117:17: error: 'MqttConfig' was not declared in this scope
  117 | {
      |                 ^
C:/Users/AE/Documents/PlatformIO/Projects/Growatt Wifi/src/ShineWiFi-ModBus.ino:117:29: error: 'config' was not declared in this scope
  117 | {
      |    `
Unfortunately I cant debug any more. Please help.

Best regards, Anders

Chreutz, 2023-01-10:
I'm having the same issue, running `eclipse-mosquitto:latest` through docker.

edstromanders, 2023-01-11:
Thank you,

it works for me as well.

However, disabling MQTT in Config.h still give the same compile errors.

PapaSpik3, 2023-01-11:
here I made some enhancements to the code to fix the error. I had the same. 

I didnt know how to make a PR: https://github.com/otti/Growatt_ShineWiFi-S/issues/81#issuecomment-1370788580

crasu, 2023-01-12:
@edstromanders/@Chreutz  Could you try if #92 fixes your issue.

edstromanders, 2023-01-13:
Yes, the MQTT client connects without problems.
No, the compile errors remains if `#define MQTT_SUPPORTED 0`

crasu, 2023-01-13:
I can build with mqtt disabled. Have you tried deleting your pio directory completly and rebuilding

edstromanders, 2023-01-13:
I tried on the feauter-fix-client-.. branch again, deleted .pio folder and got the same error again.  One error might be on line 58  in .ino file `ShineMqtt shineMqtt(espClient);` should be #ifdef, but it didnt solve the other errors

PR #92 (2023-01-14): Fix client id generation
Client ID sometimes contained special characters which made the mqtt connection fail on some devices.

Fixes #90
closed 2023-01-14 by crasu