OpenInverterGateway/OpenInverterGateway #94: "Error: Unknown Shine Stick"

tonywbond, 2023-01-18:
Hi,
 I have built the code and deployed to my USB shine wifi logger (see image)
 When the server starts I see "Error: Unknown Shine Stick" (see image) in the log and the webserver only displays 0.
 I assume there is some issue with either the chips on the logger or with the build I created.
 I am a complete newbie on building using the arduino IDE, so wondered if there is a release build that I can test with ?
Alternatively are there any diagnostics I run on the logger itself to understand why this error occurs (note using the RW Modbus fails) ?
Any help would be much appreciated
Thanks
Tony
[image]
[image]

crasu, 2023-01-18:
Which inverter are you using?

Try this in File "config.h"
#define ENABLE_DEBUG_OUTPUT 0

An reboot the inverter afterwards.

prkena, 2023-01-19:
Hi, everybody. I had the same problem on the SPH 10K tl3 bh-up. Crasu's advice solved it. Thanks!

PR #96 (2023-01-19): Disable debug output 
Seems to cause issues with stick detection

closes #94
closed 2023-01-19 by crasu