OpenInverterGateway/OpenInverterGateway #99: Not found: /firmware

AditaNi, 2023-01-28:
Hi,

good work until now, but:
- i have used a precompiled .bin file and I wanted to upload again the firmware and....: Not found: /firmware

what it can be done now?

crasu, 2023-01-29:
I had to disable the http firmware update for the precompiled, as otherwise the password would be visible in the github repo. In the future I want to make the password and parameters configurable via webinterface. But this will take some time.

crasu, 2023-01-29:
I added a note that ota is currently not included in the precompiled version.

closed 2023-01-29 by crasu

homuser2003, 2023-02-18:
I don´t really get how to compile my own image. I did open platformio, changed the 0 to 1 for #define UPDATE_SUPPORTED 1 (by the way does # as first letter not mean uncommented? and then compiled a bin. When I flash that however, the stick starts for a moment but moves into constant red flashing led soon...

The problem definitely is in front of the PC but what am I missing? I´d love to be able to update OTA. Pulling and putting the stick back in that position of the inverter sucks!

Thanks

homeuser