OpenInverterGateway/OpenInverterGateway #100: Error with JSON and growatt.cpp when compile with arduino 1.8 for esp8266

dbase25, 2023-01-29:
I try to compile for a esp8266 but i think i have a arduino json problem.

i have try a lot of version 5 & 6, but i get a error:

growatt.cpp:
"reinterpret_cast<T>(pgm_read_ptr(p));"

or similiar in *\libraries\ArduinoJson\src/ArduinoJson/*

which version is the right for the json lib or i have a huge mistake?

thanks and regards
dave

crasu, 2023-01-29:
I would advise you not to use the arduino ide to compile this project. However this is the version I use to compile with platform io. Maybe its working for you with the arduino IDE as well.
https://github.com/bblanchon/ArduinoJson#67b6797b6d19e944b01213926872f955f4b9d54d

dbase25, 2023-01-29:
Thanks for your fast answer and help.  i have make a quick check with arduino and this V6.2 and doesnt work. i test it tomorrow with platform io and take short respons.

thanks and best regards dave

dbase25, 2023-01-30:
So i have tested with platformio and i works with compile and and upload (start wifi etc).
now i try it with modbus.
thanks and regards
dave