OpenInverterGateway/OpenInverterGateway #102: PlatformIo error

mczeus, 2023-02-04:
Hello, 
I have a ShineX stick, I flashed it with the finished bin. But get displayed in the log that it is not recognized. I then tried to create one myself with platformIo, but always get this error message
"call of overloaded 'String(uint64_t)' is ambiguous" and 
"'uint16_t crc16_update(uint16_t, uint8_t)' defined but not used [-Wunused-function]"
What am I doing wrong?

crasu, 2023-02-04:
Have you fully resetted the inverter before trying the stick? The default settings should work with a mic 600 tl-x.

Which config are you using to compile the platform io project? 

Could you paste a bit more context for the error message? It looks like it is struggeling to compile the modbus lib. Is this a fresh platform io installation? Otherwise I would advise you to run `pio system prune`.

mczeus, 2023-02-05:
Sorry stick is now recognized shame on my head. 
I've done all updates from PlatformIo, weird but it's no longer needed.

closed 2023-02-05 by mczeus