OpenInverterGateway/OpenInverterGateway #111: SPH QA-question

skinkie, 2023-02-09:
Via https://github.com/johanmeijer/grott/issues/167#issuecomment-1424947302 I was pointed at this cool project. I have a question. I am currently running RS485 for an SPH10K. I receive the data using https://github.com/mbs38/spicierModbus2mqtt over a Waveshare DIN RS485 to Ethernet device.

I have noticed that the data that is sent over RS485 is consitently _very wrong_, even when I probe it with Qmodbus for development purposes. My reference documentation is "Growatt Inverter Modbus RTU Protocol v1.20".

[image]

Consistently all "P" and "S" (phase two and three) values are wrong. Line voltages are not ~230V but instead the ~400V between phase values. I wonder if any of you while developming this code have noticed these issues as well.

I have also a 6000UE which gave me also a lot of 'reverse engineering because the documentation did not line up' but for this newer device I actually wondered if the 'unbalanced output' (which does not work at all in my perspective) is broken because of these "measurement bugs".

crasu, 2023-02-12:
@skinkie I do not have an SPH inverter so I am unable to help you - sorry.

crasu, 2023-02-12:
Closing as not a  bug.

closed 2023-02-12 by crasu

skinkie, 2023-02-12:
@crasu the code within your repository suggest someone has it. 
https://github.com/otti/Growatt_ShineWiFi-S/blob/master/SRC/ShineWiFi-ModBus/Growatt124.cpp

Chreutz, 2023-02-16:
> @crasu the code within your repository suggest someone has it. https://github.com/otti/Growatt_ShineWiFi-S/blob/master/SRC/ShineWiFi-ModBus/Growatt124.cpp

What part of the code? I have a SPH10K and we can cross-reference if you want to

Chreutz, 2023-02-16:
Just looked a little bit at it. I think I will try to work out a python script to read the registers close as close as simultaneous as possible and reverse engineer from there. It might be that the values are 16 bit and not 32 bit as the doc says.

skinkie, 2023-02-16:
@Chreutz the code here only takes very few registers, seems to be only the _totals_. Can you check any S and T values? (1017, 1019, 1025, 1027, 1033, 1035)

[image]
[image]

Chreutz, 2023-02-16:
Yes, those registers return weird values.

skinkie, 2023-02-16:
> Yes, those registers return weird values.

I think there are more bugs at the SPH10K. For example, the device claims to have unbalanced output. Any other vendor explains that as: if you consume more on phase 1, we can produce that output on phase 1, preventing backflow to the grid. I notice that the SPH10K does nothing in this direction, it acts just like a stupid non-hybrid inverter: output is equal on all phases. Given that the values (but also things like line voltage) values do not agree with the documentation at all, I wonder if this device is buggy and therefore is not able to operate like it is expected to do.

Chreutz, 2023-02-16:
I have not looked into that aspect as my grid consumption metering is based on a sum of all three phases, so for me, it doesn't really matter. But it looks interesting. And I agree that there are many places where the documentation doesn't seem to be correct. I would love to talk to an expert, but it doesn't seem like Growatt is very talkative.

skinkie, 2023-02-16:
> I have not looked into that aspect as my grid consumption metering is based on a sum of all three phases, so for me, it doesn't really matter. 

Economically it does not matter for me either. From grid perspective it might matter, especially since the values over here on a sunny day already exceed 245V, having some steering happening would help to balance the grid.

> But it looks interesting. And I agree that there are many places where the documentation doesn't seem to be correct. I would love to talk to an expert, but it doesn't seem like Growatt is very talkative.

I have some contacts now. Even did the calculations for them. Then they requested to add a ShineLan to the system, I might want to add a grott-proxy so I can see what it really exports, but on the growatt webportal the data shown is less than I had with a single phase inverter from three years ago...

Chreutz, 2023-02-16:
Maybe use an ESP32 (through the RS485 port) as well as the ShineWifi/Lan at the same time, so you can cross check.

I'm still in the beginning of getting to know the system. Only set op this firmware a month ago, and haven't had too much time to work with it yet, beyond basic sensors for home assistant.

skinkie, 2023-02-16:
> Maybe use an ESP32 (through the RS485 port) as well as the ShineWifi/Lan at the same time, so you can cross check.

I have RS485 connected (spicierModbus2MQTT), this is the reason I figured out the problems in the first place.

Chreutz, 2023-02-16:
Alright. I have just flashed it to my Wifi-X, so I have no growatt server data as reference. But grott should work very nicely to chech with the "original" data