OpenInverterGateway/OpenInverterGateway #114: Home Assistant Integration

AditaNi, 2023-02-12:
Hi,

Where is the isuse hier? Bedcause is not working anymore in homeassistant.

Error: 
RELATED
This entity is no longer being provided by the rest integration. If the entity is no longer in use, delete it in settings.

mqtt:
  sensor:
    - name: "SOC"
      state_topic: "energy/solar"
      unique_id: "SOC"
      unit_of_measurement: "%"
      value_template: '{{ float(value_json.SOC)| round(1) }}'

crasu, 2023-02-12:
Can you still see the SOC value in the mqtt message? From the inverter?

The error message is a bit odd because it is talking about rest.

AditaNi, 2023-02-12:
Yes, it is there.
But that it is only an example, because nothing it is working for me.
Yes indeed i do not know from wre the error with 'rest' it is camming.

crasu, 2023-02-18:
This really seems to be more a HA issue than an issue with this firmware.

closed 2023-02-18 by crasu