OpenInverterGateway/OpenInverterGateway #115: Possible to use 2 sticks via USB HUB?

Plawasan, 2023-02-16:
I have the stick up and running, feeding my data into HA via mqtt - all works perfectly well. Question per the title - is it possible to actually have both the original unmodified stick and the custom FW at the same time one sending data to HA and the other still communicating with the Growatt server both connected to the same inverter via a USB hub?

I've tried using a USB 3.0 Hub, didn't seem to work with either stick connected via the hub (on its own) and obviously neither with both connected. Is this supposed to work in principle or is the communication between the stick and the inverter something that cannot run via an USB hub (alone or in parallel with another stick)?

Thanks!

Chreutz, 2023-02-17:
If you want this, it is possible to do it with a ESP32 flashed with this firmware, connected to the RS485 port, and then the growatt device connected to usb.

Is there a reason you would prefer two devices on the USB port?

Plawasan, 2023-02-17:
The only reason was that it was the only solution I could come up with myself...

I didn't consider the ESP32 route.. I do have a MAX3485 RS485 module and an a ESP32 sitting in my drawer so if I understand the principle correctly I can simply flash the wifi-x firmware onto the ESP32, hook it up to the RS485 and connect that to the RJ45 port on the inverter and I should get the same result (i.e. local mqtt updates) as running the stick directly via USB? That would in fact be ideal..

Any resources I can consult for the pinout of the RS485 port? I assume I just need to find A/B and Ground...

crasu, 2023-02-17:
Have a look in your inverters manual. Afaik it is pin 3/4 of the signal connector. You might have to use a different protocol/data rate. I have never tried if this really works in parallel with the original firmware.

Plawasan, 2023-02-17:
Won't be as easy as I thought it would... I flashed the nodemcu32 fw on a ESP32 and it goes into a bootloop..

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

Again.. not even sure if this is supposed to work of if the FW is exclusively meant for the shinestick...

crasu, 2023-02-17:
It works with an esp32 - I have used it with a nodemcu32s via serial and I am currently using it with a lolin32

Plawasan, 2023-02-17:
Yes, this one - https://github.com/otti/Growatt_ShineWiFi-S/releases/download/release%2Fv1.0.0/firmware-Snodemcu-32s.bin

I tried both using esptool and esphome web, same result on two different ESP32 boards..

crasu, 2023-02-17:
I justed flashed the latest master via platform io to an nodemcu32-s. I am not 100% sure if the precompiled image works for nodemcu32-s. I only tried the lolin32 image.

Plawasan, 2023-02-17:
I tried the same.. downloaded the .source zip, pip install platformio, copied config.h (wiithout any changes), built the firmware:

```
root@Elitebook:/home/martin/Downloads/Growatt_ShineWiFi-S-master# platformio run -e nodemcu-32s
...
Compiling .pio/build/nodemcu-32s/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/nodemcu-32s/libFrameworkArduino.a
Indexing .pio/build/nodemcu-32s/libFrameworkArduino.a
Linking .pio/build/nodemcu-32s/firmware.elf
Retrieving maximum program size .pio/build/nodemcu-32s/firmware.elf
Checking size .pio/build/nodemcu-32s/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  19.2% (used 62996 bytes from 327680 bytes)
Flash: [========  ]  80.1% (used 1050533 bytes from 1310720 bytes)
Building .pio/build/nodemcu-32s/firmware.bin
esptool.py v4.4
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
```

Flashed it:

```
root@Elitebook:/home/martin/Downloads/Growatt_ShineWiFi-S-master# esptool.py --port /dev/ttyUSB1 write_flash --flash_mode dio --flash_size 4MB 0x0 firmware.bin 
esptool.py v4.5
Serial port /dev/ttyUSB1
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:3a:f2:8f:7a:78
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00101fff...
Compressed 1056304 bytes to 661973...
Wrote 1056304 bytes (661973 compressed) at 0x00000000 in 58.7 seconds (effective 144.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

yet I'm still in a bootloop... (on two different boards). If you see anything I'm doing wrong, let me know.. or alternatively if you can share the bin that works for you, I can at least eliminate one variable..

crasu, 2023-02-17:
Have you tried to flash with platform io:
`platformio run --target upload --environment nodemcu-32s `

crasu, 2023-02-17:
Ok the issue seems to be that pio does not package the entire firmware (bootloader, partitions, .... into the bin file).

You have to flash the image as described here:
https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/flashing-firmware.html

This could be a solution to package everything into the bin file:
https://community.platformio.org/t/esp32-how-to-create-one-solid-firmware-package-that-will-include-bootloader-etc-in-a-single-file/24379

Plawasan, 2023-02-18:
Progress!

I merged the .bins from platformio build:

`esptool.py --chip ESP32 merge_bin -o merged-flash.bin --flash_mode dio --flash_size 4MB 0x1000 .pio/build/nodemcu-32s/bootloader.bin 0x8000 .pio/build/nodemcu-32s/partitions.bin 0x10000 .pio/build/nodemcu-32s/firmware.bin`

Now the ESP boots, I can set up my wifi, configure MQTT, it loads the homepage but I have a suspicion the partition setup might not be right - I don't see a graph (even with no values) when I go to log or AP Setup I get this:

```
File Not Found

URI: /debug
Method: GET
Arguments: 0
```

I'll try and fiddle with it, in case I succeed - what GPIO should the tx/rx pins from the modbus module be connected to?

crasu, 2023-02-18:
You have enable webdebug:
`#define ENABLE_WEB_DEBUG 1` to see something under /debug

And you probably want to disable serial debugging. The serial converter is hooked up at the normal tx/rx lines.

Plawasan, 2023-02-18:
That's both in the correct state in Config.h:

```
cat SRC/ShineWiFi-ModBus/Config.h | grep DEBUG
#define ENABLE_DEBUG_OUTPUT 0
#define ENABLE_WEB_DEBUG 1
```

crasu, 2023-02-18:
This is extremly strange. I would guess that you still have the wrong version on your stick and try to do a clean compile after erasing the stick.

Plawasan, 2023-02-18:
I'm at loss here.. I tried erasing the ESP with esptool, I tried flashing esphome on it, still the same.. so I deleted my entire build directory, got a completely different ESP32 board and compiled a new build with platformio:

`# pio run --environment nodemcu-32s`

then I merged the .bins:

```
 cd .pio/build/nodemcu-32s/
esptool.py --chip ESP32 merge_bin -o merged-flash.bin --flash_mode dio --flash_size 4MB 0x1000 bootloader.bin 0x8000 partitions.bin 0x10000 firmware.bin
```

flashed the merged_bin

`esptool.py write_flash 0x0 merged-flash.bin`

and ti still behaves exactly the same - I can configure my wifi and mqtt but I cannot open any of the links on the homepage.. all lead to a 404.

crasu, 2023-02-19:
You can flash directly via platform io:

Wipe the entire flash via es tool:
`esptool.py --chip esp32 --port /dev/ttyUSB0 erase_flash`

Clean build:
`pio run -e lolin32 --target upload `

Upload via pio:
`pio run -e lolin32 --target upload --upload-port /dev/ttyUSB0`

Plawasan, 2023-02-19:
I've tried that as well, still no luck... I really appreciate the support, normally I would have probably given up by now but I REALLY want to be able to run both at the same time..

Any chance you can share the binary that works for you?

Plawasan, 2023-02-20:
So while I was struggling with SW, I decided to try and sort out the HW to make sure the whole thing is actually feasible. 

I have flashed ESPHome on one of my esp32, connected tx/rx to the max3485, used pins 1 and 5 on RS485-3 port of the inverter, configured the growatt sensors and voila, I'm getting local data via RS485 while the stock usb stick is still sending data directly to Growatt.

However.. the only meaningful data I can get via esphome is PV power which is arguably the only thing i really need "real time", nothing else useful really works.

So now I'm REALLY motivated to get the shinestick fw working on this setup..

@crasu, what exact board are you using? I'm ok waiting a couple weeks for it to get here if it means I'll have the full data set available locally. And once again thanks for the patience, very much appreciated!

Plawasan, 2023-02-20:
I've realized I can connect to the serial port console to see what's actually going on:

ESP boots into the wifi/mqtt setup screen
```
wm:[2] <- HTTP Root 
*wm:[2] Portal Timeout In 293 seconds
*wm:[2] <- HTTP Param 
```
Hit save on MQTT parameters:
```
*wm:[2] <- HTTP Param save  
*wm:[2] Parameters 
*wm:[2] -------------------- 
*wm:[2] server: 192.168.1.10
*wm:[2] port: 1883
*wm:[2] topic: energy/growatt
*wm:[2] username: espgrowatt
*wm:[2] password: <redacted>
*wm:[2] -------------------- 
```
crash & reboot:
```
[CALLBACK] saveParamCallback fired
*wm:[2] [EVENT] WIFI_REASON:  8
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
1�[  8250][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():192]: (-4396) PEM - PEM string is not as expected : BASE64 - Invalid character in input
[  8253][E][WiFiClientSecure.cpp:135] connect(): start_ssl_client: -4396
*wm:[2] Starting Config Portal 
*wm:[2] AccessPoint set password is VALID 
*wm:[2] Enabling AP 
*wm:[1] StartAP with SSID:  GrowattConfig
*wm:[1] AP IP address: 192.168.4.1
*wm:[1] Starting Web Portal 
*wm:[2] HTTP server started 
```
If in Unplug the ESP it boots up into the home page (where the graph is supposed to be) but then opening e.g. the log link gives me the following:

```
*wm:[2] Portal Timeout In 300 seconds
[ 26255][E][WebServer.cpp:649] _handleRequest(): request handler not found
[ 28518][E][WebServer.cpp:649] _handleRequest(): request handler not found
*wm:[2] Portal Timeout In 298 seconds
[ 30253][E][WebServer.cpp:649] _handleRequest(): request handler not found
```

crasu, 2023-02-20:
I justed checked I used an esp8266 nodemcu. I have found a couple of photos how I connected it to the inverter. You might be able to use as a reference:
[serial.zip](https://github.com/otti/Growatt_ShineWiFi-S/files/10787303/serial.zip)

crasu, 2023-02-20:
This message is strange. Uncomment the cert to disable ssl support.
```
���1�[  8250][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():192]: (-4396) PEM - PEM string is not as expected : BASE64 - Invalid character in input
[  8253][E][WiFiClientSecure.cpp:135] connect(): start_ssl_client: -4396
```

Plawasan, 2023-02-20:
> This message is strange. Uncomment the cert to disable ssl support.

That got me rid of that error but not much else... I've ordered an esp8266, will try my luck with that.

Plawasan, 2023-02-22:
I'm losing all hope.. I've now tried with two different ESP8266 boards, still seeing pretty much the same behavior - setup screens comes on, let's me enter wifi details, the boards connects to my wifi (but still broadcasts the growconfig SSID), I can configure the mqtt params but the stick always reboots to the setup screen.. If I unplug it from power, then it boots to the homepage with the links to logs etc, but none of those work.

To be clear on what I'm trying to achieve - run the original FW on the stick connected to the USB port of the inverter and at the same time run this FW on a ESP32/ESP8266 connected via TX/RX to a MAX3485 RS485 to TTL converter connected to the RS-485 port of the inverter. I have this HW setup in place now running ESPhome and getting meaningful data (albeit only for PV Power).

Here's a possibly stupid idea - since I have this FW successfully installed on a second Growatt stick, can I feed it data via modbus (and the RS3485) via its TX/RX pins instead of its USB port?

Plawasan, 2023-02-22:
I have pivoted a bit.. and moved the entire setup to esphome. I have shamelessly stolen (i mean referenced...) the 124 protocol definition from this project and created a yaml that has all the same data using modbus platform. Right now I have it all set up and it seems to be working exactly as expected, numbers match up with growatt servers but let me run through an entire day with PV production to confirm it's all solid. 

I will then post the link to the yaml and will close this issue.

Plawasan, 2023-02-24:
For the record if anyone has the same inverter, here's what I ended up using as my setup... works in parallel with the stock USB stick:

https://community.home-assistant.io/t/esphome-modbus-growatt-shinewifi-s/369171/135?u=plawa

I would still want to understand Why I couldn't get this project to work but given that I found an alternative, I'm going to move on...

closed 2023-02-24 by Plawasan