OpenInverterGateway/OpenInverterGateway #116: OTA

homuser2003, 2023-02-18:
Sorry for double posting, added the same to the closed previous post but most likely no on really monitors closed posts:

I don´t really get how to compile my own image. I did open platformio, changed the 0 to 1 for #define UPDATE_SUPPORTED 1 (by the way does # as first letter not mean uncommented?) and then compiled a bin with no errors. When I flash that however, the stick starts for a moment but moves into constant red flashing led soon...

The problem definitely is in front of the PC but what am I missing? I´d love to be able to update OTA. Pulling and putting the stick back in that position of the inverter sucks!

Thanks

homeuser

crasu, 2023-02-18:
In c # is a preprocessor instruction. Have a look here:
https://en.wikipedia.org/wiki/C_preprocessor

Can you access the webinterface and get the debug output from /debug there?

Without the debug output it is impossible to tell what your issue is.

homuser2003, 2023-02-19:
Thanks for your comment,

no, the WebIF was not reachable and the above is beyond me. I´m far from beeing any kind of programmer. I meanwhile went back to your precompiled firmware and soldered a reset button to the stick. I´ll reflash once you found the time to do the changes for the password although I did not really get WHICH password you were concerned about...

homeuser2003