OpenInverterGateway/OpenInverterGateway #124: Firmware-Image working on the original Wifi-ShineX-Stick, but not in ESP32/Lolin32-Board

richardmayer, 2023-03-01:
# What happened?
Hello,
I've been using your precompiled Firmware-Image on the original ShineX-Stick for a few weeks now and am very thankful for your work.
As the original stick was pretty slow from time to time delivering the json-Data (more than 3 seconds), I decided to give the ESP32-Build a try.

# Stick type

What Stick type are you using?
I copied the firmware to two different ESP32 boards (Wroom32, Lolin32) and sucessfully configured Wifi-Access.
But the ESP32 doesn't seem to be able to connect to the Growatt Inverter via USB. Only the Wifi-Configuration page is shown, when accessing the IP-Adress of the ESP32 plugged in the USB-Port of the Growatt - no inverter data.

Now my question: 
Is the Lolin32-Build supposed to be able to communicate with the Growatt Inverter over the Built-in MicroUSB-Port?

# Log output

I have no custom build, so there seems to be no IP/debug-output.

crasu, 2023-03-01:
> Now my question: Is the Lolin32-Build supposed to be able to communicate with the Growatt Inverter over the Built-in MicroUSB-Port?
Yes - Make sure you have an CH340 chip. 

I had this issue with a none precompiled build and was able to fix it by upgrading platform io. I suspect it could be related to the flash boot loader. In my case the stick was unable to restore the wifi settings from flash after reboot.

How did you flash your lolin32?

richardmayer, 2023-03-01:
Hi, 

I don't think, it's a flash problem.
I used ESPHome-Flasher in Mac OS zu directly flash the bin-file.
The ESP32 remebers the Network-Config just fine.
But: 
Directly after pluggin in, the Webinterface will show this empty overview:
[image]

And after a few minutes, it will only show the Wifi-Manager Site (Status "Connected").
[image]

So my interpretation is that the ModBUS-Data cannot be retrieved for some reason.

crasu, 2023-03-01:
Ok good to hear that it is not the wifi config issue. Yes its normal that it will revert to the wifi manager after 5 minutes if it cannot connect.

Do you get message under http://<ip>/debug when the wifi manger is not active?

richardmayer, 2023-03-01:
Hi, 

I will check tomorrow, once the sun is up again.
For today, the inverter has shut down...:)

Best Regards

Richard

richardmayer, 2023-03-02:
So this is the /debug Output, before the WifiManager starts:

#0: Setup()
#1: ShineWiFi-X (USB) found
#2: MQTT Connect failed

(I'm not using MQTT)

crasu, 2023-03-04:
The mqtt error should not be an issue. Could you wait a bit longer? After about 30s you should get a either a `ReadData() successful` or `ReadData() unsuccessful` message.

crasu, 2023-03-05:
Just installed the release on my lolin32 there is an issue with the precompiled binary.

As a workaround you can pull A0/PIN17 to LOW with a wire. Otherwise the lolin32 will restart after 30s.

richardmayer, 2023-03-05:
Hi,
I tried your workaround, but I'm not sure, if this worked.
I put a 10k resistor between PIN17 and GND (see attached photo), but still, the WifiConfiguration is shown, when accessing the ESP32. The Inverter is already in Standby-Mode today, so I'm not sure if this is expected behaviour.
Will try again tomorrow.

I also attach the debug-log, the connection via USB seems to be sucessful (ReadData successful).

Thanks for your help.

Richard

[image]

[image]

crasu, 2023-03-05:
The issue is this line from the code. This will bring the lolin32 into config 30s after starting to read data from the inverter:
```
// On Growatt devices the Pushbutton is connected to the ADC Pin (A0) and pulled to GND when pressed
#define AP_BUTTON_PRESSED ( analogRead(A0) < 50 )
// For generic EPS8622 boards you can use a digital pin of your choice | IMPORTANT! Change the value or your device will not work properly or connect A0 to ground
// #define AP_BUTTON_PRESSED ( digitalRead(1) == HIGH )
```
I only read the comment but the comment contradicts the code. Could you try to pull the thing up instead of down.

This needs a fix, but its not that easy as I have move the config from compile time to runtime.

closed 2023-03-16 by crasu