OpenInverterGateway/OpenInverterGateway #125: MQTT not Working

free118, 2023-03-07:
# What happened?
Device is not connecting to MQTT Server. 
No entry in log. 

I run the code on a Wemos D1 mini R1. 
To get it running stable w/o ending in the WifiManager loop I've added a pullup restistor to A0 (A0 to 3.3V)
After that Converter UI appear and show values. 
MQTT_Supported is set in code to 1 

I've set and erase settings more than once. 
Reprogrammed device a couple of time. 

No MQTT connection try is showing up in log.

Once with a typo in the config I saw MQTT showing up the log. (Due to typo connection was rejected)
But after that never again. 
Looks like that setting are not sufficent stored or set. 

Could you provide a Home Assistant sonfig for a JSON Rest API sensor (RESTFul)?

Also tell us, what did you expect to happen? Tell us what you see!

# Stick type

What Stick type are you using?
- Wemos D1 mini R1      

# Log output
Please copy and paste any relevant log output from /debug and the serial console (You have to compile the firmware with the relevant settings in Config.h to get the output).

```
Insert logs here
```
#0: Setup()
#1: ShineWiFi-X (USB) found
#2: ReadData() successful
#3: ReadData() successful
#4: ReadData() successful
#5: ReadData() successful
#6: ReadData() successful

xdubx, 2023-03-13:
Double check your ip adress and the correct user/pw credentials. I replaced my rest to mqtt.

```
 - platform: rest
   resource: http://192.168.178.XXX/status
   name: growatt
   scan_interval: 3600 # replace to your pull rate
   json_attributes:
     - InputPower
     - PV1Voltage
     - PV1InputCurrent
     - PV1InputPower
     - OutputPower
     - GridFrequency
     - PVEnergyTotal
     - InverterTemperature
     - TodayGenerateEnergy
     - TotalGenerateEnergy
   value_template: '{{ value_json["InverterStatus"] }}'
  

- platform: template
  sensors:
    growatt_inputpower:
      unique_id: '0a7476cc-d6c1-40ba-8ae1-606518c3497f'
      value_template: "{{ state_attr('sensor.growatt', 'InputPower') }}"
      device_class: "power"
      unit_of_measurement: "W"
      friendly_name: "Input Power"
    growatt_pv1voltage:
      unique_id: 'd56f851d-b196-4aed-9965-6c31a53cd18a'
      value_template: "{{ state_attr('sensor.growatt', 'PV1Voltage') }}"
      unit_of_measurement: "V"
      device_class: "voltage"
    growatt_pv1inputcurrent:
      unique_id: 'b0e6c0b1-2f3d-49ff-86ee-44f8c04a0e10'
      value_template: "{{ state_attr('sensor.growatt','PV1InputCurrent') }}"
      unit_of_measurement: "A"
      device_class: "Current"
    growatt_pv1inputpower:
      unique_id: 'e363cfdd-398c-4602-87b2-894d0b4f447a'
      value_template: "{{ state_attr('sensor.growatt','PV1InputPower') }}"
      unit_of_measurement: "W"
      device_class: "power"
    growatt_outputpower:
      unique_id: '989ef68b-ec2d-40e0-80ab-049b7854111c'
      value_template: "{{ state_attr('sensor.growatt', 'OutputPower') }}"
      device_class: "energy"
      unit_of_measurement: "W"
    growatt_gridfrequency:
      unique_id: 'ae43485e-ca7e-46cb-96ff-04ee674cac3f'
      value_template: "{{ state_attr('sensor.growatt', 'GridFrequency') }}"
      device_class: "frequency"
      unit_of_measurement: "Hz"
    growatt_pvenergytotal:
      unique_id: '5574aaeb-12c0-4646-b656-7599a28e39d3'
      value_template: "{{ state_attr('sensor.growatt', 'PVEnergyTotal') }}"
      unit_of_measurement: "kWh"
      device_class: "energy"
    growatt_invertertemperature:
      unique_id: 'd4af77b7-ff09-4e02-b9d6-f592e1082f59'
      value_template: "{{ state_attr('sensor.growatt', 'InverterTemperature') }}"
      unit_of_measurement: C
      device_class: temperature
    growatt_todaygenerateenergy:
      unique_id: '93822e6d-44fa-4185-97f9-f3500da28655'
      value_template: "{{ state_attr('sensor.growatt', 'TodayGenerateEnergy') }}"
      device_class: "energy"
      unit_of_measurement: "kWh"
    growatt_totalgenerateenergy:
      unique_id: '4968dd1b-a95d-48bb-aad1-83088fd64475'
      value_template: "{{ state_attr('sensor.growatt', 'TotalGenerateEnergy') }}"
      device_class: "energy"
      unit_of_measurement: "kWh"
```

i recommend to remove the rest api log from the logbook its spaming the entries if the device is offline.

crasu, 2023-03-16:
Are you sure its not sending any messages to the mqtt broker? There is no log message on success.

closed 2023-03-17 by free118