OpenInverterGateway/OpenInverterGateway #128: EVCC einbinden Probleme

Sommerwl, 2023-03-12:
# What happened?

Ich möchte den ShineWiFi-S gerne per Modbus einbinden
Bekomme es nicht hin, benötige bitte Hilfe.

Alle Werte werden in der WEB-Oberfläche und in der JSON Datei angezeigt.

Habe ehrlich gesagt keine Ahnung.

# Stick type

What Stick type are you using?

- Shine S
  

# Log output
Please copy and paste any relevant log output from /debug and the serial console (You have to compile the firmware with the relevant settings in Config.h to get the output).

```
Insert logs here
```

Sommerwl, 2023-03-16:
Anbindung hat per Json geklappt.
Bekomme bei MQTT immer eine Fehlermeldung.
Wir kann ich MQTT aktivieren verwenden?

#0: Setup()
#1: Error: Unknown Shine Stick
#2: MQTT Connect failed
#3: MQTT Connect failed
#4: MQTT Connect failed
#5: MQTT Connect failed
#6: MQTT Connect failed
#7: MQTT Connect failed
#8: MQTT Connect failed
#9: MQTT Connect failed
#10: MQTT Connect failed
#11: MQTT Connect failed
#12: MQTT Connect failed
#13: ShineWiFi-S (Serial) found
#14: MQTT Connect failed
#15: ReadData() successful
#16: MQTT Connect failed
#17: ReadData() successful
#18: MQTT Connect failed
#19: ReadData() successful
#20: MQTT Connect failed
#21: ReadData() successful
#22: MQTT Connect failed
#23: ReadData() successful
#24: MQTT Connect failed
#25: ReadData() successful
#26: MQTT Connect failed
#27: ReadData() successful
#28: MQTT Connect failed
#29: ReadData() successful
#30: MQTT Connect failed
#31: ReadData() successful
#32: MQTT Connect failed
#33: ReadData() successful
#34: MQTT Connect failed
#35: ReadData() successful
#36: MQTT Connect failed
#37: ReadData() successful

crasu, 2023-03-21:
It looks like your stick is connecting successfully. You have to enter the correct mqtt settings for your broker on the config page (via start config access point).

crasu, 2023-04-01:
Closing this due to inactivity

closed 2023-04-01 by crasu