OpenInverterGateway/OpenInverterGateway #131: RS485 instead of USB ?

Keucar, 2023-03-19:
# What happened?
it is possible to use to modify the Code to use RS485 instead of USB ?
GROWATT SPH 10000 TL3 BH UP has 3 RS485 Ports and Dual Use (Cloud+Local) will be possible.

# Stick type
XTVTX ESP32 + SP3485
SP3485 is connected to D16, D17 for Serial and D5 for RTS

Sorry but i don`t understand the source code to 100% 

Problem1:
to test i flash the Lolin32 binary
First Time after ESP boot i see the webinterface "Growatt Inverter"
But none of the links ist working.
File Not Found

URI: /firmware (also all other)
Method: GET
Arguments: 0

The ESP mainpage fall back to the wifimanager page

When i reboot the ESP, i got for a few seconds access to the "status" subpage, when i refresh, site is empty.
When i call the mainpage.. fallback to wifi manager

With other .bin like Nukihub, Tasmota etc. are no issues. 
Can flash just the bin @0x10000 and working fine.

Maybe the Lolin use a different partition table and/or bootloader ?

crasu, 2023-03-19:
Yes people have attached an growatt SPH inverter via serial. The pre build lolin32 binary has an issue. It is propably the best way to build the firmware yourself.

Alternativly you can add a resistor to the lolin32 as described here:
https://github.com/otti/Growatt_ShineWiFi-S/issues/124

Keucar, 2023-03-20:
With self compiled Firmware was the same problem, thanks for the hint with the resistor.

crasu, 2023-03-21:
You have to set these values in Config.h for the lolin32 to work without the resistor:
```
#define AP_BUTTON_PRESSED (0)
#define ENABLE_DOUBLE_RESET 1
```

closed 2023-03-21 by crasu