OpenInverterGateway/OpenInverterGateway #132: Firmware cant detect which stick im using - ShineWifi-X

Sleazified, 2023-03-25:
# What happened?
I flashed the ShineWifi-X with this firmware, and it seems like it can´t detect automatically which stick is in use. Can i do this manually?
[image]

# Stick type

What Stick type are you using?
- Shine X

# Log output
Please copy and paste any relevant log output from /debug and the serial console (You have to compile the firmware with the relevant settings in Config.h to get the output).

```
#0: Setup()
#1: Error: Unknown Shine Stick
#2: Error: Unknown Shine Stick
```

Treffer-hh, 2023-03-26:
Did you try to put it on the inverter USB Port? I had the same issue yesterday while it was plug in in the USB Port of my computer. After pluging it directly to the inverter the detection was succesful and the stick works.

crasu, 2023-03-26:
I do not have an SPH inverter afaik people only had success when connecting with a ESP via serial/rs485.

Sleazified, 2023-03-26:
i managed to flash my seriel stick instead, and it worked almost immidiatly. thanks for the help though.

crasu, 2023-03-27:
Good to hear

closed 2023-03-27 by crasu