OpenInverterGateway/OpenInverterGateway #144: MQTT-settings not being saved

BerndOffGrid, 2023-04-01:
Hey guys,

I managed to flash the "release" firmware via esptool on a raspberry pi, which was kind of a though one for me as a non-pro. 
But now I have the stick up and running again. After configuring the Wifi via the AP-mode I get all the data from the inverter.

But, when trying to enter and save the MQTT-settings afterwards the stick seems to not keep the information.
It exits the AP-mode after pressing "save" immediatly or within a few seconds without any confirmation. 
Should there be any? Am I missing something?
I would have expected it to go back to the main menu within the 300sec of AP-mode or show a confirmation before leaving.

What I tried is adding user&password to the mqtt-broker, which I had not so far, hoping that it would be needed instead of empty strings to successfully save the settings.
Also tried flashing the stick again, hoping that the wifi-information will be gone, so I can enter the MQTT-settings first.
But the Wifi-information from my local network is still there afterwards. So where is the different informations being stored?

Would definitly appreciate anybody to show me a workaround, preferably without flashing the stick again. 
I seem to be lucky that it worked at all for 2 times, because there were multiple attempts needed with really low baud rates (2400-9600)!

Stick used:  ShineWifi-X
Inverter used: MIC-1500-TL-X

Log output:
[image]

Thanks in advance for your support,
Best regards
Bernd.

crasu, 2023-04-10:
The ap needs to restart once fully configured and will change it ssid. For this reason it is quite difficult to redirect the browser to the main inverter page after saving settings.

The wifi info is stored outside of the firmware. You have to completly erase the sticks memory. This is more an esp related issue and has little to do with this firmware.

crasu, 2023-04-12:
The wifi settings are saved by the esp firmware itself. The mqtt are stored on the "filesystem" (littlefs) inside the esp8266. I guess for some reason your esp cannot write to littlefs. 

Can you turn on serial debuging and attach a serial monitor?

BerndOffGrid, 2023-04-12:
Thank you @crasu ,
not sure if i know what to do exactly. 
How do I actively have to "turn on" serial debugging ?

What I tried now is to "screen" the stick's USB-connection to my raspberry. This is the output while entering AP-mode and trying to save MQTT-parameters:
[image]

Btw, I noticed, that my stick doesn't have the CR1220 battery tray soldered on, like I saw it here:
https://user-images.githubusercontent.com/6315513/120305298-e079d900-c2d0-11eb-9a33-e67d96989be5.jpg
Does this affect any function, like keeping information?

Best regards, Bernd.

Sand0rf, 2023-04-16:
I'm having the same issue, I can't seem to save the MQTT settings. Configuring WiFi and reading from the modbus server of the inverter is fine however setting the MQTT settings is not working. Not before the wifi settings nor after setting wifi settings. It is a brand new ShineWifi-X stick (which, just as Bernd, does not have a CR1220 battery so probably a hardware revision).

crasu, 2023-04-16:
@BerndOffGrid @Sand0rf Could you try this change please #163.

@BerndOffGrid The messages in the log look normal. I added an additional debug message on the pr above.

Sand0rf, 2023-04-20:
@crasu that solved the issue. I used the master version as your pr was already merged back

Thanks for you help!

BerndOffGrid, 2023-04-20:
@crasu @Sand0rf 
I couldn't manage to build a binary myself.
Could one of you be so kind and attach a file or publish a new release with that change included? 
Thank you!

closed 2023-04-20 by crasu

BerndOffGrid, 2023-04-21:
Ok, in the meantime, I managed to build a file from the current master via the git clone option, as of todays early state, which contains the latest changes (right?). At least I can see the changes in the files.

I still have the same behaviour on my stick. 
Tested it on the inverter, mqtt-settings still not being saved. MQTT-broker was running as well.
Same log messages than before. Nothing new showing up on my MQTT-explorer.
Also keeping AP-Button pressed doesn't do anything. Thought this would be different after issue#162 being solved.
Firmware update via the webserver doesn't work as well, I get "Not found: /firmware".
All tested with different browsers (firefox, edge).

Recognized my stick having a the red LED on constantly, green one blinking somehow. No idea what that means.

[firmware.zip](https://github.com/otti/Growatt_ShineWiFi-S/files/11296088/firmware.zip)

Sommerwl, 2023-04-23:
Hi, vielen Dank für die tolle Arbeit.

Ich bin etwas überfordert und komme nicht weiter.

Nutze aktuell die fertige  firmware-ShineWifiX.bin  vom Jan 28.
Lese die Parameter aus der JSON Datei aus und nutze sie für EVCC.

Wäre es möglich dass ihr eine fertige BIN Datei erstellt für einen WIFIX-Stick?
- MQTT aktiviert
- Firmware update aktiviert
- neustes Release 

Oder eine Anleitung für Dummys ...wie ich das zusammengestellt bekommen.

Danke und viele Grüße

BerndOffGrid, 2023-04-23:
Hallo @Sommerwl ,
kann dein Anliegen gut verstehen, bin selber neu in dem Thema (github, compilieren, usw.).
Ich habe an meinen Post davor eine Binärdatei angehängt, kann Du die sehen, runterladen?
Darin habe ich Firmwareupdate per Weboberfläche aktiviert und im Master waren die neusten Änderungen drin.

MQTT sollte standardmäßig aktiviert sein, siehe Zeile "#define MQTT_SUPPORTED 1" in der config.h.
Nur bei mir funktioniert es halt nicht die Eingaben für den Broker zu speichern. Blöd.

Ich habe eine TX-L Inverter und scheinbar funktioniert das eingestellte Modbus-Protokoll 124 damit.
Ansonsten auch in der config.h ändern.

Wo und wie kann man eine JSON-Datei auslesen?

Neue Firmware kannst du selbst relativ einfach (ich weiß, sagt sich leicht hinterher) erstellen.
Was bei mir funktioniert hat war Visual Studio Code auf Win10 mit PlatformIO-Erweiterung. Dann per git clone (Anleitungen gibt auf Youtube usw.) den aktuellen Master (oder eine branch Variante auswählen) lokal speichern und ins VSCode laden (den Growatt...Unterordner per File/open folder). Da kann/sollte man dann auch direkt die config.h.example umbenennen (Rechtsklick auf Datei in SRC-Unterordner/rename) oder Anpassungen machen, wie eine Flag setzen und speichern.
Dann in der blauen Leiste unten die Umgebung (env) einstellen.
Dann build (Haken in der Leiste) drücken und das Programm machen lassen. Die *.bin kannst dann aus dem Unterordner .pio/build/... rausfischen. Kopier sie und speichere sie an anderer Stelle, denn bei neuen Aktionen wird das Verzeichnis wieder geleert.

Gruß Bernd.

Sommerwl, 2023-04-25:
Hallo @BerndOffGrid,

danke für die Antwort.
Ich werde mal deine Datei auf meinen Stick spielen und schauen was passiert.
Hatte mich nur gewundert warum die so klein ist, im Vergleich zu der bereitgestellten Bin.

Der Link zur Json wird auf der WEB-Oberfläche angezeigt.
[image]

In der Json sind alle wichtigen Daten aus meinem SPH4600 enthalten.

kleine Fehler (Beispiel Output-Leistung kann eigentlich nicht größer als Input-Leistung sein) 
InputPower: 160 W
OutputPower: 186.7 W  
BatteryTemperature: 2.7 C -> der Fehler mit der Nachkommastelle

Lese die Parameter aus und nutze sie für meine anderen Anwendungen.
Über MQTT wäre es denke ich einfacher....auch das bekomme ich nicht hin :-(

Ansonsten muss ich mich mal mit dem erzeugen einer Bin-Datei beschäftigen....ist ja nicht wirklich einfach wenn man das noch nie gemacht hat.

Hatte gehofft es gibt bald eine neue vorkompilierte Version.

Viele Grüße 
Sommerel

Sommerwl, 2023-04-28:
@BerndOffGrid ,

bei mir funktioniert jetzt alles ...eigentlich ganz einfach :-)
Mann muss nur die Anleitung befolgen...

sveba, 2023-08-06:
Ich habe auch das Problem, dass die MQTT-Settings nicht gespeichert werden können. 
Ich benutze die 2.0.1 vorkompilierte Version mit einem Wemos D1 Mini Serial C340. 
JSON-Abruf gunzt gut.
Mein WR ist ein min-4600tl-x

Ich habe versucht auch selber zu kompilieren. Hat auch alles fehlerfrei geklappt, allerdings war der Wemos danach in restart Loop. Keine Ahnung warum. Vllt. liegt es an meinem MacBook M1 (glaube eher nicht)!?  In der seriellen Konsole kam jede Sekunde folgende Meldung:
```
ets Jan 8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
```

Kann man für MQTT auch FQDN benutzen? Oder nur IPs? Auch public IPs?

Auf jeden Fall danke für das tolle Projekt!

crasu, 2023-08-06:
Could you open a new issue and attach the logfile when you save your settings.

sveba, 2023-08-06:
I found out what my problem was. I've set the AP password to a less than 8 characters. This let the ESP to go into boot loop. Now everything is OK