OpenInverterGateway/OpenInverterGateway #147: Battery State is always 0

FirstWithThisName, 2023-04-01:
# What happened?
The battery is state is allways 0, as well as all the other battery values.

# Stick type

What Stick type are you using?
- Shine X

Which Inverter are you using?
- MOD 8000TL3-XH

tbasnoopy, 2023-04-02:
Hi,

which protocoll do you use?

FirstWithThisName, 2023-04-02:
124, I tried the others as well.
But 124 seams to be the only that supports  battery information.

FirstWithThisName, 2023-04-02:
> Can you make a manual Modbus query (/postCommunicationModbus) with following settings?
1041

Read 16b Input register 1041 with value 0

tbasnoopy, 2023-04-02:
> > Can you make a manual Modbus query (/postCommunicationModbus) with following settings?
> > 1041
> 
> Read 16b Input register 1041 with value 0

Hm... looks like the Information could be read and is 0.
Soo.... It could be, that your Inverter is not (fully) supported.

Are you able to reverse engineer which register could be the battery state?
Can you provide some information about the protocol (google?).

tbasnoopy, 2023-04-02:
Do you want to test if your specification is like this?
https://www.photovoltaicsolar.in/Growatt_Manual/MAX%20Series%20Modbus%20RTU%20Protocol.pdf
There are a lot of battery specific register.

tbasnoopy, 2023-04-03:
You can take a look into my fork/branch. I will implement some of the protocols from the pdfs I found.
No guarantee that this branch is building/working. It's work in progress and I cannot test these protocols.
https://github.com/tbasnoopy/Growatt_ShineWiFi-S/tree/moreProtocols

When it's done, i will make a push request...

FirstWithThisName, 2023-04-03:
I don't know how to find the correct registers, because I don't know the exact battery state.

[out-sorted.csv](https://github.com/otti/Growatt_ShineWiFi-S/files/11141752/out-sorted.csv)

I read out the first 4096 registers, I think the battery status is around 50 percent but there are very much values.
Register | Value
--- | ---
104 | 14
186 | 14
3086 | 14
3238 | 14
3189 | 15
3195 | 16
307 | 25
3134 | 30
287 | 32
3171 | 32
3215 | 32
113 | 35
289 | 36
305 | 36
293 | 37
3200 | 37
3201 | 37
291 | 38
295 | 38
297 | 39
39 | 41
43 | 41
47 | 41
303 | 42
299 | 45
301 | 46
3190 | 57
3101 | 59
114 | 60
3115 | 60
3035 | 71
3027 | 72
3031 | 72

These are my guesses but they don't seam to match with anything in the docs.

tbasnoopy, 2023-04-03:
Puh... when you have to wait until I have done all the protocols I found.
Or you look directly in the papers:

- https://watts247.com/manuals/gw/GrowattModBusProtocol.pdf  (this is my 011)
- https://www.amosplanet.org/wp-content/uploads/2022/12/%E9%80%9A%E4%BF%A1%E5%8D%8F%E8%AE%AEMAX_Growatt-Inverter-Modbus-RTU-Protocol_II-V1_13_20190116-English.pdf
- https://www.scribd.com/document/592776725/Growatt-Inverter-Modbus-RTU-Protocol-II-V1-24-English-new
- https://forum.iobroker.net/assets/uploads/files/1550402978050-growatt-pv-inverter-modbus-rs485-rtu-protocol-v3.05-%E5%AE%A2%E6%88%B7%E7%89%88.pdf
- https://www.photovoltaikforum.com/core/attachment/265072-pdfcoffee-com-growatt-inverter-modbus-rtu-protocolii-v120-english-pdf-free-pdf/   (this could be 120)

FirstWithThisName, 2023-04-03:
3171 seams to be a hot candidate for soc. The value seams reasonable.
I just changed the line in Growatt124.cpp to 
```
  Protocol.InputRegisters[P124_SOC] = sGrowattModbusReg_t{
      3171, 0, SIZE_16BIT, "SOC", 1, PRECENTAGE, true, true};  // #34
```
The returned value in the json is still 0.
But the value returned from postCommunicationModbus is for the register 3171 is 10.

tbasnoopy, 2023-04-04:
It is not as easy as this. Sorry.
This is untested but should work for you:
https://github.com/tbasnoopy/Growatt_ShineWiFi-S/tree/FirstWithThisName

FirstWithThisName, 2023-04-04:
> It is not as easy as this. Sorry. This is untested but should work for you: https://github.com/tbasnoopy/Growatt_ShineWiFi-S/tree/FirstWithThisName

Thank you, i tried it but the value is for SOC is still 0%, should it be called something else?
I also needed to remove the ByteOrder_t things because the aren't defined

tbasnoopy, 2023-04-05:
Oh. Both are called SOC. I renamed it to "SOC_NEW" in my branch.

FirstWithThisName, 2023-04-05:
Ok, but SOC_NOW is not find in the json or in the web frontend.

tbasnoopy, 2023-04-05:
Did you pulled and built the last version of my fork (for you)? Because I tested it on my Inverter and the value is at least in the JSON (one of the last).

But my Inverter is another model, so I got no value here.

FirstWithThisName, 2023-04-05:
Jup, just replaced the ByteOrder_t stuff with the old code.

FirstWithThisName, 2023-04-06:
Nope, that is not the problem, I changed the html and that worked.

tbasnoopy, 2023-04-08:
So does the new value work for you?

jkoner, 2023-06-09:
I've added the list of fields that should work for you to the following comment:
https://github.com/otti/Growatt_ShineWiFi-S/issues/197#issuecomment-1583037566

crasu, 2023-07-29:
Feel free to open a PR for the missing values. I do not have a MOD 8000TL3-XH to test this.

closed 2023-07-29 by crasu

FirstWithThisName, 2023-08-21:
@tbasnoopy   thank you the latest version is working now.

WaizenErnter, 2023-10-16:
I have the exact same hardware, but the latest reposetory doesnt work for me. When i try to build the fork from @tbasnoopy i get an error for missing files. What is your working config @FirstWithThisName? Can you post your working .bin file here Please.