OpenInverterGateway/OpenInverterGateway #148: MQTT Password lenght

AditaNi, 2023-04-02:
Hi Otti,

Please change the mqtt password length inside of ShineWifi-ModBus.ino file, from: 
custom_mqtt_pwd = new WiFiManagerParameter("password", "mqtt password", mqttConfig.mqttpwd.c_str(), 40);
to:
custom_mqtt_pwd = new WiFiManagerParameter("password", "mqtt password", mqttConfig.mqttpwd.c_str(), **64**);

This is because the latest version HA with Mosquitto MQTT Broker generates by default a password wich is 64 characters long.

This is a easy one! :)

Regards
Adi

tbasnoopy, 2023-04-03:
> This is a easy one! :)

When why not make a pull request? ;-)

AditaNi, 2023-04-03:
Hi Otti,

i do know myself with this part(pull request). I'm learning but too slow.
But in one day i will manage .

I would like to update also the web part....

Regards
Adrian

tbasnoopy, 2023-04-03:
https://github.com/otti/Growatt_ShineWiFi-S/pull/149

tbasnoopy, 2023-04-03:
I made a pull request with this change.

BTW. I'm not Otti

PapaSpik3, 2023-04-03:
Hi tbasnoopy,

I have also some changes and I'm unable to make a pull request. Is it posible that you merge them :)?

best regards

tbasnoopy, 2023-04-03:
Hi @PapaSpik3 
I'm not able to merge, too. But everyone can make a pull request. I'm not sure what exactly you meann

PapaSpik3, 2023-04-03:
eg I added "wm.setHostname(HOSTNAME);" in ShineWiFi-ModBus.ino. Now the device will be displayed in DHCP/DNS correctly with the hostname (before it was the ESP ID). Or I changed "bool res = wm.autoConnect("GrowattConfig", APPassword);" to "bool res = wm.autoConnect(HOSTNAME, APPassword);" beecause if you have more then one you couldn't decide between the APs...

I have it local but i dont know how to send it from platform.io to a pull request.

crasu, 2023-04-04:
@PapaSpik3  Contributions are always welcome. Please have a look here how the github flow which includes opening a pr works: https://docs.github.com/de/get-started/quickstart/github-flow

Its a bit of a learning curve but at the end you will know how to contribute to almost every github project.

crasu, 2023-04-04:
original issue solved on master

closed 2023-04-04 by crasu