OpenInverterGateway/OpenInverterGateway #155: replace the firmware over SUB-D of the Shine-S without level-shifting and inverting the Tx/Rx signal ?

JoostDkr, 2023-04-07:
# What happened?
I like to replace the firmware of the Shine-S with a SUB-D 9  RS232 connector. 
Can I for that connect the Shine-S to the PC with a "RS232 null modemcabel  9-pins SUB-D with two female connectors"? 
Only Tx, Rx and ground are connected  between the connectors **without level-shifting and inverting the Tx/Rx signal** ! 
The 8 volt is connected to pin 9 of the Shine-S.

# Stick type

What Stick type are you using?
- Shine S
- 
Which Inverter are you using?
Growatt 3000S

crasu, 2023-04-12:
Here is the description how to flash the shine s https://github.com/otti/Growatt_ShineWiFi-S/blob/master/Doc/ShineWiFi-S/ShineWiFi-S.md
I do not have one so I cannot help you.

closed 2023-04-12 by crasu

JoostDkr, 2023-04-15:
It works fine! No level-shifter/invertor needed!) I used:
- a usb ->rs232 cable (SUB-D9 male)
- a null model rs232 cable (SUB-D9 female-female. Rx-Tx crossed)
- Connect Sub-D: Pin-9 to +9Volt and Pin-5 to ground.

1/Switch off ShineWifi-S
2/ Connect GPIO-0 to ground
3/ Switch on ShineWifi-S
4/ Disconnect GPIO-0
5/ Start flash from flash program (I used VB-PlatformIO) with boudrate 57600 (115200 gives problems)
6/ Switch off, on the ShineWifi-S to leave the flashmode.

[image]