OpenInverterGateway/OpenInverterGateway #162: ShineWiFi-X wont compile, PlatformIO

patsik, 2023-04-14:
Thank you for your great work! 
PlatformIO noob here, installed vscode and platformIO on Win10 just for this project. Also, this is the first time im posting an issue to Github. Please forgive my mistakes. I am not a programmer, only experience is Arduino IDE. I'd like to learn and understand why env ShineWiFi-X won't compile (while env nodemcu-32s and lolin32 compile without any issues).
Installed precompiled Bin without issues. It's working but i like to adjust some info on the shinewifix ip page/chart. I'd like to remove some of the (battery) entries. Would like to add PV1 voltage & amps, grid voltage.

# What happened?
Installed vscode and platformIO on Win10. Cloned Git project using repository URL. Chose correct env like instructed in readme.md file. Renamed config.h.example to config.h without any further changes.
The project started compiling, i copied this from terminal: (sorry for overflow..)

Processing ShineWifiX (platform: espressif8266; board: esp07s; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp07s.html
PLATFORM: Espressif 8266 (4.2.0) > Espressif Generic ESP8266 ESP-07S
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 3.30102.0 (3.1.2)
 - tool-esptool @ 1.413.0 (4.13)
 - tool-esptoolpy @ 1.30000.201119 (3.0.0)
 - toolchain-xtensa @ 2.100300.220621 (10.3.0)
Converting ShineWiFi-ModBus.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep+, Compatibility ~ soft
Found 43 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoOTA @ 1.0
|   |-- ESP8266WiFi @ 1.0
|   |-- ESP8266mDNS @ 1.2
|   |   |-- ESP8266WiFi @ 1.0
|-- PubSubClient @ 2.8.0
|-- ModbusMaster @ 2.0.1
|-- ESP8266-ping @ 2.0.1
|   |-- ESP8266WiFi @ 1.0
|-- WiFiManager @ 2.0.12-beta+sha.94bb903
|   |-- ESP8266WiFi @ 1.0
|   |-- ESP8266WebServer @ 1.0
|   |-- DNSServer @ 1.1.1
|   |   |-- ESP8266WiFi @ 1.0
|-- ESP_DoubleResetDetector @ 1.3.1+sha.bce10ef
|-- ArduinoJson @ 6.19.4+sha.67b6797b
|-- ESP8266WiFi @ 1.0
|-- ESP8266WebServer @ 1.0
|-- DNSServer @ 1.1.1
|   |-- ESP8266WiFi @ 1.0
|-- LittleFS @ 0.1.0
Building in release mode
Compiling .pio\build\ShineWifiX\src\Growatt.cpp.o
Compiling .pio\build\ShineWifiX\src\Growatt120.cpp.o
Compiling .pio\build\ShineWifiX\src\Growatt124.cpp.o
The system cannot find the file specified.
The system cannot find the file specified.
Compiling .pio\build\ShineWifiX\src\Growatt305.cpp.o
The system cannot find the file specified.
Compiling .pio\build\ShineWifiX\src\GrowattSPF.cpp.o
The system cannot find the file specified.
The system cannot find the file specified.
Compiling .pio\build\ShineWifiX\src\ShineMqtt.cpp.o
The system cannot find the file specified.
Compiling .pio\build\ShineWifiX\src\ShineWiFi-ModBus.ino.cpp.o
Compiling .pio\build\ShineWifiX\src\WebDebug.c.o
*** [.pio\build\ShineWifiX\src\Growatt.cpp.o] Error 1
*** [.pio\build\ShineWifiX\src\Growatt120.cpp.o] Error 1
*** [.pio\build\ShineWifiX\src\Growatt124.cpp.o] Error 1
*** [.pio\build\ShineWifiX\src\Growatt305.cpp.o] Error 1
*** [.pio\build\ShineWifiX\src\GrowattSPF.cpp.o] Error 1
*** [.pio\build\ShineWifiX\src\ShineMqtt.cpp.o] Error 1
The system cannot find the file specified.
*** [.pio\build\ShineWifiX\src\ShineWiFi-ModBus.ino.cpp.o] Error 1
The system cannot find the file specified.
[... 477 lines cut ...]
Building .pio\build\lolin32\firmware.bin
esptool.py v4.5
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
========================================================================== [SUCCESS] Took 49.05 seconds ==========================================================================

Environment    Status    Duration
-------------  --------  ------------
ShineWifiX     FAILED    00:00:03.422
nodemcu-32s    SUCCESS   00:04:08.330
lolin32        SUCCESS   00:00:49.054
====================================================================== 1 failed, 2 succeeded in 00:05:00.806 ======================================================================

 *  The terminal process "C:\Users\[]\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 

# Stick type
What Stick type are you using?
- Shine X

Which Inverter are you using?
MIN 5000TL-XE
"The system cannot find the file specified." < this is driving me insane. Obviously the files are available because nodemcu and lolin32 compile without issues.
Tried troubleshooting basic stuff like windows file permissions and windows file path length, virus scanner disabled, reinstall platformio, used Google and tried some basic things like adding -I SRC/ to the ini file. This was without succes and i seem unable to understand why shinewifix won't compile while nodemcu and lolin32 do. Does it have something to do with the difference in platforms ea Espressif8266 vs Espressif32? 
Any help would be deeply appreciated!

tbasnoopy, 2023-04-15:
I have a build issue, too.
Does it help to change the line platformio.ini (line 38) from
    '-DAP_BUTTON_PRESSED=analogRead\(A0\)\<50'
to
    '-DAP_BUTTON_PRESSED=0'

patsik, 2023-04-15:
Thanks a lot for your input! With your suggested change i'm able to compile the project, have to confirm on the inverter  tomorrow because for me it's night time now.
I have 1 more question though: Would you like to share the troubleshooting steps you took to solve this problem? I mean, how did you get from 'compiler failed' to this working solution? I hope i'm not asking to much, just trying to learn (:

crasu, 2023-04-16:
We probably have an escaping issue here for windows users ;-(. I cannot really debug this becuase I am on linux. @patsik could you try removing one char at a time to figure out which character is making the compilation fail?

crasu, 2023-04-16:
@patsik Could you check if #162 fixes your issue? You have to copy the Config.h file again.

patsik, 2023-04-16:
Renamed existing Platformio.ini and config.h, added the files from the commit. Cleaned project, build.

Failed throwing the following:
Building in release mode
Compiling .pio\build\ShineWifiX\src\Growatt.cpp.o
Compiling .pio\build\ShineWifiX\src\Growatt120.cpp.o
Compiling .pio\build\ShineWifiX\src\Growatt124.cpp.o
Compiling .pio\build\ShineWifiX\src\Growatt305.cpp.o
Compiling .pio\build\ShineWifiX\src\GrowattSPF.cpp.o
Compiling .pio\build\ShineWifiX\src\ShineMqtt.cpp.o
Compiling .pio\build\ShineWifiX\src\ShineWiFi-ModBus.ino.cpp.o
Compiling .pio\build\ShineWifiX\src\WebDebug.c.o
xtensa-lx106-elf-g++: error: SRC\ShineWiFi-ModBus\ShineWiFi-ModBus.ino.cpp: No such file or directory
xtensa-lx106-elf-g++: fatal error: no input files
compilation terminated.
*** [.pio\build\ShineWifiX\src\ShineWiFi-ModBus.ino.cpp.o] Error 1

After this, hit build again, succeeded. 
Uploaded firmware.bin found at .pio\build\ShineWifiX\
Nothing happens when pressing the button.

----update about 2h later:-----
I took a dive in the code, i think i found the culprit of the ap button not working in config.h:
Changed this 
#if AP_BUTTON_PRESSED == 1
to this
#if ENABLE_AP_BUTTON == 1

Now when keeping the button pressed for a few seconds ap mode is started. Awesome!
thanks a lot!

closed 2023-04-16 by crasu

tbasnoopy, 2023-04-16:
> Thanks a lot for your input! With your suggested change i'm able to compile the project, have to confirm on the inverter tomorrow because for me it's night time now. I have 1 more question though: [...]

I work a lot with these options and I thought I broke this only local. So.... sorry but I have no clue how the issue was reported.