OpenInverterGateway/OpenInverterGateway #164: Shine X Stick with CH340

rober1982, 2023-04-15:
# What happened?
I was able to flash and configure the latest release to my Shine X Wifi Stick. After connecting it to the Inverter it does not receive any data. When I checked the log I found two entries about "unknown hardware stick". When I checked the stick I found a CH340 Chip - not an exar. Is there a way to make it work? 

# Stick type

What Stick type are you using?
- Shine X

Which Inverter are you using?
- MIC-2000-TL-X

# Log output
#0: Setup()
#1: Error: Unknown Shine Stick

Any help is welcome.

Kind regards

Robert

patsik, 2023-04-15:
I really not the person to offer you the best technical assistance (because i'm a noob, lol), but i can confirm 1 Shinewifi X with CH340 working and connected to the inverter. I uploaded the precompiled release provided at the main page of this repository. Our inverters are different though (i have a MIN-5000-TL-X).

crasu, 2023-04-16:
CH340 or exar serial chip should not matter at all. Try to turn off the inverter via the DC disconnect switch and try again.

closed 2023-04-16 by crasu

rober1982, 2023-04-16:
Thanks, it did work this time...

Closed #164<https://github.com/otti/Growatt_ShineWiFi-S/issues/164> as completed.