OpenInverterGateway/OpenInverterGateway #167: Bat First holding registers not saving

schmellic, 2023-04-16:
# What happened?
Trying to change bat first charge periods on holding register 1100 and 1101, UI says value written but upon read value has not changed

# Stick type

What Stick type are you using?
- Shine S

Which Inverter are you using?
SCH6000

schmellic, 2023-04-18:
Just been having a look at the 1.24 modbus protocol documents online and it looks like the holding registers for battery charge times are not writable.

The only writable register I can find that is related to time is 3038 and 3039 but I'm not sure these are the correct ones for setting battery charge times, or what value I should be populating them with if they are the correct registers.

Has anybody else had success in changing the AC charge times as I'd like to be able to set them to match the cheapest Octopus Agile rates which I used to do via the growatt API.

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/e0402865/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/e0402865/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">
<span style='font-variant-ligatures: normal;font-variant-caps: normal;
orphans: 2;text-align:start;widows: 2;-webkit-text-stroke-width: 0px;
text-decoration-thickness: initial;text-decoration-style: initial;text-decoration-color: initial'>

Register   NO. | VariableName | Description | Writeor not | Value | Unit | Initialvalue | Note
-- | -- | -- | -- | -- | -- | -- | --
1091 | wBatFirst stop SOC | Stop Charge soc when Bat First | W | 0-100 | 1% |   | Stop Charge soc when Bat First
1092 | AC charge Switch | When Bat First     Enable:1     Disable:0 |   | Enable:1     Disable:0 |   |   | AC Charge Enable
1100 | Bat First Start Time 1 | High eight bit:   hour     Low eight bit: minute |   | 0-23     0-59 |   |   |  
1101 | Bat First Stop Time 1 | High eight bit:   hour     Low eight bit: minute |   | 0-23     0-59 |   |   |  
1102 | BatFirst on/off Switch 1 | Enable:1     Disable:0 |   | 0 or 1 |   |   |  
  |   |   |   |   |   |   |  
3038 | Time 1(xh) | Period 1:   [Start Time ~ EndTime], [Charge/Discharge],[Disable/Enable]          3038 enable, charge and discharge, start time, endtime 3039 | R/W |   |   |   | Bit0-7: minutes;     Bit8-12: hour;     Bit13-14,     0: load priority;     1: battery priority;     2: Grid priority;     Bit15,     0: prohibited;      1:enabled
3039 |   |   | R/W |   |   |   | Bit0-7: minutes;     Bit8-12: hour;     Bit13-15: reserved

<span style='font-variant-ligatures: normal;font-variant-caps: normal;
orphans: 2;text-align:start;widows: 2;-webkit-text-stroke-width: 0px;
text-decoration-thickness: initial;text-decoration-style: initial;text-decoration-color: initial'>
</body>

</html>

schmellic, 2023-04-21:
I've been doing a bit of digging on the HA forum around this issue (mainly because I don't have the technical knowledge to much else) and another user using ESPHome and the same inverter has also been having this issue.  

They have done a fair big of digging on the issue and have found that all 3 registers for start time, end time and enable need writing at once.

Link to forum post and more information below
[https://community.home-assistant.io/t/esphome-modbus-growatt-shinewifi-s/369171/228?u=schmellic]

crasu, 2023-04-23:
Closing this as this is more an issue with the inverter itself and less with the sticks firmware.

closed 2023-04-23 by crasu

schmellic, 2023-04-24:
Not sure I agree it's an "issue" with the inverter, I can save the values fine with ESPHome when writing the 3 registers at the same time. 

More the web r/w functionality of the web server doesn't support this function / inverter.

PR #210 (2023-06-29): Add battery / grid first commands
This PR adds new commands for the v1.24 modbus version that allows the user to read and update the various battery and grid first settings, which allows them to e.g. charge their battery from the grid overnight. It is a 1:1 match of the 'Battery First' and 'Grid First' settings on the official Growatt server config page.

I was unable to get the 'Battery First1' and 'Grid First1' equivalents functional as the inverter responded very strangely to setting the values as documented so I have left them out, it is unlikely to have more than 3 schedules anyway.

I ha [... cut]