OpenInverterGateway/OpenInverterGateway #175: Cant connect with NodeMCU ESP8266

robeving, 2023-04-30:
# What happened?
I took the latest master branch and compiled it for my NodeMCU ESP8266 Lolin V3. I need to change the platformio.ini to include:
```
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
lib_deps =
    ArduinoOTA
    knolleary/PubSubClient@^2.8
    4-20ma/ModbusMaster@^2.0.1
    bluemurder/ESP8266-ping@^2.0.1
    vshymanskyy/Preferences@2.1.0
    https://github.com/tzapu/WiFiManager#94bb90322bf85de2c9ec592858f20643161bc11f
    https://github.com/khoih-prog/ESP_DoubleResetDetector#bce10ef01f3d4864a07ef31fdec2ad65adb3e5b8
    https://github.com/bblanchon/ArduinoJson#67b6797b6d19e944b01213926872f955f4b9d54d
build_flags =
    ${env.build_flags}
    "-D MQTT_MAX_PACKET_SIZE=2048"    
    "-D ENABLE_AP_BUTTON=1"
```

Successfully wrote to the device, wifi set up. But when I plug the device into the USB port I get `{"InverterStatus": -1 }` over MQTT. I already have a black stick of some kind plugged into the serial connection which works via the GroWatt portal.

# Stick type
NodeMCU ESP8266 Lolin V3 - says it uses CH340

Which Inverter are you using?
- SPH-5000TL3 BL-UP

# Log output

```
File Not Found

URI: /debug
Method: GET
Arguments: 0
```

crasu, 2023-04-30:
Did you enable debug logging in the config?

robeving, 2023-04-30:
```
#define ENABLE_WEB_DEBUG 1
```

The log output above is what was returned from the /debug endpoint

robeving, 2023-04-30:
OK so i've swapped to use a NODE MCU ESP32-S. Seem to have deployed better but I get the following in the log 

```
#0: Setup()
#1: Error: Unknown Shine Stick
```

robeving, 2023-04-30:
Still getting 
```
{"InverterStatus": -1 }
```
in the MQTT topic

robeving, 2023-05-01:
OK giving up on this. I've tried a lot of different combinations and even another project and I cant get my inverter to talk modbus. I'm going to buy a Shine-S and replace the firmware on that and talk to the RS232 port directly.

If someone in the future finds this by searching for USB or similar...give up

closed 2023-05-01 by robeving

crasu, 2023-05-03:
I am pretty sure that it works. I guess I have to get a BL-UP myself so I can support people better ;-).

robeving, 2023-05-03:
All I can say is that I've tried two different embedded boards. Confirmed they have the CHG340 on Linux, even taken out the serial checks in Setup() so it just assumes its connected to USB. I've tried another project that also says it can talk USB and that failed in the same place. 

In all cases it tries to talk modbus and get invalid results back.

If you've got any other ideas I'm willing to try them.

robeving, 2023-05-03:
Tried everything again and with a full reboot

puchisco, 2023-08-26:
Same results here with a NodeMCU ESP8266 v1.0 and v3 on Inverter SPF3500ES. (CHG340 both)
Tried too with diferents projects and nothing. I can read all the records through the USB-B port connected to the PC with Modbus Poll, but it is impossible through the "Wifi/GPRS" port.
I am open to suggestions and if someone with this investor model has obtained results, I would appreciate your help