OpenInverterGateway/OpenInverterGateway #178: Shiniwifi-X precompiled does not open AP

Ac0ps, 2023-05-02:
# What happened?
Also tell us, what did you expect to happen? Tell us what you see!

The issue is that after the flash of the pre compiled firmware and then re plugging the usb stick it doesnt open an wifi ap. 

So, i opened the housing of the PCB, brigde the ground with GPIO0. 
Downloaded the firmware for the WIFI-X, open the https://espressif.github.io/esptool-js/.
Now i connect the stick in to my pc and the esp tool sees the serial port and connects.

*see logs*

Now i select the correct firmware downloaded from github and flash it.

*see logs*

So i disconnect my stick, unbridge GND and GPIO0. reconnect the stick.
But now webserver / AP gets formed

# Stick type

What Stick type are you using?
- Shine X
=

Which Inverter are you using?
- MIC-2000-TL-X

# Log output
pre compiled

Insert logs here

ptool.js
Serial port WebSerial VendorID 0x1a86 ProductID 0x7523
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: b4:8a:0a:c0:91:0f
Uploading stub...
Running stub...
Stub running...
Changing baudrate to 921600
Changed
Compressed 411888 bytes to 285913...
Writing at 0x1000... (5%)
Writing at 0x6409... (11%)
Writing at 0xc090... (16%)
Writing at 0x11ee9... (22%)
Writing at 0x17ffa... (27%)
Writing at 0x1df69... (33%)
Writing at 0x23774... (38%)
Writing at 0x29098... (44%)
Writing at 0x2e945... (50%)
Writing at 0x33af0... (55%)
Writing at 0x38abb... (61%)
Writing at 0x3dc65... (66%)
Writing at 0x4352e... (72%)
Writing at 0x48ca1... (77%)
Writing at 0x4e047... (83%)
Writing at 0x5361e... (88%)
Writing at 0x5c065... (94%)
Writing at 0x611da... (100%)
Wrote 411888 bytes (285913 compressed) at 0x1000 in 5.284 seconds.
Hash of data verified.
Leaving...

I hope you can help me

BerndOffGrid, 2023-05-05:
Hi, 
I'm not a specialist and no idea what the issue could be, but as I used the esptool as well, it might be helpful to talk about that first. 
I used the installed versions for Win10 as well as for the latest debian OS.
Both basically worked, but there were options to choose from and not very much documentation of it.
I generally used lower baud rates, but the process would not complete if this was the issue I guess.
I also used the precompiled one first, which worked with the MIC TL-X series.

Maybe you have to completely erase the ESP8266 flash memory first?

Regards Bernd.

v1rus317, 2023-05-10:
Hi,
i had the same issue that you had. I also used the Espressif ESP-Tool in the Browser.

The solution was to at first erase the Flash again and then flash the Image again but start the Flash not at address 0x1000 but at address 0x0000.

crasu, 2023-05-10:
@Ac0ps do you see anything on the serial monitor?

Ac0ps, 2023-05-10:
> Hi, i had the same issue that you had. I also used the Espressif ESP-Tool in the Browser.
> 
> The solution was to at first erase the Flash again and then flash the Image again but start the Flash not at address 0x1000 but at address 0x0000.

**This seemed to be the solution**

closed 2023-05-10 by Ac0ps