OpenInverterGateway/OpenInverterGateway #179: Protokoll 120 Problem

Bernd9000, 2023-05-03:
1.)
In der Growatt.cpp Zeile 391 steht folgendes:

[image]

Müßte hier nicht im markierten Bereich ebenfalls "HoldingRegisters" stehen ?

2.)
Ich habe beim Protokoll 120 das Problem das die Holding Register nicht im MQTT Broker veröffentlicht werden.
Das scheint daran zu liegen das es im Protokoll 120 sehr viele InputRegister in der "Growatt120.cpp" gibt.
Es scheint beim "JSONAddReg" irgendeine Grenze zu geben die nicht mehr als 64? einträge zu läßt.
Vertausche ich in der "Growatt.cpp" die beiden folgenden For Schleifen dann sind die Holding Register vorhanden und stehen am Anfang, dafür Fehlen aber einige Input Register.

`#if SIMULATE_INVERTER != 1
  for (int i = 0; i < _Protocol.InputRegisterCount; i++)
    JSONAddReg(&_Protocol.InputRegisters[i], &doc);

  for (int i = 0; i < _Protocol.HoldingRegisterCount; i++)
    JSONAddReg(&_Protocol.HoldingRegisters[i], &doc);`

Hat jemand eine Idee?

crasu, 2023-05-10:
You are probably hitting MQTT_MAX_PACKET_SIZE limit. However as you have not specified which inverter and stick you are using its hard to tell.

Bernd9000, 2023-05-10:
Yes, sorry
Have a Growatt MIC 600 TL-X and tested it with a LOLIN32 Lite.

`build_flags` =   in platformio.ini
That was the Problem, Thanks

What is with Line 391 in `Growatt.cpp` ? Is that correct ?

crasu, 2023-06-29:
@Bernd9000 Yes line 391 is a bug. However there are no holding registers with a multiplier other then 1.0 so this is never triggered.

Bernd9000, 2023-06-29:
okay

closed 2023-06-29 by Bernd9000