OpenInverterGateway/OpenInverterGateway #183: NodeMCU ESP8266EX CH340C not working- should it?

kkro, 2023-05-09:
I have one ShinX with a self compiled version. Everthing ok.
Log: 
#0: Setup()
#1: ShineWiFi-X (USB) found
#2: ReadData() successful
#3: ReadData() successful
#4: ReadData() successful
#5: ReadData() successful
#6: ReadData() successful

When I use this: "AZDelivery D1 Mini V3 NodeMCU ESP8266EX " the inverter will not respond.
#0: Setup()
#1: Error: Unknown Shine Stick
#2: Inverter.last_res: 0xe2
#3: Error: Unknown Shine Stick
#4: Inverter.last_res: 0xe2

#2, #4 is from me because I wanted to know what was the result of 
    res = Modbus.readInputRegisters(0, 1);
in file Growatt.cpp Line 74.

I put this stick to my PC an started a serial sniffer and the output is as expected: 
 01 04 00 00 00 01 31 CA 
= Modbus.readInputRegisters(0, 1);

On my ESP8266EX I connected the LEDs to the same pins as ShinX. Same for the switch at A0. USB/Serial chip: CH340C
https://www.az-delivery.de/products/d1-mini-v3
May the last "C" form CH340C a problem?

I also removed the 9600 baud test for ShineS but no success.

What can I do more? 
Or do the Inverter some check of the bytes that come out of the esp8266 at boot time to do so kind of hardware bundleing?

Inverter: MOD-10KTL3-X from 2022, Protocol 120

crasu, 2023-05-10:
Could you check with lsusb which usb id you are getting?

kkro, 2023-05-10:
Shine-X:
Bus 001 Device 006: ID 1a86:7523 QinHeng Electronics CH340 serial converter

my ESP8266EX:
Bus 001 Device 007: ID 1a86:7523 QinHeng Electronics CH340 serial converter

Thank you very much. This question helped me to solve the issue. 

It did not show up any ID on my Debian but on Windows 11. So it could only be the USB cable and it was the problem. Two different cables did not work, the third worked on Debian and then also on the inverter. 
Unbelivable.

closed 2023-05-10 by kkro